dresden-elektronik/deconz-rest-plugin #3232: How do you set up a IAS?

fribse, 2020-09-05:
## Describe the question or issue you are having

I've just received a series of Heiman smoke alarms (HS1SA), that I've joined to the deconz network. That went rather smootly (a few quirks, but nothing much).
I tried finding out how to set up an IAS so they could warn each other directly, but I couldn't find a guide on how to do that.

## Environment

 - Host system: PVE on NUC
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26580700)
 - deCONZ version: (2.05.79)
 - Device: ConBee II
 - Do you use an USB extension cable: yes - USB3

SwoopX, 2020-09-05:
Sort answer for the moment: This is not a matter of setup, it's not possible.

ebaauw, 2020-09-05:
The whole IAS setup doesn’t cater for IAS device to communicate directly to each other, unlike e.g. wireless controller that control lights directly.  You need a gateway that receives the messages from the device, and translates them into messages for another device.  In our case, the REST API plugin would be the gateway, and you’d use rules to activate the sirens of the other devices when one device reports an alarm.

The _IAS Zone_ cluster, reporting alarms, should be exposed as a `/sensors` cluster, depending on the device type.  The _IAS WD_ cluster, controlling the siren (and strobe light) should be exposed as a `/lights` resource.  So your rules would have a condition on the sensor, and an action on the light.

Do the Heiman sensors have a functional _IAS WD_ cluster?  The loaner I had a couple of years ago had this cluster, but it wasn’t doing anything.  Can you activate the siren in the GUI?

fribse, 2020-09-06:
Looking in the IAS WD I can see two sections.
[image]
The (1) section doesn't do anything.
The (2) section makes noise.

fribse, 2020-09-06:
I've built up a nice flow in Node-Red to handle the alarming now, also announcing on the Google devices where the smoke is detected, so that part is fine. The devices will of course also make noise for themselves, so even if the central system is the cause of the problems, sound will be made :-)

ebaauw, 2020-09-06:
> The (1) section doesn't do anything.

You probably need to set a siren level and warning mode.

> The (2) section makes noise.

That’s promising.  I don’t think I’ve exposed the _Squawk_ command thru the API, but if _Start Warning_ doesn’t work, we could use _Sqawk_ instead.  How long does the noise last?

SwoopX, 2020-09-06:
This is indeed interesting. Might be a small challenge to find the right combination to have the siren triggered. As os now, we have devices reacting on different combinations for the start warning command, which are not necessarily logical. You may check here for further background https://github.com/dresden-elektronik/deconz-rest-plugin/pull/1593#issuecomment-504700737

I quickly checked the specs for the squawk command, no way to define any duration there. It is originally meant to give a "short" audible feedback (e.g. to indicate a system is armed), so it's probably up to the device, how long the signal lasts.

fribse, 2020-09-07:
Ok, I did some testing now that the wife is out the door :-)

The (1) does start if I set warning duration.
Siren level 0,1 gave nothing
Strobe makes the red light in it flash (the number of flashes is controlled by the number given in warning duration (the number 1 gives 1 flash, 2 gives two flases etc.). I'm not sure what the Strobe Duty cycle does?
The warning mode 0, 1 gives siren
The warning mode 2, 3 gives nothing

The (2) gave these results.
The sqawk level 0, 1 and strobe goes for 3 seconds (all with the same wiuwiu sound sirene and flashing), the sound for disarmed goes for 5 seconds?

ebaauw, 2020-09-07:
> The (1) does start if I set warning duration.

Cool.  Normally the siren should apply the default duration (from the corresponding attribute) when none is given, but the GUI probably sends the value from the input field even when it's `0`.

> Siren level 0,1 gave nothing
> The warning mode 0, 1 gives siren
> The warning mode 2, 3 gives nothing

The _Options_ parameter consists of a 2-bit enum for _Siren Level_, a 1-bit flag for _Strobe_ and a 4-bit enum for the _Warning Mode_.  The GUI cannot handle the enum subfields, so I exposed them as individual bits.  Setting only bit 0 in the _Warning Mode_ should work, at least for all sirens I've seen.  The _Siren Level_ is a bit more difficult: some devices will sound the siren at _Siren Level_ 0 (both bits cleared), others remain silent at level 3 (both bits set).  The _Stobe_ bit should control whether the light flashes, but not all devices implement it.  By default, the REST API Plugin sends:
- _Warning Mode_ 1 (bit 0 set), _Stobe_ and _Siren Level_ 3 (both bits set) on `"alert": "select"` and `"alert": "lselect"`;
- All bits cleared on `"alert": "none"` (this should silence the alarm); and
- Only _Strobe_ on `"alert": "blink"`.
The duration is 1 for `"select"` and 300 for `"lselect" and `"blink"`; this can be overwritten by specifying `"ontime"` (so `{"alert": "blink", "ontime": 10}` should flash the lights for 10 seconds.

> I'm not sure what the Strobe Duty cycle does?

It should control the flash pattern: how long the flash is on (in 0.1s) per second.  Again, not all devices implement this.

fribse, 2020-09-07:
Ok, the Duty Cycle certainly changes the flashing pattern as described, do you need me to test further?

ebaauw, 2020-09-07:
Did the API plugin already create a light resource for the Siren?  If not, can you please post a screenshot of the _Basic_ cluster (after reading the attributes) and of the _Node Info_ panel?

fribse, 2020-09-07:
Yes it shows up in deconz like this:

[image]

ebaauw, 2020-09-07:
Then you should be able to control it from the API, doing a PUT of `{"alert": "select"}` to the light state.  Doing a PUT of `{"alert": "all"}` should sound all sirens in the network.  This is a dirty hack of mine to send a broadcast _Start Warning_.  Better specify `"ontime"` and cover your ears.  

Next step is to setup rules to sound the siren(s) when fire is detected.

ebaauw, 2020-09-19:
> I've just received a series of Heiman smoke alarms (HS1SA)
> Looking in the IAS WD I can see two sections.

@fribse, is that the _IAS WD_ cluster of the HS1SA or of the (standalone) Heiman siren?

In case of the HS1SA: which ones are these?  Could you post a screenshot of the _Basic_ cluster (after reading the attributes) and of the _Node Info_ panel?  You might need to briefly press the reset button (hole) to wake the sensor, for the _Read_ to work.

I got my hands on a couple of Zigbee 3.0 HS1SA sensors, with _Model Identifier_ `SmokeSensor-N-3.0`, _Date Code_ `2018.9.27`, and _SW Buid ID_ `2000-0001`, but they report _Unsupported Custer_ (0xC3) when trying to interact with their _IAS_ WD_ cluster.

The good news is that they seem to support over-the-air firmware updates, but I've yet to see a firmware file for a Heiman device...

fribse, 2020-09-19:
That is the Heiman siren
The smoke sensor looks like this:
[image]
[image]

ebaauw, 2020-09-19:
> That is the Heiman siren

Thanks, that makes sense.  The smoke sensor also advertises an _IAS WD_ cluster, so you’d think/hope you could control its siren over Zigbee, but it doesn’t work.

fribse, 2020-09-21:
Hi @ebaauw 
Is it possible to send out the 'alert' command via the API? So I could have my node-red alerting do the 'zone' alert.
I know I can do it via HA to the siren itself, as that is a switch and a light, but as the smoke sensors does not show a switch for this, maybe I can do it directly to the api?

SwoopX, 2020-09-21:
You cannot send a command to a non-existing target. If the functionality is supported, then typically a corresponding resource item (alert) is exposed via the resource (light). If I understand your situation correctly, you neither have any of those.

Besides, HA uses the API.

fribse, 2020-09-22:
Hi @SwoopX 
I was just hoping, since I could activate the alarm from the VNC GUI, that this could be done through the API as well.

ebaauw, 2020-09-22:
@frisbe, are you talking about the smoke sensor or the siren?  You should be able to activate the sound on the siren through the API.  Afaik, the smoke sensor doesn’t support activating the sound over Zigbee, neither in the GUI nor through the API.

fribse, 2020-09-22:
I have the siren working, but you are right, even though the buttons are in the GUI for the smoke sensor, they don't work.
I think it goes in a very deep sleep, because the GUI shows the flashing red when I try to send the command.

ebaauw, 2020-09-22:
It does go into deep sleep, but when you wake the sensor before sending the command, it returns the status for an unsupported cluster, as I mentioned above.

closed 2020-10-20 by fribse