dresden-elektronik/deconz-rest-plugin #3233: How do I delete resourcelinks?

fribse, 2020-09-05:
## Describe the question or issue you are having

I have a 'ghost' item shown in VNC:
I don't get any info when clicking the 'read' attributes, it doesn't 'flash' in the VNC view when doing so.
If I go and just ask for all devices via the REST API, the only place that the MAC address appeares is under resourcelinks.

`"  resourcelinks": {

    "3": {
      "classid": 3010,
      "description": "Resourcelink for switch rule",
      "links": ["/groups/3", "/sensors/14:b4:57:ff:fe:ca:41:b2-01-1000", "/rules/9", "/rules/10", "/rules/11"],
      "name": "opt_pushDimUp",
      "owner": "96B965D58F",
      "recycle": false,
      "type": "Link"
    },
    "4": {
      "classid": 3010,
      "description": "Resourcelink for switch rule",
      "links": ["/groups/3", "/sensors/14:b4:57:ff:fe:ca:41:b2-01-1000", "/rules/12", "/rules/13", "/rules/14"],
      "name": "opt_pushDimUp",
      "owner": "96B965D58F",
      "recycle": false,
      "type": "Link"
    },`

But how do I delete those, so that the item is removed from the VNC? The rules are not there any longer.

## Screenshots

[image]

## Environment

 - Host system: PVE on NUC
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26580700)
 - deCONZ version: (2.05.79)
 - Device: ConBee II
 - Do you use an USB extension cable: yes - USB3

Smanar, 2020-09-05:
Hello, https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3222
Unfortunately delete something in the api have not a lot of impact on deconz.

To delete the ressource you can use curl

fribse, 2020-09-06:
Ok, so even if I delete every trace, it will still be in deconz?

Smanar, 2020-09-06:
yep, but ebaawn have explain why on your other issue.

closed 2020-09-29 by fribse