dresden-elektronik/deconz-rest-plugin #3234: Hive Active Heating Receiver SLR2b (upgraded version of SLR2)

fryc88, 2020-09-05:
## Device

- Product name: Hive Active Heating Dual Channel Recviever
- Manufacturer: Computime
- Model identifier: SLR2b
- Device type : 
  - Thermostat

Endpoints and clusters of the node
[image]

Node Info panel
[image]

Basic Cluster attributes in the Cluster Info panel.

[image]
[image]
[image]
[image]
[image]
[image]

## Screenshots

Smanar, 2020-09-06:
Hello.
I m comparing the 2 devices, what is the difference ?

I can add the device in 5s, but can you test it (if there is difference, because I don't think I can make something bad just with adding this model)

fryc88, 2020-09-06:
Visually there is nearly no difference. By hardware it only says it's been upgraded, and i spoke to peterguy04, who suggested me to add it here, because he says it's more like automated one, so not sure if that will cause any problem.
It's not been connected to boiler yet, but i've got it in my hands, so i can do whatever you want. I've got hub Hive Nano 2 V2 as well, but i'm not planning to use it, but if it can help someone else, i can do some stuff with it as well.

Smanar, 2020-09-06:
If you want to try it.
The procedure is here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
Just replace step 1 by
`git clone --branch slr2b https://github.com/Smanar/deconz-rest-plugin.git`

and skip
`git checkout -b mybranch HEAD`

But I don't think I can be wrong, if you have problem you will just have it on next release.

fryc88, 2020-09-06:
Got it ready, i'll test it tonight. Thanks

fryc88, 2020-09-06:
[image]

All works perfect, thanks to Pete Guy for his time to explain it to me how to get it working, but all seems to be fine (except of 'auto' button, but he explained why, so that's not an issue, i'm not going to use their hub&cloud), and of course you, Smanar for adding it for me.
So now i need to wait for official release and hopefully HA will update it soon :)
Thanks!

Smanar, 2020-09-07:
Yep @ peterguy04 know realy well this stuff ^^.
He have sniffed the hardware, all the protocol used in deconz is from him

fryc88, 2020-09-10:
> 
> 
> Yep @peterguy04 know realy well this stuff ^^.
> He have sniffed the hardware, all the protocol used in deconz is from him

What's gonna be the version number of deconz which will contain it and possibly some ETA? So i'll keep my eye on it to see if HA will bring me the right version :)

Smanar, 2020-09-10:
Probably the next one in some days (new version every 15).
But the PR need to be validated (I don't think we can have problem with this light one) and HA need to use the new version too.

closed 2020-09-15 by manup