dresden-elektronik/deconz-rest-plugin #3238: Recommendation: run deconz as user or root?

Stefan-83, 2020-09-06:
Hi,
I noticed, that there two instances of deconz running?

```
> pi@homebridge:~ $ systemctl restart deconz.service
> ==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
> Authentication is required to restart 'deconz.service'.
> Multiple identities can be used for authentication:
>  1.  ,,, (pi)
>  2.  root
> Choose identity to authenticate as (1-2):
```

Which one is the correct one?
Should I deactivate one?

Thanks again,
Stefan

Smanar, 2020-09-06:
The security recommend don't use root, so better to use pi.

Mimiix, 2020-09-06:
@Stefan-83 Use the following command:

sudo systemctl restart deconz.service

Stefan-83, 2020-09-06:
So, i should deactivate the root-instance, correct?

Mimiix, 2020-09-06:
No, just run the command i wrote :)

Stefan-83, 2020-09-06:
ok, when i fire up your command: "sudo systemctl restart deconz.service" nothing happens.

When i type "systemctl restart deconz.service", i got the choice of root or pi (the user).
Should I deactivate the root-instance?

Mimiix, 2020-09-07:
@Stefan-83 I guess. Because Sudo is like "start as administrator" in Windows.

Stefan-83, 2020-09-07:
when i'm doing "sudo systemctl start deconz", i don't get any errors.. but nothing happens:
[image]

Mimiix, 2020-09-07:
@Stefan-83 Means it started :)
Check the IP of it for it running.

Stefan-83, 2020-09-07:
yes, it's running. When i'm typing "http://phoscon.de/pwa/" into the browser, I got this:
[image]
(with IP 192.168.178.94)
But when i'm typing "http://192.168.178.94:8080/" into the browser, I got this:
[image]

Is that correct?

closed 2020-10-11 by stale[bot]