dresden-elektronik/deconz-rest-plugin #3239: Generic tuya PIR sensor zigbee

SilverWx, 2020-09-06:
## Device
https://www.aliexpress.com/item/4000829510396.html?spm=a2g0s.9042311.0.0.5a056c37KXsW1Q

- Product name: NoName - Generic
- Manufacturer: _TYZB01_zwvaj5wy
- Model identifier: TS0202
- Device type : 
  - Sensor

## Screenshots

[image]
[image]
[image]

SilverWx, 2020-09-11:
@peterguy04 why zigbee 3 sensor (I have Heiman PIR HS1MS-E too) does not work directly with deconz ?

Smanar, 2020-09-12:
Hello.
I think we miss a message ^^.

I can try to include your device, but there is a problem, it s tuya stuff, and I don't have all tuya value, I have one for presence detection but I can't promise it will be the one used for your device.
So you will probably need to use the debug mode, to find the good value (not hard).

Do you know how to compile code for testing ?

SilverWx, 2020-09-14:
Hello, Thank you for answer me :)

I have noticed that Zone type change to "Motion Sensor" and Zone Status to 0x0001 when motion is detected, maybe this can help.

I could try to build debug version of deconz but I run deconz inside hassio, So I have to stop the addon and install deconz in linux ?

Smanar, 2020-09-15:
> I have noticed that Zone type change to "Motion Sensor" and Zone Status to 0x0001 when motion is detected, maybe this can help.

Yep, It mean it will be easier ^^, It seem I don't need to use the tuya cluster, but the IS can be enought.
BTW the HS1MS-E have the same reaction ? Because all seem fine in the code, a problem in enrollement ?

Yes, you need a deconz version on linux. You can stop the plugin (test just take 10mn) or you can configure HA to use a distant deconz installation (But don'i ask me how).
If you just turn off the addon for some minuts, you will not have problem, Xiaomi need more time before leaving.

SilverWx, 2020-09-15:
thank you for the quick return, I will check your modification as soon I can (this week for sure).
It could be great to find the battery information too (I have also konke sensor that doesn't report battery so I will check that).

Thank for solving the heiman sensor problem :D

SilverWx, 2020-09-15:
I got Project ERROR: Unknow module(s) in QT: websocket

i have installed qt5-qmake and qt5-default

Smanar, 2020-09-16:
Have you tried to install the dev package ?

`sudo apt install deconz-dev`

If you have an error here, you can stop direclty.

SilverWx, 2020-09-17:
It worked ! Thank you

heiman motion worked great, appeared in phoscon app, same for konke but konke doesn't seem to report battery info (do not respond when I ask read cluster for power).
Tuya TS0202 motion detection is based on IAS Zone Status Change (0x0000 for nothing, 0x0001 for presence) bot does not respond to read for power cluster.

So, here is the zcl debug log :

0x0C43 is TS0202 (recognized as TS0202)

'01:39:54:621 ZCL got data for node=0x0C43, cl=0x0500, at=0x0011, status=0x00, type=0x20
01:39:54:628 ZCL got unsupported status: 0x86 for mandatory attribute
01:39:54:628 ZCL got data for node=0x0C43, cl=0x0500, at=0x8000, status=0x86, type=0x00
01:39:54:722 ZCL got data for node=0x0C43, cl=0x0500, at=0x0000, status=0x00, type=0x30
01:39:54:722 ZCL got data for node=0x0C43, cl=0x0500, at=0x0001, status=0x00, type=0x31
01:39:54:722 ZCL got data for node=0x0C43, cl=0x0500, at=0x0002, status=0x00, type=0x19
01:39:54:722 ZCL got data for node=0x0C43, cl=0x0500, at=0x0010, status=0x00, type=0xF0
01:42:59:494 ZCL cmd-req nwk: 0x0C43, profile: 0x0104, cluster: 0x0500 cmd: 0x00
01:42:59:494 ZCL read cluster: 0x0500, attribute: 0x0011
01:42:59:494 ZCL cmd-req nwk: 0x0C43, profile: 0x0104, cluster: 0x0500 cmd: 0x00
01:42:59:494 ZCL read cluster: 0x0500, attribute: 0x8000
01:42:59:494 ZCL cmd-req nwk: 0x0C43, profile: 0x0104, cluster: 0x0500 cmd: 0x00
01:42:59:494 ZCL read cluster: 0x0500, attribute: 0x8001
01:42:59:494 ZCL cmd-req nwk: 0x0C43, profile: 0x0104, cluster: 0x0500 cmd: 0x00
01:42:59:494 ZCL read cluster: 0x0500, attribute: 0xFFF0
01:42:59:494 ZCL cmd-req nwk: 0x0C43, profile: 0x0104, cluster: 0x0500 cmd: 0x00
01:42:59:494 ZCL read cluster: 0x0500, attribute: 0xFFF1
01:42:59:494 ZCL cmd-req nwk: 0x0C43, profile: 0x0104, cluster: 0x0500 cmd: 0x00
01:43:08:196 IAS Zone Status Change, status: 0x0001, zoneId: 255, delay: 0
01:45:08:169 IAS Zone Status Change, status: 0x0001, zoneId: 255, delay: 0
01:45:09:297 ZCL got data for node=0x0C43, cl=0x0500, at=0x0000, status=0x00, type=0x30
01:45:09:297 ZCL got data for node=0x0C43, cl=0x0500, at=0x0001, status=0x00, type=0x31
01:45:09:297 ZCL got data for node=0x0C43, cl=0x0500, at=0x0002, status=0x00, type=0x19
01:45:09:297 ZCL got data for node=0x0C43, cl=0x0500, at=0x0010, status=0x00, type=0xF0
01:45:09:304 ZCL got data for node=0x0C43, cl=0x0500, at=0x0011, status=0x00, type=0x20
01:45:09:327 ZCL got unsupported status: 0x86 for mandatory attribute
01:49:18:941 IAS Zone Status Change, status: 0x0000, zoneId: 255, delay: 0
01:52:16:154 ZCL cmd-req nwk: 0x0C3A, profile: 0x0104, cluster: 0x0500 cmd: 0x00
01:52:16:155 ZCL read cluster: 0x0500, attribute: 0x0011
01:52:16:155 ZCL cmd-req nwk: 0x0C3A, profile: 0x0104, cluster: 0x0500 cmd: 0x00
01:52:16:155 ZCL read cluster: 0x0500, attribute: 0x8000
01:52:16:155 ZCL cmd-req nwk: 0x0C3A, profile: 0x0104, cluster: 0x0500 cmd: 0x00
01:52:16:155 ZCL read cluster: 0x0500, attribute: 0x8001
01:52:16:155 ZCL cmd-req nwk: 0x0C3A, profile: 0x0104, cluster: 0x0500 cmd: 0x00
01:52:16:155 ZCL read cluster: 0x0500, attribute: 0xFFF0
01:52:16:155 ZCL cmd-req nwk: 0x0C3A, profile: 0x0104, cluster: 0x0500 cmd: 0x00
01:52:16:155 ZCL read cluster: 0x0500, attribute: 0xFFF1
01:52:16:155 ZCL cmd-req nwk: 0x0C3A, profile: 0x0104, cluster: 0x0500 cmd: 0x00
01:52:17:272 ZCL got data for node=0x0C3A, cl=0x0500, at=0x0000, status=0x00, type=0x30
01:52:17:272 ZCL got data for node=0x0C3A, cl=0x0500, at=0x0001, status=0x00, type=0x31
01:52:17:273 ZCL got data for node=0x0C3A, cl=0x0500, at=0x0002, status=0x00, type=0x19
01:52:17:273 ZCL got data for node=0x0C3A, cl=0x0500, at=0x0010, status=0x00, type=0xF0
01:52:17:279 ZCL got data for node=0x0C3A, cl=0x0500, at=0x0011, status=0x00, type=0x20
01:52:17:999 IAS Zone Status Change, status: 0x0021, zoneId: 100, delay: 0
01:52:18:262 ZCL got unsupported status: 0x86 for mandatory attribute
**01:52:18:262 ZCL got data for node=0x0C3A, cl=0x0500, at=0x8000, status=0x86, type=0x00**
01:55:02:699 ZCL read cluster: 0x0001, attribute: 0x0000
01:55:02:699 ZCL read cluster: 0x0001, attribute: 0x0001
01:55:02:699 ZCL read cluster: 0x0001, attribute: 0x0010
01:55:02:699 ZCL read cluster: 0x0001, attribute: 0x0011
01:55:02:699 ZCL cmd-req nwk: 0x0C43, profile: 0x0104, cluster: 0x0001 cmd: 0x00
01:55:02:699 ZCL read cluster: 0x0001, attribute: 0x0012
01:55:02:699 ZCL read cluster: 0x0001, attribute: 0x0013
01:55:02:699 ZCL read cluster: 0x0001, attribute: 0x0020
01:55:02:699 ZCL read cluster: 0x0001, attribute: 0x0021
01:55:02:699 ZCL cmd-req nwk: 0x0C43, profile: 0x0104, cluster: 0x0001 cmd: 0x00
01:55:02:699 ZCL read cluster: 0x0001, attribute: 0x0030
01:55:02:699 ZCL cmd-req nwk: 0x0C43, profile: 0x0104, cluster: 0x0001 cmd: 0x00
01:55:02:699 ZCL read cluster: 0x0001, attribute: 0x0031
01:55:02:699 ZCL read cluster: 0x0001, attribute: 0x0032
01:55:02:699 ZCL read cluster: 0x0001, attribute: 0x0033
01:55:02:699 ZCL read cluster: 0x0001, attribute: 0x0034
01:55:02:699 ZCL cmd-req nwk: 0x0C43, profile: 0x0104, cluster: 0x0001 cmd: 0x00
01:55:02:699 ZCL read cluster: 0x0001, attribute: 0x0035
01:55:02:699 ZCL read cluster: 0x0001, attribute: 0x0036
01:55:02:699 ZCL read cluster: 0x0001, attribute: 0x003E

'

SilverWx, 2020-09-17:
Since I have tested your code, Heiman sensor started working in my hassio too, how it's possible ?

Smanar, 2020-09-17:
What is your konke device ?

You have used a database copy ? on the previous code the heiman was not in the supported list, but one time the device is included, it was not checked a second time. Else, maybe planet alignement ^^.

The detection is working ?
BTW how have you enabling this log ? It s the result with debug=2 ?

You are saying the power confguration cluster is not working (no usefull value in deconz) ?

For the attribute 0x8000 it can use the same than for develco
```
				<attribute-set id="0x8000" description="Develco specific" mfcode="0x1015">
					<attribute id="0x8000" name="Zone Status Interval" type="u16" access="rw" required="o" mfcode="0x1015"></attribute>
					<attribute id="0x8001" name="Alarm Off Delay" type="u16" access="rw" required="o" mfcode="0x1015"></attribute>
				</attribute-set>

```

status = 0x86 mean ERROR.

And I don't know at all thoses one
```
01:52:17:273 ZCL got data for node=0x0C3A, cl=0x0500, at=0x0010, status=0x00, type=0xF0
01:52:17:279 ZCL got data for node=0x0C3A, cl=0x0500, at=0x0011, status=0x00, type=0x20
```

I will make search for them, can you just check power cluster to be sure the device don't use it, pls.

SilverWx, 2020-09-19:
It's konke pro motion sensor, this one https://fr.banggood.com/KONKE-Zigbee-3_0-Open-Protocol-Smart-Home-PIR-Human-Body-Sensor-Wireless-Motion-Sensor-Switch-Support-Google-Assistant-Amazon-Alexa-Siri-p-1551514.html?cur_warehouse=GWTR

I use --dbg-zcl=1 to get only zigbee cluster info, as I found info debug too noisy.

Reading power configuration of konke using read button inside deCONZ does not let to get data about battery, exept for one value (battery voltage) : 
[image]

I didn't copy database so i don't understand why my heimann started working well after testing your code, But I don't understand everything in deCONZ and Zigbee stack so :P 

The deCONZ app runing your code started discovery all the nerwork without requiring me to pair my device again, so I guess the pairing is only based on MAC or something like that x)

Okay, thanks for finding info on this attribute, Zone Status Interval, but what does it mean ? Can we change the second between two position check ? 

Thank you again for your help :D

Smanar, 2020-09-19:
I assume the konke is this one  > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1607 ?
I have changed the code to ue the attribute 0x0020 instead of the 0x0021.

And same for the tuya, as you haven't answered, I assume it s the same problem ?

You can try the new code

```
     cd deconz-rest-plugin
     git pull
     qmake && make -j2
     sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

You need perhaps a new inclusion to configure binding, but not sure.

SilverWx, 2020-09-20:
Yes, konke is this one.

Battery is correctly reported for heimann, not for konke and  TS0202.

I will try the new code tinight ;)

SilverWx, 2020-09-21:
Using deCONZ addon in hassio I got that yersterday
[image]
[image]

I have not tested your new code yet, I will do today after my work

SilverWx, 2020-09-21:
So, I have finally tested the last version of your code :+1: 

Both Heimann and konke are recognized as motion sensor and appears in Phoscon, Heimann battery is ok, I'm not sure for konke since is reporting low battery but the battery inside is pretty new.

TS0202 reports to message on motion, not one as reported before
'
TS 0202 Zone ID on motion
01:42:34:223 IAS Zone Status Change, status: 0x0001, zoneId: 100, delay: 0
01:42:35:226 IAS Zone Status Change, status: 0x0001, zoneId: 255, delay: 0
'
Log: 
'
TS0202 Battery Cluster
01:33:51:331 ZCL read cluster: 0x0001, attribute: 0x0000
01:33:51:331 ZCL read cluster: 0x0001, attribute: 0x0001
01:33:51:331 ZCL read cluster: 0x0001, attribute: 0x0010
01:33:51:331 ZCL read cluster: 0x0001, attribute: 0x0011
01:33:51:331 ZCL cmd-req nwk: 0x0C43, profile: 0x0104, cluster: 0x0001 cmd: 0x00
01:33:51:331 ZCL read cluster: 0x0001, attribute: 0x0012
01:33:51:331 ZCL read cluster: 0x0001, attribute: 0x0013
01:33:51:331 ZCL read cluster: 0x0001, attribute: 0x0020
01:33:51:331 ZCL read cluster: 0x0001, attribute: 0x0021
01:33:51:331 ZCL cmd-req nwk: 0x0C43, profile: 0x0104, cluster: 0x0001 cmd: 0x00
01:33:51:331 ZCL read cluster: 0x0001, attribute: 0x0030
01:33:51:331 ZCL cmd-req nwk: 0x0C43, profile: 0x0104, cluster: 0x0001 cmd: 0x00
01:33:51:331 ZCL read cluster: 0x0001, attribute: 0x0031
01:33:51:331 ZCL read cluster: 0x0001, attribute: 0x0032
01:33:51:331 ZCL read cluster: 0x0001, attribute: 0x0033
01:33:51:331 ZCL read cluster: 0x0001, attribute: 0x0034
01:33:51:331 ZCL cmd-req nwk: 0x0C43, profile: 0x0104, cluster: 0x0001 cmd: 0x00
01:33:51:331 ZCL read cluster: 0x0001, attribute: 0x0035
01:33:51:331 ZCL read cluster: 0x0001, attribute: 0x0036
01:33:51:331 ZCL read cluster: 0x0001, attribute: 0x003E
01:33:51:331 ZCL cmd-req nwk: 0x0C43, profile: 0x0104, cluster: 0x0001 cmd: 0x00
01:33:51:789 ZCL got unsupported status: 0x86 for mandatory attribute
01:33:51:790 ZCL got data for node=0x0C43, cl=0x0001, at=0x0000, status=0x86, type=0x00
01:33:51:790 ZCL got unsupported status: 0x86 for mandatory attribute
01:33:51:790 ZCL got data for node=0x0C43, cl=0x0001, at=0x0001, status=0x86, type=0x00
01:33:51:790 ZCL got unsupported status: 0x86 for mandatory attribute
01:33:51:790 ZCL got data for node=0x0C43, cl=0x0001, at=0x0010, status=0x86, type=0x00
01:33:51:790 ZCL got unsupported status: 0x86 for mandatory attribute
01:33:51:790 ZCL got data for node=0x0C43, cl=0x0001, at=0x0011, status=0x86, type=0x00
01:33:52:125 ZCL got data for node=0xA9E5, cl=0x0000, at=0x0006, status=0x00, type=0x42
01:33:52:841 ZCL got data for node=0x398D, cl=0x0006, at=0x0000, status=0x00, type=0x10
01:33:53:854 ZCL got unsupported status: 0x86 for mandatory attribute
01:33:53:854 ZCL got data for node=0x0C43, cl=0x0001, at=0x0012, status=0x86, type=0x00
01:33:53:854 ZCL got unsupported status: 0x86 for mandatory attribute
01:33:53:854 ZCL got data for node=0x0C43, cl=0x0001, at=0x0013, status=0x86, type=0x00
01:33:53:854 ZCL got data for node=0x0C43, cl=0x0001, at=0x0020, status=0x00, type=0x20
01:33:53:854 ZCL got data for node=0x0C43, cl=0x0001, at=0x0021, status=0x00, type=0x20
01:33:53:867 ZCL got unsupported status: 0x86 for mandatory attribute
01:33:53:867 ZCL got data for node=0x0C43, cl=0x0001, at=0x0030, status=0x86, type=0x00
01:33:53:874 ZCL got unsupported status: 0x86 for mandatory attribute
01:33:53:874 ZCL got data for node=0x0C43, cl=0x0001, at=0x0031, status=0x86, type=0x00
01:33:53:874 ZCL got unsupported status: 0x86 for mandatory attribute
01:33:53:874 ZCL got data for node=0x0C43, cl=0x0001, at=0x0032, status=0x86, type=0x00
01:33:53:874 ZCL got unsupported status: 0x86 for mandatory attribute
01:33:53:874 ZCL got data for node=0x0C43, cl=0x0001, at=0x0033, status=0x86, type=0x00
01:33:53:874 ZCL got unsupported status: 0x86 for mandatory attribute
01:33:53:874 ZCL got data for node=0x0C43, cl=0x0001, at=0x0034, status=0x86, type=0x00

Heimann Battery Cluster
01:36:20:843 ZCL read cluster: 0x0001, attribute: 0x0020
01:36:20:843 ZCL read cluster: 0x0001, attribute: 0x0021
01:36:20:843 ZCL cmd-req nwk: 0x0C43, profile: 0x0104, cluster: 0x0001 cmd: 0x00
01:36:20:844 ZCL read cluster: 0x0001, attribute: 0x0035
01:36:20:844 ZCL read cluster: 0x0001, attribute: 0x0036
01:36:20:844 ZCL read cluster: 0x0001, attribute: 0x003E
01:36:20:844 ZCL cmd-req nwk: 0x0C43, profile: 0x0104, cluster: 0x0001 cmd: 0x00
01:36:20:973 ZCL got data for node=0x0C43, cl=0x0001, at=0x0020, status=0x00, type=0x20
[... 37 lines cut ...]
01:38:45:002 ZCL got unsupported status: 0x86 for mandatory attribute
01:38:45:002 ZCL got data for node=0x9AF2, cl=0x0001, at=0x0012, status=0x86, type=0x00
01:38:45:002 ZCL got unsupported status: 0x86 for mandatory attribute
01:38:45:002 ZCL got data for node=0x9AF2, cl=0x0001, at=0x0013, status=0x86, type=0x00
01:38:45:002 ZCL got data for node=0x9AF2, cl=0x0001, at=0x0020, status=0x00, type=0x20
01:38:45:002 ZCL got unsupported status: 0x86 for mandatory attribute
01:38:45:003 ZCL got data for node=0x9AF2, cl=0x0001, at=0x0021, status=0x86, type=0x20
01:38:46:321 ZCL got data for node=0xA9E5, cl=0x0006, at=0x0000, status=0x00, type=0x10
01:38:46:601 ZCL got data for node=0xA9E5, cl=0x0000, at=0x0006, status=0x00, type=0x42
01:38:46:958 ZCL got unsupported status: 0x86 for mandatory attribute
01:38:46:958 ZCL got data for node=0x9AF2, cl=0x0001, at=0x0030, status=0x86, type=0x00
[... cut ...]
01:38:46:965 ZCL got unsupported status: 0x86 for mandatory attribute
01:38:46:965 ZCL got data for node=0x9AF2, cl=0x0001, at=0x0032, status=0x86, type=0x30
01:38:46:965 ZCL got unsupported status: 0x86 for mandatory attribute
01:38:46:965 ZCL got data for node=0x9AF2, cl=0x0001, at=0x0033, status=0x86, type=0x30
01:38:46:965 ZCL got unsupported status: 0x86 for mandatory attribute
01:38:46:966 ZCL got data for node=0x9AF2, cl=0x0001, at=0x0034, status=0x86, type=0x30
'

Phoscon
[image]
[image]

I hope this can help you :D

Smanar, 2020-09-22:
It seem fine both are supporting the attribute 0x0020

>01:33:53:854 ZCL got data for node=0x0C43, cl=0x0001, at=0x0020, status=0x00, type=0x20
>01:38:45:002 ZCL got data for node=0x9AF2, cl=0x0001, at=0x0020, status=0x00, type=0x20

On the Json you will have the value in percent.

```
TS0202 reports to message on motion, not one as reported before
TS 0202 Zone ID on motion
01:42:34:223 IAS Zone Status Change, status: 0x0001, zoneId: 100, delay: 0
01:42:35:226 IAS Zone Status Change, status: 0x0001, zoneId: 255, delay: 0
```
I don't understand, It don't work ? You have the Device Json too ?

SilverWx, 2020-09-23:
Perfect :D

No, I have not using Rest API myself to get Json, I will try as soon I can.

TS0202 doesn't appear in phoscon, but maybe I have to pair it again to see it.

Mimiix, 2020-09-23:
@SilverWx Phoscon need to add support too.

SilverWx, 2020-09-23:
Ok, so do you need me to do some extra test or it's ok ?

Smanar, 2020-09-23:
So all 3 devices are working ? I can make the PR ?

SilverWx, 2020-09-23:
I think so, thank you !

Here are the json from the REST API

**TS0202**
`  "5": {
    "config": {
      "battery": 100,
      "duration": 60,
      "on": true,
      "pending": [],
      "reachable": false
    },
    "ep": 1,
    "etag": "b2d1224b10825e0317676558070a8e12",
    "lastseen": null,
    "manufacturername": "Tuya",
    "modelid": "TS0202",
    "name": "TS0202",
    "state": {
      "lastupdated": "2020-09-21T23:42:35.226",
      "lowbattery": false,
      "presence": false,
      "tampered": false
    },
    "type": "ZHAPresence",
    "uniqueid": "68:0a:e2:ff:fe:71:b8:27-01-0500"
  },`

**Heiman**
`  "6": {
    "config": {
      "battery": 77,
      "duration": 60,
      "on": true,
      "pending": [],
      "reachable": false
    },
    "ep": 1,
    "etag": "b2d1224b10825e0317676558070a8e12",
    "lastseen": null,
    "manufacturername": "Heiman",
    "modelid": "PIRSensor-EM",
    "name": "Motion Sensor",
    "state": {
      "lastupdated": "2020-09-21T23:43:50.695",
      "lowbattery": false,
      "presence": false,
      "tampered": false
    },
    "swversion": "1.1.1",
    "type": "ZHAPresence",
    "uniqueid": "00:0d:6f:00:14:bb:90:64-01-0500"
  },`

**Konke**
`    "config": {
      "battery": null,
      "duration": 60,
      "on": true,
      "pending": [],
      "reachable": false
    },
    "ep": 1,
    "etag": "b2d1224b10825e0317676558070a8e12",
    "lastseen": null,
    "manufacturername": "Konke",
    "modelid": "3AFE28010402000D",
    "name": "Motion Sensor (2)",
    "state": {
      "lastupdated": "2020-09-21T23:42:34.223",
      "lowbattery": false,
      "presence": false,
      "tampered": false
    },
    "swversion": "190829",
    "type": "ZHAPresence",
    "uniqueid": "14:b4:57:ff:fe:07:63:01-01-0500"
  }`

Smanar, 2020-09-24:
Ok I have find another problem for the konke device > https://github.com/Smanar/deconz-rest-plugin/commit/45b8b8823dc06d73b4feba5724bcd370d38f77b4
I think it will help for battery

SilverWx, 2020-09-24:
I have build deCONZ using your last commit but I still get no battery information, it's strange since deCONZ Power Configuration Cluster report the battery voltage.

If you can't it's not a big deal, but I think this sensor will be widely used as it cheaper than aqara and sponsored by xiaomi too.

'  "8": {
    "config": {
      "battery": null,
      "duration": 60,
      "on": true,
      "pending": [],
      "reachable": true
    },
    "ep": 1,
    "etag": "521b0cce1a6bebaeaec0a5fc933bcfef",
    "lastseen": "2020-09-24T21:00Z",
    "manufacturername": "Konke",
    "modelid": "3AFE28010402000D",
    "name": "Motion Sensor (2)",
    "state": {
      "lastupdated": "2020-09-24T20:59:59.276",
      "lowbattery": false,
      "presence": true,
      "tampered": false
    },
    "swversion": "190829",
    "type": "ZHAPresence",
    "uniqueid": "14:b4:57:ff:fe:07:63:01-01-0500"
  }'

Smanar, 2020-09-25:
I m sorry, but I have again some missing code > https://github.com/Smanar/deconz-rest-plugin/commit/5b3411e1e70c6f762e175e7cbf544a66f9d5f247

Have just see it.

SilverWx, 2020-09-26:
Hey !
Unfortunately it doesn't seems to fix it, now I have battery indicated to be *almost empty* which I think is not possible because I have tested it with two brand new battery.

`  "7": {
    "config": {
      "battery": null,
      "duration": 60,
      "on": true,
      "pending": [],
      "reachable": true
    },
    "ep": 1,
    "etag": "5fbb983fbad5e9db674cde72d354efd0",
    "lastseen": "2020-09-26T16:41Z",
    "manufacturername": "Konke",
    "modelid": "3AFE28010402000D",
    "name": "Motion Sensor (2)",
    "state": {
      "lastupdated": "2020-09-26T16:41:40.938",
      "lowbattery": true,
      "presence": true,
      "tampered": true
    },
    "swversion": "190829",
    "type": "ZHAPresence",
    "uniqueid": "14:b4:57:ff:fe:07:63:01-01-0500"
  }`

Smanar, 2020-09-26:
But this time, I don't find more problem in the code ^^. No more idea.
Perhaps it can take same time ?

SilverWx, 2020-09-26:
What do you mean by Perhaps it can take same time ? :p

Smanar, 2020-09-26:
Lol, on the code the setting is 6 mn, but "hope brings life".

No seriously, I don't have more idea, I make a last check tomorrox and make the PR with the missing battery.

SilverWx, 2020-09-27:
ahah :P

I got this for battery voltage :D
```
            rq.attributeId = 0x0020;
            rq.minInterval = 3600;
            rq.maxInterval = 7200;
            rq.reportableChange8bit = 0;
```

Smanar, 2020-09-28:
Hu ?
I have checked the code those value are not from deconz. Deconz never use 3600 and 7200 for the same device.

have you tried a new device inclusion ?

SilverWx, 2020-09-28:
I have pair it again and I get those value :+1: 

I got this for battery voltage :D
```
            rq.attributeId = 0x0020;
            rq.minInterval = 0;
            rq.maxInterval = 7200;
            rq.reportableChange8bit = 0;
```

Smanar, 2020-09-29:
It s like the binding/attribute setting is not working ...

Perhaps something visible on logs ?

SilverWx, 2020-09-29:
I will check that :)

Smanar, 2020-10-02:
BTW I m making the PR now, I have missed that last one ^^.

Can add the battery later.

SilverWx, 2020-10-04:
OK, thank you ! I will check if can get any info on battery stat chen I get some time :D

closed 2020-10-05 by manup

Mtpaulsen, 2020-11-23:
Hi everyone,

Thanks for all the work, on basically everything... I often buy an item, and before having to submit request for support for it here already :-)
And I must admit, that I like to shop around, to test different versions, and find the one I fancy the most...
Have just received on of these:
https://www.aliexpress.com/item/4000829510396.html?spm=a2g0s.9042311.0.0.5a056c37KXsW1Q

And I'm super happy with the design, the choice to add a battery source with a bit more juice, and the overall design with the clever magnetic mount, where you easily can add/remove the PIR as well as adjust and point it to different direction.

However, I don't seem to be able to connect it... My preferred setup is throught a deConz Conbee 2 usb stick, and then access it either through Homey or Home Assistant, but I can't even locate it in Phoscon to Conbee... Has anyone success with this??

It abvious connects find to a generic Tuya hub, I have for test purposes, and I also managed to add it to a Samsung Smartthings v3 Hub.... But would prefer to collect all ZigBee nodes to Conbee, so any helpful proposals are very welcome? :-)

Best regards.

Smanar, 2020-11-23:
It s realy hard to reconize a tuya device, they can be exactly same, just with a different MAC adress base.
To be sure the only way is checking the values visible in the GUI and compare them.