dresden-elektronik/deconz-rest-plugin #3240: Centralite Pearl Support

cbrightly, 2020-09-06:
## Device

- Product name: Centralite Pearl Zigbee Thermostat
- Manufacturer: Centralite
- Model identifier: 3157100
- Device type:
  - Thermostat

## Screenshots

Please see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2630#issuecomment-687902181 for an identical description of the issue.

> I was able to successfully add a Centralite 3157100 to my conbee I stick with 2.05.78 and do see the thermostat temperature sensor in the API and home assistant. Thanks for getting this working in #2630 However currently it seems to only expose the current temperature and not any of the other information such as temperature set point, fan setting, cooling/heating mode. If the AC/heater is actually currently running. It also doesn't appear to be settable through the API, or at least it doesn't appear in the lights section of the API. The thermostat control in home assistant therefore also doesn't appear to be able to set the thermostat. I was able to set the temperature set point directly in deconz gui and read the information from the respective zigbee clusters. Is it easy to expose these other information through the API as well? I would like to write some automations to control the A/C through a temperature sensor in a different room instead of the room the thermostat is currently physically located.

[image]
[image]
[image]

Mimiix, 2020-09-07:
Hi @cbrightly Information seems to be missing. Cluster information in this case. Could you please add that?

cbrightly, 2020-09-07:
Of course! Apologies for neglecting to add that!
[image]

Mimiix, 2020-09-08:
@cbrightly Yes, but also the clusters being read 😄

cbrightly, 2020-09-08:
Thanks for your help and your patience! I’m unsure specifically what you are asking for though; can you point me to how or where to gather the information you need?

Mimiix, 2020-09-08:
No worries :)

You did a read of the basic cluster already, but you need to do that with all clusters and provide screens of those.

cbrightly, 2020-09-08:
Here are screenshots of the readable attributes. If you need the rest of the cluster information panels (exec functions, etc.) please let me know and I can repost them.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Mimiix, 2020-09-08:
@SwoopX I think you can knock yourselve out with this one

SwoopX, 2020-09-08:
I can at least have a look later. There's been lots of work and improvements on the thermostat functionality throughout the last 2 versions...

SwoopX, 2020-09-08:
Could you please provide me the output of the REST API from `/sensors` and `/lights`. Based on the node name in deconz GUI, I doubt that everything's set up correctly or even anything at all.

cbrightly, 2020-09-08:
Here are the results of the API calls. I will grab those screenshots and post them a bit later. Thanks for your assistance!

`/sensors`
>{
  "25" : {
    "config" : {
      "battery" : 31,
      "reachable" : true,
      "on" : true
    },
    "manufacturername" : "Samjin",
    "state" : {
      "lastupdated" : "2020-09-08T19:43:40.315",
      "vibration" : false,
      "orientation" : [
        -36,
        -1054,
        11
      ]
    },
    "uniqueid" : "28:6d:97:00:01:0c:81:f3-01-fc02",
    "ep" : 1,
    "lastseen" : "2020-09-08T19:43Z",
    "modelid" : "multi",
    "type" : "ZHAVibration",
    "name" : "Back Door | Multi 2",
    "etag" : "41037b6a17d24f1e9b523dda627e356f"
  },
  "10" : {
    "config" : {
      "battery" : 100,
      "reachable" : false,
      "on" : true
    },
    "manufacturername" : "LUMI",
    "state" : {
      "lastupdated" : "2020-07-17T08:44:04.925",
      "pressure" : 1009
    },
    "swversion" : "20161129",
    "uniqueid" : "00:15:8d:00:04:11:d2:fa-01-0403",
    "ep" : 1,
    "lastseen" : null,
    "modelid" : "lumi.weather",
    "type" : "ZHAPressure",
    "name" : "Leather Traveldor",
    "etag" : "18936d258057fe04fd9a65644c5a4ec1"
  },
  "26" : {
    "config" : {
      "battery" : 100,
      "reachable" : true,
      "on" : true
    },
    "manufacturername" : "Samjin",
    "state" : {
      "lastupdated" : "2020-09-08T19:15:40.591",
      "open" : false
    },
    "uniqueid" : "28:6d:97:00:01:0b:df:9b-01-0500",
    "ep" : 1,
    "lastseen" : "2020-09-08T19:15Z",
    "modelid" : "multi",
    "type" : "ZHAOpenClose",
    "name" : "Safe Door | Multi 1",
    "etag" : "3f09864c1e95b2c4ffee608dca4277fc"
  },
  "27" : {
    "config" : {
      "reachable" : true,
[... 434 lines cut ...]
    "lastseen" : "2020-09-08T19:43Z",
    "modelid" : "LWB014",
    "type" : "Dimmable light",
    "name" : "Dad’s Nightstand Lamp",
    "etag" : "1e8ee7611a09ddfcbbfa331a966f1b6e"
  },
  "5" : {
    "manufacturername" : "Philips",
    "hascolor" : false,
    "lastannounced" : "2020-09-06T10:21:15Z",
    "state" : {
      "bri" : 254,
      "reachable" : false,
      "alert" : "none",
      "on" : false
    },
    "swversion" : "1.23.0_r20156",
    "uniqueid" : "00:17:88:01:03:4c:bd:2d-0b",
    "lastseen" : "2020-09-07T01:25Z",
    "modelid" : "LWB014",
    "type" : "Dimmable light",
    "name" : "Mom’s Nightstand Lamp",
    "etag" : "dd6662730b23e53025595171663e0187"
  }
}

SwoopX, 2020-09-10:
I'm afraid the list of sensors and lights does not correspond to your description of the issue. You don't have anything related to the device in the lists. From that perspective, I'd expect the device not to be paired correctly and I also wonder how you can determine if anything is missing when nothing is available.

cbrightly, 2020-09-12:
It appears the device is showing in the GUI but not the API? What could cause this?

ebaauw, 2020-09-12:
See _deCONZ for Dummies_.

Typically all ZHA, ZLL, and Zigbee 3.0 devices devices appear in the GUI when they’ve joined the Zigbee network.  The core also supports ZGP devices, but they’re not shown.

The API creates resources for supported device endpoints and clusters.  The device needs to match the conditions for the resource to be created.  For this, the gateway needs to read the device configuration.  This takes multiple requests.  When one of those requests goes unanswered, the process stalls.  Unfortunately, this is not uncommon for battery-powered devices.  Try manually reading the descriptors and the _Basic_ cluster attributes, while the API is searching for new devices, and while forcing the device to be awake.

cbrightly, 2020-09-12:
After trying that, curiously, it now shows up under sensors in the API after adding it as such through the web interface, but still does not show via the web interface at all. Here is the new addition to the /sensors output:

  "19" : {
    "config" : {
      "offset" : 0,
      "schedule_on" : null,
      "heatsetpoint" : 2000,
      "on" : true,
      "battery" : null,
      "reachable" : true,
      "schedule" : {

      }
    },
    "manufacturername" : "Centralite",
    "state" : {
      "lastupdated" : "2020-09-12T09:48:30.849",
      "temperature" : 2958,
      "on" : null
    },
    "uniqueid" : "00:0d:6f:00:0b:4d:0c:75-01-0201",
    "ep" : 1,
    "lastseen" : "2020-09-12T09:50Z",
    "modelid" : "3157100",
    "type" : "ZHAThermostat",
    "name" : "3157100",
    "etag" : "c6e9828351a4382cb08588192938fefc"
  }

I hope this is of help! Thanks again for your help and time!

ebaauw, 2020-09-12:
> but still does not show via the web interface at all.

It needs to be supported by Phoscon before it shows there.  Best open an issue in the Phoson repo, asking for support.

cbrightly, 2020-09-12:
Thanks for the tip; are you referring to the beta app repo?

SwoopX, 2020-09-13:
@cbrightly  Alright, this is looking better now. However, you'd still need to go through the attributes as mentioned here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3240#issuecomment-689078201 to gain a better understanding on what the device capabilities are.

Also, can you please verify if the device can handle on-device schedules? Just use the commands offered above the attributes to play with.

cbrightly, 2020-09-15:
The unit doesn't appear to support on-device scheduling. I have attached the requested screenshots; thanks again! I do use HA, but have deConz running on a separate Raspberry Pi, if that is of assistance.[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2020-09-17:
@cbrightly Thanks for following up!

> The unit doesn't appear to support on-device scheduling.

Have you tried setting some? Z2m indicates they might be able too support it.

So now, you need to help me with any missing features. I understood that you want/need the fan control cluster and settings to be exposed. Which other features (that means attributes) from the thermostat cluster do you want/need/does the device support? You can all try and set then via GUI.

stevewull, 2020-10-08:
Jumping into the middle of your discussion. I can start a separate issue if you would like...  I have the same thermostat and was working with SwoopX (issue #2630) before events forced me to abruptly put my project on hold.

I recently installed deConz version 2.05.81. The GUI detects my Pearl Thermostat. My cluster info looks the same as the screenshots posted by cbrightly. The device shows up in the API. Home Assistant's deConz integration detects it as a thermostat and can display the temperature. You made great progress!

Reviewing the Pearl's manual and picking up on SwoopX's last question about missing features. Starting with controlling using the deConz gui and hoping all controls will eventually be available in Home Assistant.

1. The Pearl has four modes. Pressing MODE button cycles between them.  GUI should be able turn on and off each mode
a. HEAT
b. COOL
c. Emergency Heat (if 2nd-stage heat is available) not a priority for me
d. OFF

2. Independently, a FAN button toggles between two modes. GUI should be able to switch between them.
a. Fan On Mode
b. Fan Auto Mode

3. The Setpoint temperature is controlled by a slider to increase and decrease. The GUI should be able to change the Setpoint temperature

4. There is no evidence in the documentation or the user interface that the Pearl supports on-device scheduling. Setting schedules in the GUI (and Home Assistant) to control the Pearl is a high priority.

5. A HOLD button enables a hold function that locks out all scheduled system changes, both internal schedules and external schedules from a controller. Turning the hold on and off from the GUI would be good to have.

6. Pearl has various menu functions and options. The four that would be nice to control from the GUI are below. This is a low priority
a. Minimum Heatpoint
b. Maximum Heatpoint
c. Minimum Coolpoint
d. Maximum Coolpoint

stevewull, 2020-10-08:
Note: I tested the Setpoint Raise/Lower function in the Thermostat cluster.  This works!

stevewull, 2020-10-08:
Setpoint raise/lower also works from Home Assistant

stevewull, 2020-10-08:
I experimented with a few of the read/write attributes and **succeeded** in controlling the Thermostat for the following

- Fan Mode:  Set to Auto or On
- System Mode:  Set to OFF, COOL, HEAT, Emergency Heating, fan only
- Temperature setpoint hold:  Turn on and off
- Occupied Cooling setpoint: can write a specific temperature
- Occupied Heating Setpoint: can write a specific temperature
- Presumably min/max heat and cool setpoint limits also work

- The setpoint raise/lower command works for both the Heat and Cool setpoints

Overall, it looks like control of the Thermostat using the cluster attributes works

cbrightly, 2020-10-09:
@stevewull  I'm glad that some progress has been made! After a bit of working with it I decided that as a 'smart' thermostat the Centralite Pearl wasn't particularly intelligent and have since returned it. Best of luck with getting the integration tightened up!

@SwoopX et. al. Thank you for your assistance!

SwoopX, 2020-10-09:
@stevewull Many thanksQ That's exactly what we need here and valuable input. The changes should make it to the next beta release the coming week.

stevewull, 2020-10-09:
@SwoopX . Thank you.

Note also:  There are two setpoints.  HEATING and COOLING.

SwoopX, 2020-10-09:
Got that, working on it anyway ;)

SwoopX, 2020-10-24:
Just a quick update: Haven't forgotten about that, but I need to wait for other PRs to be merged first before I can bring this one in.

stevewull, 2020-10-26:
Thanks for the update.

apmon, 2020-11-04:
Thanks for all of the hard work, and I am looking forward to seeing those changes as I have now finally gotten around to installing my centralite pearl thermostat that I had lying around for a while.
I have it show up in the rest api and home assistant sees it (as far as it is currently implemented). However, one thing, that I am not sure has been mentioned yet (sorry, if I missed it in this long thread), is that I don't seem to get it to automatically update if things change on the thermostat side (e.g. current temperature). I.e. attribute reporting/polling doesn't currently work.  If I manually read the cluster in deconz it does update, and does pass those updates through to home assistant.

SwoopX, 2020-11-04:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to...#manually-configure-attribute-reporting for thermostat cluster and the attributes of interest.

SwoopX, 2020-11-22:
Closing this one as it is supported with version 2.06.00.

stevewull, 2020-11-23:
Thank you!!
Looking forward to the version 2.06.00 release.

xbmcnut, 2022-05-12:
> Thanks for all of the hard work, and I am looking forward to seeing those changes as I have now finally gotten around to installing my centralite pearl thermostat that I had lying around for a [...]

I too have these issues many years latter running deCONZ v2.14.1 under the Home Assistant  add-on version 6.12.0. The guys on the forums ask me to post here and the full thread is [](here https://forum.phoscon.de/t/centralite-pearl-3-thermostat-not-presenting-all-hvac-modes-in-home-assistant/1803)

Now that I've added `coolsetpoint` using DDF and added 'off' as the default for `mode`, everything is working with the exception of mode commands from the physical controller to Home Assistant as noted by @apmon above.

If I send a mode control (heat/cool/off) or a temp setpoint from HA to the physical thermostat, everything is sent. If I send a setpoint back to HA from the physical controller that works too. If however I send a mode command (heat/cool/off) to HA, nothing is sent.

[image]

Mimiix, 2022-05-12:
Re-opened on request.

snel6424, 2023-02-13:
Not entirely related, but has anyone found the zigbee settings to allow external sensor temperature overrides? According to the docs, this is possible, but I have not been able to figure it out