dresden-elektronik/deconz-rest-plugin #3243: Eurotronics Spirit ZigBee SPZB0001 - Expose "Set Valve Position" to REST-API

titus-leistner, 2020-09-07:
## Feature request type

Add the "Set Valve Position" attribute to the REST-API.

## Description

By default Eurotronics SPZB0001 is used by sending a target temperature set point.
However, it also has two manufacturer specific attributes: "TRV Mode" and "Set Valve Position".
If "TRV Mode" is set to "Unknown 2" (which should be 0x01), the valve can be controlled by sending an 8 Bit integer (0 - fully closed, 255 - fully opened) to "Set Valve Position". In this mode, the LCD  shows the valve position in percent.

[image]

## Considered alternatives

My goal is to control the thermostat using an external temperature sensor. Usually, people propose to set the offset accordingly. However, to me this seems like some kind of dirty hack that may cause all kinds of unexpected side effects (like oscillation, etc.).

## Additional context

The ability to set the valve position manually using the REST-API would allow to implement some PID-algorithm on the gateway (using NodeRed, etc.). I think this would be a "clean solution" for usage with an external temperature sensor.
All parameters of the PID-algorithm could be adjusted by hand, which might solve problems like overheating (as reported for the Z-Wave variant of this device in #1933 ).

I'm not sure if it's advisable to add manufacturer specific attributes to the API, but for this application it would be great imho.

ebaauw, 2020-09-08:
I’m still having issues with TRVs becoming unreachable.  I won’t be exposing setting the valve position manually, until these are resolved.  I think it would be irresponsible to do so before, as your PID algorithm would no longer be able to update the valve position.

See #1098, somewhere in the 300+ hidden remarks.

Mimiix, 2020-09-08:
**Changed from FR to DR**

titus-leistner, 2020-09-08:
@ebaauw Yes, this makes sense of course. I did not have this issue yet, but probably because my TRV was connected to the RaspBee directly.

closed 2020-10-11 by stale[bot]

Prior99, 2021-10-04:
Was there any update on this? I have the exact same usecase.