dresden-elektronik/deconz-rest-plugin #3245: Conbee 2 on Debian 10

Sasch72, 2020-09-10:
Hello everybody,

I'm trying to install the Conbee 2 Stick on my Debian. Unfortunately, I made the mistake of installing it as root. I then made up for it again under the user with su. The stick is recognized and deConz also runs. I can also call up the deConz site via the IP of my Beelink. The stick is not displayed under Gateway. Where is my mistake?
Many thanks.

## Describe the question or issue you are having

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II

### deCONZ Logs

## Additional context

Sasch72, 2020-09-10:
Hi Mimiix,

I have adjusted the enviroment. I hope that's enough for now. Tonight I'll plug the stick into a USB extension cable and then try your instructions.
Thanks first of all.

Sasch72, 2020-09-10:
Hi,
I have now connected a USB extension cable to my Beelink and misplaced the stick a bit. Then I carried out the installation as in the attached link. Now I can't even get to the "start page". No GUI is installed on the Debian.

Smanar, 2020-09-10:
Take care, the procedure is for desktop and you are using GUI. Are you sure you are not running both service in same time ?

```
sudo systemctl status deconz
sudo systemctl status deconz-gui
```

Can you make a try on another machine ? It pehaps a problem on the device (firmware, the famous usb disconnect loop)

what is the result of command
`GCFFlasher_internal -l`

Sasch72, 2020-09-10:
Hi,
I just quickly put on a Raspi and installed the Conbee 2. The data are in the environment.

These are the results of the query:

`root@debianiobroker:/home/pi1# sudo systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: e
   Active: active (running) since Thu 2020-09-10 17:46:44 CEST; 1h 31min ago
 Main PID: 475 (deCONZ)
    Tasks: 2 (limit: 4569)
   Memory: 34.0M
   CGroup: /system.slice/deconz.service
           └─475 /usr/bin/deCONZ -platform minimal --http-port=80`

`root@debianiobroker:/home/pi1# sudo systemctl status deconz-gui
● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/lib/systemd/system/deconz-gui.service; disabled; vendor preset: enabled)
   Active: inactive (dead)`

Smanar, 2020-09-10:
Not realy, all seem fine ^^.

>I just quickly put on a Raspi and installed the Conbee 2

And it have worked on the pi ?

>Now I can't even get to the "start page"

Even on http://phoscon.de/app ?

Sasch72, 2020-09-10:
The stick worked without any problems on the Pi.

When I call up the link, it looks for a gateway and finds nothing.

Smanar, 2020-09-10:
Have you tried to run deconz in debug mode ?

`
deCONZ --dbg-error=2 --dbg-info=2`

Sasch72, 2020-09-10:
This is the result:
`root@debianiobroker:/home/pi1# deCONZ --dbg-error=2 --dbg-info=2
qt.qpa.screen: QXcbConnection: Could not connect to display
Could not connect to any X display.`

Sasch72, 2020-09-11:
Good Morning,
I've finished the project for now. I installed a backup on my Raspberry Pi and will reinstall the Beelink with the Debian and then look again.
Many thanks for the support.

closed 2020-09-11 by Sasch72

Smanar, 2020-09-12:
Ha ?
So you have an idea the problem was from ?