dresden-elektronik/deconz-rest-plugin #3251: Tuya TS0215 Remote Control (Optical like Heiman HS1RC /HS1RC-M )

rweise, 2020-09-12:
The Device will connect with Phoscon but is not shown in Phoscon or IOBroker
I think it is a Heiman HS1RC clone.

Here are the Screenshots with my network settings ...

[image]
[image]
[image]

Smanar, 2020-09-20:
Yep, not exactly same cluster but can make a try.

Do you know how to compile code totest, anf have you an Unix machine ?

Kenny-59, 2020-10-17:
Hello,
I've both remotes:
the tuya/smart9 one whose reference on the package is "Zigbee Smart Remote Controller - Y"
And tje Heiman whose reference is "Zigbee Smart Remote Controller - 2-E"; as you can see on the picture.
Remote are physically the same.

The heiman is correctly associate since the Venus 2_05_81, but Tuya is not.
I've a great hope with the beta Hydaspe 2_05_85 whose some Tuya devices are supported now, but not...
[image]
[image]
[image]
Any chance of beeing supported by deconz?

Smanar, 2020-10-17:
It s possible https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

SwoopX, 2020-12-16:
I take care of it

SwoopX, 2021-02-15:
Added in 2.10.00-beta

closed 2021-03-04 by manup

Kenny-59, 2021-03-09:
hi,
nothing new for me with the 2.10.1 and my tuya "Zigbee Smart Remote Controller - Y",
still not supported
regards.