dresden-elektronik/deconz-rest-plugin #3252: deconz is constantly logging

DenisLug, 2020-09-12:
## Describe the question or issue you are having
Every few seconds I'm getting these messages on my console output:
```
16:30:59:109 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration             
16:30:59:109 Skip automatic channel change, TODO warn user                                           
16:31:07:404 don't close database yet, keep open for 900 seconds
```
This leads to constant hard drive writes on my Synology which prevents them to go into stand by.

## Environment

 - Host system: NAS Synology DS218+
 - Running method: Marthoc Docker container
 - Firmware version: 26580700
 - deCONZ version: 2.05.78 / 22.5.2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes

Smanar, 2020-09-12:
You have used the conbee with zigbee2mqtt ?

Can you show the setting panel from deconz GUI ?

DenisLug, 2020-09-13:
I haven't used the Conbee with Zigbee2MQTT before.

Do you mean the network settings? Here you can see them:
[image]

Smanar, 2020-09-13:
Try that
-Disconnect from network
-Change setting APS USE EXT PAN ID > 0x0000000000000000
- Reconnect to network

DenisLug, 2020-09-13:
I disconnected the Conbee over the GUI but after that I can't change the network settings. I've now set the APS USE EXT PAN ID to 0x0000000000000001, disconnected and connected the Conbee and also restarted the container. The error messages are almost the same:
```                                                 
17:31:35:746 0x00178801084639F4 error APSDE-DATA.confirm: 0xD0 on task                               
17:31:44:108 apsUseExtPanid is 0x0000000000000001 but should be 0, start reconfiguration             
17:31:44:108 Skip automatic channel change, TODO warn user                                           
17:31:45:367 don't close database yet, keep open for 900 seconds   
```

Smanar, 2020-09-13:
>apsUseExtPanid is 0x0000000000000001

Do you have the "1" too in your setting ?

DenisLug, 2020-09-13:
I thought you've written a greater than sign :) Now I've set it to 0 at the end. The "start reconfiguration" error has gone but these ones are left:
```                            
20:16:11:345 don't close database yet, keep open for 900 seconds                                     
20:16:11:537 0x0017880108461B3E error APSDE-DATA.confirm: 0xD0 on task   
```

Smanar, 2020-09-13:
The first one is useless and is removed on last deconz version.
The second one mean connection error, but can be minor warning too > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log

DenisLug, 2020-09-18:
I updated the container to the latest version and could now get rid of the messages above. Thank you Smanar.

One left problem now is that I'm getting (fortunately at least not so frequently than the previous messages) an error about removing websockets:
```
08:18:53:605 Remove websocket 192.168.178.75:57066 after error Unknown error                         
08:18:53:608 New websocket 192.168.178.75:57074 (state: 3)  
```
I saw that this is a already known issue (#2422) without any solution yet. What is about this?

Smanar, 2020-09-18:
Hawww, this is not good at all, not possible using deconz with that.
And 'Unknow error" is very usefull ^^.
For me it s a network problem (redirection on virtual machine ?)

Mimiix, 2020-09-18:
@DenisLug 
Websocket are managed by the user of the socket, not the server.

closed 2020-10-18 by stale[bot]

BabaIsYou, 2022-03-30:
Hi. Got the same trouble here. Put the apsUseExtPanid back to 0x000000000000000 but the ConBee II seems continue to ignore the other devices and the log contains lot of 
`6:16:26:678 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
16:16:26:923 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
16:16:29:320 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
16:16:31:723 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
16:16:34:121 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
`
Trying to get more info with INFO and APS log checked :
`6:05:44:896 Websocket 192.168.2.172:41212 send message: {"attr":{"id":"1","lastannounced":null,"lastseen":"2022-03-30T20:05Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Clé ConBee II","swversion":"0x26660700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:08:09:2c-01"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"00:21:2e:ff:ff:08:09:2c-01"} (ret = 338)
16:05:46:679 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
16:05:47:195 APS-DATA.request id: 59, addrmode: 0x03, addr: 0x00212effff08092c, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
16:05:47:196 	asdu (length: 2): 9600
16:05:47:212 APS-DATA.confirm id: 59, status: 0x00 SUCCESS
16:05:47:213 APS-DATA.confirm request id: 59 -> confirmed, timeout 0
16:05:47:246 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 117, rssi: 32
16:05:47:248 	asdu: 9600000000
16:05:47:250 APS-DATA.indication request id: 59 -> finished
16:05:47:252 APS-DATA.request id: 59 erase from queue
16:05:47:253 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
16:05:49:595 APS-DATA.request id: 70, addrmode: 0x03, addr: 0x00212effff08092c, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
16:05:49:595 	asdu (length: 2): 9700
16:05:49:607 APS-DATA.confirm id: 70, status: 0x00 SUCCESS
16:05:49:608 APS-DATA.confirm request id: 70 -> confirmed, timeout 0
16:05:49:635 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 117, rssi: 32
16:05:49:636 	asdu: 9700000000
16:05:49:637 APS-DATA.indication request id: 70 -> finished
16:05:49:638 APS-DATA.request id: 70 erase from queue
16:05:49:640 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
16:05:51:995 APS-DATA.request id: 82, addrmode: 0x03, addr: 0x00212effff08092c, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
16:05:51:996 	asdu (length: 2): 9800
16:05:52:039 APS-DATA.confirm id: 82, status: 0x00 SUCCESS
16:05:52:040 APS-DATA.confirm request id: 82 -> confirmed, timeout 0
16:05:52:096 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 117, rssi: 32
16:05:52:098 	asdu: 9800000000
16:05:52:099 APS-DATA.indication request id: 82 -> finished
16:05:52:100 APS-DATA.request id: 82 erase from queue
16:05:52:102 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
16:05:54:394 APS-DATA.request id: 93, addrmode: 0x03, addr: 0x00212effff08092c, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
16:05:54:395 	asdu (length: 2): 9900
16:05:54:410 APS-DATA.confirm id: 93, status: 0x00 SUCCESS
16:05:54:411 APS-DATA.confirm request id: 93 -> confirmed, timeout 0
16:05:54:441 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 117, rssi: 32
16:05:54:444 	asdu: 9900000000
16:05:54:445 APS-DATA.indication request id: 93 -> finished
16:05:54:445 APS-DATA.request id: 93 erase from queue
16:05:54:446 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
16:05:56:678 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
16:05:56:795 APS-DATA.request id: 105, addrmode: 0x03, addr: 0x00212effff08092c, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
16:05:56:796 	asdu (length: 2): 9a00
16:05:56:827 APS-DATA.confirm id: 105, status: 0x00 SUCCESS
16:05:56:828 APS-DATA.confirm request id: 105 -> confirmed, timeout 0
16:05:56:854 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 117, rssi: 32
16:05:56:856 	asdu: 9a00000000
16:05:56:857 APS-DATA.indication request id: 105 -> finished
16:05:56:858 APS-DATA.request id: 105 erase from queue
16:05:56:859 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
16:05:59:195 APS-DATA.request id: 117, addrmode: 0x03, addr: 0x00212effff08092c, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
16:05:59:196 	asdu (length: 2): 9b00
16:05:59:213 APS-DATA.confirm id: 117, status: 0x00 SUCCESS
16:05:59:215 APS-DATA.confirm request id: 117 -> confirmed, timeout 0
16:05:59:269 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 117, rssi: 32
16:05:59:271 	asdu: 9b00000000
16:05:59:272 APS-DATA.indication request id: 117 -> finished
16:05:59:274 APS-DATA.request id: 117 erase from queue
16:05:59:274 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
[... cut ...]
16:06:23:194 APS-DATA.request id: 233, addrmode: 0x03, addr: 0x00212effff08092c, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
16:06:23:203 APS-DATA.confirm id: 233, status: 0x00 SUCCESS
16:06:23:204 APS-DATA.confirm request id: 233 -> confirmed, timeout 0
16:06:23:217 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 117, rssi: 32
16:06:23:218 APS-DATA.indication request id: 233 -> finished
16:06:23:218 APS-DATA.request id: 233 erase from queue
16:06:23:219 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000`

Smanar, 2022-03-31:
What you have done with the conbee ? a firmware update ?
Can you show a capture from the setting panel ?
You have not a backup to use ? Have you tried the hidden restoration ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-issues

And I don't see something special on your logs, but not easy to read them.

BabaIsYou, 2022-03-31:
Hi,
Just trying to use another tool (zigbee2mqtt in docker) ;-(
Seems to have change conbee2 configuration. Restoring from backup did not really help.

Then i did a reset from deConz and a new join from my devices. Had to cut electricity  from thermostat to have it to rejoin but all is ok now.

Thanks to your plugin with Domoticz I did not have to change anything for that part ;-)

After all returns to normal I also did a manual firmware update and all is still fine.