dresden-elektronik/deconz-rest-plugin #3253: Aqara Smart Wireless Switch WXKG11LM #917 - Adding as button

Siggi0904, 2020-09-12:
The Aqara Smart Wireless Switch WXKG11LM #917 Button should be added as "button" not as sensor.

In Phoscon App it is added as button, in IOBroker ist will be shown under sensors, what is wrong.

Thanks.

Smanar, 2020-09-12:
Can you take a look in the JSON, I have take a look on your link and it s seem you have :

>"type": "ZHASwitch",

Third application can display the device as they want, so not sure the problem is from deconz.

Siggi0904, 2020-09-12:
Hi, where can I find it?
I'm using the phoscon app for adding and checking devices
and iobrocker deconz adapter to adding it in home automation.

Smanar, 2020-09-12:
Can take a look direclty in the API
`http://IP:PORT/api/KEY/sensors`

IP and PORT are the same used than for deconz.

rjenx, 2021-02-27:
Not sure this is related to the problem described here. I have tried adding an Aqara Smart Switch to my Home Assistant. I get it to join the Deconz network using Phoscon Sensor add, however only the battery is added to HA. Well if I look at the entity created for the battery it shows a related device "switch 14" but that does not exist as a separate entity. Tried adding multiple times and two different devices without success. 

I am running Deconz 2.09.01 the device firmware is WXKG11LM Rev.2 and HA is on the latest version 2021.2.0 

Any suggestions?

Smanar, 2021-02-27:
Yes, same, without the device JSON not possible to guess if the problem is from HA or deconz.

rjenx, 2021-02-27:
Pure guess, but I think the problem is with HA, I can see two devices in Deconz VNC and they appear to be communicating with the co-ordinator. I also have two devices listed in HA, but the only entities that have been created are the battery levels.

Smanar, 2021-02-27:
But HA devs will need the JSON too, so will ne nice if you can find it.

rjenx, 2021-02-27:
Hi Smanar, can you point me to where I can find it?  Is it in the Dconz docker container?

rjenx, 2021-02-27:
Further to my earlier posting, upgraded HA to version 2021.2.3 as there were some upgrades to deconz integration shown in the notes. But it made no difference.

Smanar, 2021-02-28:
You have the procedure here > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3253#issuecomment-691477297
If you don't find an api key
`curl http://IP:PORT/api/ -X POST -d '{"devicetype": "TestKey"}'`
Can need to authenticate app.

You can too use the developper tab in your browser when using phoscon.

rjenx, 2021-02-28:
I was not very successful, cannot find an api key in the developer tools view, the curl command as written results in an error message "unmatched closed brace / bracket in column 8".  If one of the devs wants to discuss or send me explicit instructions on how to get the information they need, I will be happy to help.  But this is all new to me.

Smanar, 2021-02-28:
Can you show the command and the result you have to get an api key ?

rjenx, 2021-03-02:
Hi Smanar, 
I used the curl command provided by you. As the procedure in 3253 is for IObroker and I have no idea how to obtain the API key. 

C:\Users\Rick>curl http://192.168.1.200:5900/api/ -X POST -d '{"devicetype": "TestKey"}'
**RFB 003.008
curl: (3) [globbing] unmatched close brace/bracket in column 8**

I also set HA to debug mode, this is what i captured for the switch when adding via phoscon. 
'uniqueid': '00:15:8d:00:04:77:00:a3-01-0500'},
'17': {'config': {'battery': 100,
'on': True,
'reachable': True,
'temperature': 2500},
'ep': 1,
'etag': '58ffae012961c85a96a155880a20f77d',
'lastseen': '2021-03-02T19:52Z',
'manufacturername': 'LUMI',
'mode': 1,
'modelid': 'lumi.remote.b1acn01',
'name': 'aqara_switch',
'state': {'buttonevent': 1002,
'lastupdated': '2021-02-28T13:35:33.214'},
'swversion': '20180525',
'type': 'ZHASwitch',

But again only the battery entity is created, no switch entity.

Smanar, 2021-03-02:
Ha ! I haven't see you was on windows, never tried curl on it, can be the problem, but yes it s exactly the JSON.
Can you explain me how you have done with HA, for other user, pls ?

And I can confirm the problem is from HA.

`'type': 'ZHASwitch',`
and
`state': {'buttonevent': 1002 }`

You don't need more to reconize a battery switch.

If you know the HA github you can share this JSON.

rjenx, 2021-03-02:
Thanks for your help Smanar, appreciate your patience.

mac-city, 2021-04-01:
I just purchased this switch and found this issue.  does it seem like this will be solved?  the issue dates back to 2018 from what I can tell.  the button shows up in home assistant only as a battery.  thanks for any feedback.

rjenx, 2021-04-01:
Hi Mac-City, these switches do not create entities other than the battery, you need to create some code to recognise the events created when the switch is activated. 
Here is mine as an example. 
`  - alias: toggle alarm_away when button pressed
    initial_state: 'off'
    trigger:
      platform: event
      event_type: deconz_event
      event_data:
        id: smart_switch
        event: 1002
    condition:
      - condition: state
        entity_id: alarm_control_panel.alarm
        state: 'disarmed'
    action:
      - service: script.set_away
      - service: notify.notify
        data:
          message: "Alarm On - Away Mode msg007" 
          title: "Alarm on"`
You can see the events in developer tools, bottom on events page there is a tool to listen to events. Use "deconz_event".

By the way the event codes for this switch are:

Events 
1000: click, 
1001: hold,
1002: click_up, 
1003: hold_up, 
1004: 2_click, 
1005: 3_click, 
1006: 4_click, 
1010: 5_click

Hope this helps.

mac-city, 2021-04-01:
thanks for replying, just a few minutes later and I found this solution worked well for me.  I swear, I tried this on my own and it didn't work, but tried again after restarting ha and it works fine....

https://community.home-assistant.io/t/how-to-use-stateless-programmable-switch/161810/32

Smanar, 2021-08-24:
So to resume you have the "long press" but not the "long press release" ?
Can you check logs when making this try ? with "info" and "info_l2". And BTW what is your deconz version ?

stefanuytterhoeven, 2021-08-25:
version: 2.12.06 / 19-8-2021
Firmware: 26660700
Conbee2

You're right. I get a "long press" but no "long press release"
I just tried your suggestion about "info" and "info_l2"
(I only found out the "debug view"  a few days ago :) )
but... when selecting "info" and "info_l2", I get so many lines.... I can't
follow what's happening there... I don't see a way to filter certain
devices?
Almost all my devices work via deconz (zigbee), ....
I there another way ?

manup, 2025-04-24:
Closing old issue. The WXKG11LM 2016 and 2018 switches are integrated.
For the 2018 there will be an DDF soon.

closed 2025-04-24 by manup