dresden-elektronik/deconz-rest-plugin #3258: Elko ESH 316 Endevender RF - device support request

Knodd, 2020-09-13:
This is the battery operated wireless remote for the 316GLED RF requested here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2497

Phoscon will not find it but it shows up in deCONZ after pairing. I can bind it to a light to turn on/off and dim but cannot bind to groups.

## Device

- Product name: Elko ESH 316 Endevender RF
- Manufacturer: ELKO
- Model identifier: ElkoDimmerRemoteZHA
- Device type : Please remove all unrelated device types. 
  - Remote
  - Switch

## Screenshots
Basic cluster
[image]

Node info
[image]

WTell, 2020-09-17:
I would be happy to help with testing with my ESH 316 remotes. In my experience, binding is hit-or-miss, and some remotes refuse to show up in deCONZ at all.

SwoopX, 2020-09-17:
I guess I can give it a look within the next days. It would really help if anybody could compile the plugin then for testing.

WTell, 2020-09-18:
@SwoopX I would love to help in any way I can, as long as I get a little bit of guidance on the way!

The product in question here is described by the manufacturer on https://proff.elko.no/316-endevender/category2990.html .

The battery operated remote mentioned above is meant to work in conjunction with an in-wall mounted dimmer that features the same design. The dimmer is called _Elko ESH 316GLED dimmer RF_, and is described on https://proff.elko.no/smart-316gled/category2968.html .

I have with success been able to add the in-wall dimmers to my ZigBee network, and they show up as Lights in deCONZ and Phoscon. They are have model _ElkoDimmerZHA_, and vendor either _Heiman_ **or** _ELKO_ depending on what hardware version it is and when it was manufactured. It seems reasonable to add support for the _ElkoDimmerZHA_ along with the _ElkoDimmerRemoteZHA_ that @Knodd mentioned above. Should I add the requested screenshots for the _ElkoDimmerZHA_ in this thread, or should I start a separate thread?

WTell, 2020-09-18:
@Knodd What is your experience with the _ElkoDimmerRemoteZHA_ in your ZigBee network? My experience is that they sometimes seem to "forget" their bindings, or fall deeply asleep requiring me to jiggle them a bit before they do anything. I also have a few devices that seem to refuse to be part of my network. What devices are you binding them to?

SwoopX, 2020-10-04:
Ok, so I guess I got everything ready for a test. Someone who doesn't run HA, but deconz on a bare Raspbian can give it a try. It requires the "deconz-dev" package to be installed. Just in case, I'd recommend to make a backup in Phoscon ;)

Steps should be as follows:
```
sudo systemctl stop deconz
sudo systemctl stop deconz-gui
git clone --branch elko https://github.com/SwoopX/deconz-rest-plugin.git elko
cd elko
qmake && make -j2
mkdir .local/share/dresden-elektronik/deCONZ/devices/
cp button_maps.json .local/share/dresden-elektronik/deCONZ/devices/
sudo cp /usr/share/deCONZ/plugins/libde_rest_plugin.so /usr/share/deCONZ
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

WTell, 2020-10-04:
@SwoopX Installing! Will report back.

WTell, 2020-10-05:
@SwoopX Compiled seemingly successfully, but I'm stuck just after compilation. I'm sorry to be a newbie in this area – I have little experience with qmake and make tools.

I'm assuming that `mkdir .local/share/dresden-elektronik/deCONZ/devices/` should be `mkdir ~/.local/share/dresden-elektronik/deCONZ/devices/` (is this correct?).

However, `button_maps.json` does not exist anywhere in the new `elko` directory. There only one I can find is at `/usr/share/deCONZ/devices/button_maps.json`, but this file is dated September 26. Where should I be able to find button_maps.json?

For the second-to-last line, `/usr/share/deCONZ/plugins/libde_rest_plugin.so` is also still dated September 26, does it make sense to copy this file to `/usr/share/deCONZ`?

The last line still assumes that I'm in the newly created elko folder, which would mean that it actually executes `sudo cp elko/../libde_rest_plugin.so`. Does this make sense? I created the elko folder in my home folder, was this wrong on my part?

SwoopX, 2020-10-05:
> I'm assuming that mkdir .local/share/dresden-elektronik/deCONZ/devices/ should be mkdir ~/.local/share/dresden-elektronik/deCONZ/devices/ (is this correct?).

Yes, indeed

> For the second-to-last line, /usr/share/deCONZ/plugins/libde_rest_plugin.so is also still dated September 26, does it make sense to copy this file to /usr/share/deCONZ?

The compiled plugin is from today or yesterday. The intention here is to preserve the originally installed plugin so it can be restored after testing or when needed.

> The last line still assumes that I'm in the newly created elko folder, which would mean that it actually executes sudo cp elko/../libde_rest_plugin.so. Does this make sense? I created the elko folder in my home folder, was this wrong on my part?

You should be still in the elko directory, so "../XYZ" is fine as the compiled plugin is one directory up.

WTell, 2020-10-05:
> > I'm assuming that mkdir .local/share/dresden-elektronik/deCONZ/devices/ should be mkdir ~/.local/share/dresden-elektronik/deCONZ/devices/ (is this correct?).
> 
> Yes, indeed

Thank you.

> The compiled plugin is from today or yesterday. The intention here is to preserve the originally installed plugin so it can be restored after testing or when needed.

I understand. I'll make sure I make a copy.

> You should be still in the elko directory, so "../XYZ" is fine as the compiled plugin is one directory up.

I had so far failed to notice `libde_rest_plugin.so` appearing a level above `elko/`. PEBKAC.

`cp button_maps.json ~/.local/share/dresden-elektronik/deCONZ/devices/` still makes no sense, though, as there is no new `button_maps.json` anywhere. In fact, the only place I can find this in my entire file system is at `/usr/share/deCONZ/devices/button_maps.json`. What am I missing? Am I supposed to issue
`cp /usr/share/deCONZ/devices/button_maps.json ~/.local/share/dresden-elektronik/deCONZ/devices/`?

SwoopX, 2020-10-05:
Ah, forgot about the maps, sorry. That won't be necessary as the code still uses the "old" approach

WTell, 2020-10-13:
@SwoopX I'm having trouble paring my devices, but I believe it's a lower level Zigbee issue so I'll give it a bit more time (some of my devices simply don't appear in either Phoscon or the fat GUI).

SwoopX, 2020-10-13:
All good, just let me know when there's any news.

closed 2020-10-15 by manup

oywino, 2021-01-24:
Hi @SwoopX Sorry for my late arrival. Was this device eventually added - or not ?
Mine shows up in deConz and in Phoscon, but I'm uncertain how to bind it to a light (or even better; a group).
I tried to bind it, but it failed. Event though it is visible in Phoscon, it cannot be added to a group.
The REST-API post looks like this:

`{
    "config": {
        "group": "9",
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "8f3ff2b123697e66f702c8ce7e2685fa",
    "lastseen": "2021-01-24T19:10Z",
    "manufacturername": "ELKO",
    "mode": 1,
    "modelid": "ElkoDimmerRemoteZHA",
    "name": "ElkoDimmerRemoteZHA 8",
    "state": {
        "buttonevent": null,
        "lastupdated": "none"
    },
    "swversion": "0.0.1",
    "type": "ZHASwitch",
    "uniqueid": "00:0d:6f:00:0c:b0:8b:cc-01-0006"
}`

Any help is appreciated

SwoopX, 2021-01-24:
You need to do that via the old webapp, there you can directly bind lights to a switch.

If you want to have button events to be emitted, you'd need to get me some certain messages from the deconz debug log.

Knodd, 2021-01-24:
I was able to link it with the bind box in deCONZ but not in the old webapp.
Mine never showed up in Phoscon, how did you link it @oywino ?
I started a search for switch in Phoscon and when I got an error saying "could not connect" or somethin I checked deCONZ and found it.
Linking in deCONZ seems to be maximum 3 lights (level control and on-off) and then I get binding table full

oywino, 2021-01-24:
Hi @SwoopX and thanks for getting back to me.
So, I followed your advice and in the old web app, the Elko switch/Dimmer shows up as a "group" on its own. Binding a light to it by adding it to this "group" worked fine - apparently, but the switch isn't able to control the light.

[image]

I'd love to give you info from the deConz log, but i'm afraid I do not know what you mean by "some certain messages" ?  Can you be more specific?

oywino, 2021-01-24:
@Knodd Are you using the same Elko switch/dimmer as I am?   Mine is the one Elko calls "Endevender" (ESH RS16 316GLED).  This is the one that runs on two CR2032 batteries (No Mains).

Knodd, 2021-01-24:
@oywino yes thats the one. The normal 316 ESH showed up as a light in Phoscon, but the "endevender" does not show up

oywino, 2021-01-24:
Mine shows up in Phoscon like this:

[image]

oywino, 2021-01-24:
When I added a light to the switch (using the old Web App), this is what I got in the log (INFO):

```
22:48:02:777 delay sending request 169 dt 0 ms to 0xCCCCCCFFFEC2E48A, ep: 0x01 cluster: 0x0003 onAir: 1
22:48:02:783 group 10 member 1
22:48:04:624 Add to group response for light 1. Status:0x00, capacity: 254
22:48:06:782 ZCL attribute report 0xCCCCCCFFFEC2E48A for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
22:48:20:180 ZCL attribute report 0xCCCCCCFFFEC2E48A for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
22:48:20:217 ZCL attribute report 0xCCCCCCFFFEC2E48A for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
22:48:27:258 Current channel 11
22:48:27:268 Device TTL 5160 s flags: 0x7
22:48:43:743 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
```

WTell, 2021-01-24:
@Knodd @oywino Is this the first time your are adding the the _Elko ESH 316 Endevender RF_ to your network? I've got around twelve of these added, with various degrees of success.

First of all, they are notoriously dependent on having an extremely potent power source – I've found that some CR2032 3V cells are simply not up to the task, even when brand new. Some brand new cells measuring in at 3.15 V is not enough, and I've had to dig around for ones clocking in above this (they are rare!) to get the device to pair successfully to the network. (This was using the GUI to add the device, before the update in #3409 that @SwoopX links to above).

Secondly, the devices are really bad at paying attention and caching up. I've had to try multiple times to get them to bind, and most importantly, _I tend to be more successful when I keep the device awake_ by pressing the "button"-button on the PCB every two seconds while performing the binding and repeatedly up to ten seconds afterwards. Give this a chance and see what you get!

SwoopX, 2021-01-24:
You guys should look out for those kind of messages messages coming from your switch:

`[INFO] - No button map for: eTRV0100 endpoint: 0x01 cluster: 0x000A command: 0x0A payload[0]: 000`

Knodd, 2021-01-24:
Like this @SwoopX ?
I created at binding to deCONZ
Toggle on/off:
[INFO] - No button handler for: ElkoDimmerRemoteZHA endpoint: 0x01 cluster: ONOFF (0x0006) command: TOGGLE (0x02) payload[0]: 000

Dim up:
[INFO] - No button handler for: ElkoDimmerRemoteZHA endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STEP (0x02) payload[0]: 000

Dim down:
[INFO] - No button handler for: ElkoDimmerRemoteZHA endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STEP (0x02) payload[0]: 001

SwoopX, 2021-01-25:
Yes, exactly. I'd assume those were short presses only. You should also check if there's further possibilities (bouble press, long press). Please take note that especially the long press most likely fires consecutive events, so you should ensure to look for them as well.

SwoopX, 2021-01-25:
Meanwhile, based on the information above so far, that would be the additions to `button_maps.json`, which you can basically do your selfes. The below contains the new map till the end of the file and the map above (you need to be very thorough with the brackets)

```
        },
        "lidlMap": {
            "vendor": "LIDL Livarno Lux",
            "doc": "LIDL / Livarno Lux Remote Control (_TYZB01_bngwdjsr)",
            "modelids": ["HG06323"],
            "map": [
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "ONOFF", "LIDL", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On (0)"],
                [1, "0x01", "ONOFF", "LIDL", "1", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "On (1)"],
                [1, "0x01", "ONOFF", "LIDL", "2", "S_BUTTON_1", "S_BUTTON_ACTION_TREBLE_PRESS", "On (2)"],
                [1, "0x01", "ONOFF", "LIDL", "3", "S_BUTTON_1", "S_BUTTON_ACTION_QUADRUPLE_PRESS", "On (3)"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "DimUp Press"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE", "0", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "DimUp Hold"],
                [1, "0x01", "LEVEL_CONTROL", "STOP", "0", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "DimUp Release"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "1", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "DimDown Press"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE", "1", "S_BUTTON_3", "S_BUTTON_ACTION_HOLD", "DimDown Hold"],
                [1, "0x01", "LEVEL_CONTROL", "STOP", "1", "S_BUTTON_3", "S_BUTTON_ACTION_LONG_RELEASED", "DimDown Release"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"]
            ]

        },
        "elkoMap": {
            "vendor": "ELKO",
            "doc": "Elko ESH 316 Endevender RF",
            "modelids": ["ElkoDimmerRemoteZHA"],
            "map": [
                [1, "0x01", "ONOFF", "TOGGLE", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Dim up"],
                [1, "0x01", "LEVEL_CONTROL", "STEP", "1", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Dim down"]
            ]

        }
    }
}

```

Knodd, 2021-01-25:
Thanks @SwoopX 
The toggle was a short press, this is a rotary dimmer wheel so a long press normally triggers the same action as a short press
The dim action is turning the rotary wheel and is fired multiple times during dimming so turning x degrees fires x/(device sesitivity) events

SwoopX, 2021-01-25:
I'm afraid I can't really help you with that file, since I don't run HA. On a bare Linux, it is to be found in `/usr/share/deCONZ/devices`

Knodd, 2021-01-25:
I found it in the docker containers under /var/lib/docker/overlay2 by searching for folder "deconz"

Toggle works great, and the log shows "[INFO] - Button 1002 On ElkoDimmerRemoteZHA"

Dimming seems to be expecting single button pushes and not continuous dimming with a rotary wheel.
I can only get it to dim one step and the next events are ignored

11:02:31:040 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
11:02:31:076 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
11:02:31:155 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
11:02:31:198 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
11:02:31:220 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
11:02:31:256 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
11:02:31:279 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
11:02:31:299 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
11:02:32:199 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
11:02:32:218 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
11:02:32:238 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
11:02:32:377 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
11:02:32:402 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
11:02:32:415 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA

We might need to tweak the dimming, but this switch just got a lot more useful!

SwoopX, 2021-01-25:
Seems we're getting there. However, I'd expect to also see some `STOP` commands or anything alike... If not, than it's gonna be tough.

Please double check if there's literally **any** further info message from the device.

Knodd, 2021-01-25:
I double checked and the are no stop messages.
Also there are no attribute reports like when toggling, is that normal?

The dimmer has a potensiometer that clicks every x degrees when turning. Is seems one dim buttonevent is sent for every click.
[image]

13:32:31:822 [INFO] - Button 1002 On ElkoDimmerRemoteZHA
13:32:31:932 ZCL attribute report 0x000D6FFFFE4C9350 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:32:31:949 ZCL attribute report 0x000D6FFFFE4CBF1C for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:32:31:001 ZCL attribute report 0x000D6FFFFE53A920 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:32:32:007 ZCL attribute report 0x000D6FFFFE4C91E5 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:32:34:867 [INFO] - Button 1002 On ElkoDimmerRemoteZHA
13:32:34:969 ZCL attribute report 0x000D6FFFFE4C9350 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:32:34:984 ZCL attribute report 0x000D6FFFFE53A920 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:32:35:039 ZCL attribute report 0x000D6FFFFE4C91E5 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:32:35:044 ZCL attribute report 0x000D6FFFFE4CBF1C for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
13:32:37:080 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:112 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:154 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:219 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:273 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:304 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:339 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:363 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:410 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:420 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:451 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:469 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:501 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:516 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:540 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:551 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:562 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:566 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:590 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:603 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:620 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:37:637 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
13:32:38:732 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
13:32:38:755 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
13:32:38:761 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
13:32:38:778 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
13:32:38:875 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
13:32:38:886 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
13:32:38:895 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
13:32:38:908 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
13:32:38:980 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
13:32:39:014 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
13:32:39:116 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
13:32:39:137 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
13:32:39:240 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
13:32:39:374 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA

SwoopX, 2021-01-25:
> Also there are no attribute reports like when toggling, is that normal?

That depends on the reacting devices if they support reporting on the level control cluster as well or need to be polled. I'm not too familiar with lights, since I barely have any, but I'd expect to seem some reports.

It would now be interesting to see which command is send unicast and which broadcast. On the broadcast, there should be additional messages, so I'd assume it is unicast. That would in turn also mean the lights don't get dimmed?

Anyway, I planned to polish the debug output a bit to get some additional information. Maybe there's also something hidden for the Dim commands we currently don't see in the debug output.

Knodd, 2021-01-25:
Let me know what I can do to help with debugging!

I've tested a bit and the lights don't seem to change brightness when dimming so maybe we need a different button mapping?

oywino, 2021-01-25:
@Knodd - you seem far ahead of me now.  Did you have to change to new batteries to get it kicking - like @WTell suggested ?
Mine is visible in both Phoscon as well as deConz, but a GET at the REST-API reveals absolutely nothing whether I turn or click. It just says "buttonevent": null
Could you give me som guidance to get going?

Knodd, 2021-01-25:
@oywino I didn't change the batteries and I can still not see it in Phoscon.
What I did was use the deCONZ GUI and create a binding between both cluster 6 and 8 and the deCONZ node with the switch as source and deCONZ as destination. That makes the switch send the button events to deCONZ.

If you have not used the binding box in deCONZ before this is how to do it:
Click "open web gui" and choose "deCONZ"
[image]

Under Panels menu, check "Bind Dropbox"
[image]

Click and drag cluster "0006 on/off" of the dimmer remote to source
Click and drag from the top of the configurator (I renamed it deCONZ) to destination
[image]

Important:
Before you press "Bind", press the "push button" on the elko dimmer (the little white one under the rotary wheel), this wakes the button so it's not in sleep mode.

Now press "bind" and make sure it says "success", if not then try again.

Then you need to Click and drag cluster "0008 Level control" to source and press bind again.

After this the button events for toggle and dim will be sent to deCONZ

oywino, 2021-01-25:
Hi, binding it to Conbee II (deconz) persistantly resulted in **failure: timout**
I deleted and repaired the dimmer several times but to no avail. I also replaced the batteries.
However, binding it an IKEA Leptiter light was immediately successful and I could use it to control the light easily. But of course, that binding was the same as "TouchLink"  The signals passes directly between the dimmer and the light, bypassing the coordinator (Conbee II).
Looking at your dialog with @SwoopX, I get the impression that a modification of the button_maps.json was done (?)  Did you have to do that to get it to work?

Knodd, 2021-01-25:
Did you wake the switch before binding?

Touchlink is how I've been using it so far, but the binding table is very small (6 spots) so you can only control 3 lights with on/off and dimming.
I have several up/down wall lamps with IKEA bulbs so that's 1,5 lamp for 1 dimmer :-)

After binding to deconz i got this in the log:
[INFO] - No button handler for: ElkoDimmerRemoteZHA endpoint: 0x01 cluster: ONOFF (0x0006) command: TOGGLE (0x02) payload[0]: 000

After editing button_maps, on/off worked for the group created in "old web app" but dimming is not working

oywino, 2021-01-25:
Did you edit the button map before or after binding it to the coordinator (Conbee II) ?

Knodd, 2021-01-25:
After, but I don't think it matters

oywino, 2021-01-25:
Can you see what firmware version your dimmer has ?

WTell, 2021-01-25:
> Touchlink is how I've been using it so far, but the binding table is very small (6 spots) so you can only control 3 lights with on/off and dimming.

Do you know how to clear the binding table? I have one remote that I can't seem to be able to add, and I believe that it might be because the binding table is full – but it doesn't seem to be bound to anything but one lamp.

Knodd, 2021-01-25:
software 0.0.14, I got a file to update from Elko so it should be the latest.
I never found out how to clear the binding table when I didn't know what binding to unbind so I used  "delete node" in the deconz gui and then repaired it

oywino, 2021-01-25:
After playing around with it and attempting several binds, I eventually got an error message **binding table: full**
Then I just removed the batteries and held the little switch for 10 seconds and re-inserted the batteries again and I could continue to experiment with binding. So my impression is that this did empty the table - but I'm not sure. I haven't had that error message again.

Knodd, 2021-01-25:
Great tip. Did you not have to pair it again?
I've been googling for hours how to factory restore this thing before I found out I could delete it and pair again

oywino, 2021-01-25:
Could you post the update file? My SW version is 0.0.1

Knodd, 2021-01-25:
Here it is, you have to extract the .bin file to a micro SD card.
**MAKE SURE THE CARD IS FORMATTED TO FAT32** and not FAT, or the switch dies...
[ELKDF001.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5869393/ELKDF001.zip)

oywino, 2021-01-25:
No, I did not have to pair it again.  The touchlink connection works directly without the need for coordinator assistance. It actually never disappeared from deConz at all.  This is typical for all ZigBee devices I have ever used. Even if I leave them unused for month at a time, they are online as soon as they get power (batteries or mains). Never needed to re-pair anything.
Thanks for the file 👍

oywino, 2021-01-25:
According to Elko docs, there are two different update files; One for the "Endevender SW" and the other is for the "ZigBee radio FW" The update procedure is somewhat different. Do you now which one you sent me?

Knodd, 2021-01-25:
this is the "Endevender SW"

Knodd, 2021-01-25:
Just remembered I also updated the zigbee FW a few months ago.
Found it here, on the bottom of the page: [https://www.elko.se/support/programvara/](https://www.elko.se/support/programvara/)

oywino, 2021-01-25:
Now the update went well, and binding too went well. Success on the first attempt.
But then what?  Did you manage to get events from it into HA ?
Is now the time to edit the binding table?

Knodd, 2021-01-25:
Great!
No need to edit the binding table, just bind the two clusters (on/off and level control) to the configurator.
Then you get the add the lines @SwoopX listed in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3258#issuecomment-766639222
I found the file by searching for "button_maps" in "/var/lib/docker" folder of my Debian install (HA host system since I installed HA in docker). Strangely I had the file in two locations so I just edited them both.

oywino, 2021-01-26:
I have searched everywhere for "button_maps*" but I cannot find any match anywhere. I've search both inside the container as well as in the host system. I would expect to find  such a file inside the container (?) as this must be a file closely linked to Home Assistant, right ?  (It has no relevance to the host system).
This is the content of my /var/lib/docker/ folder in the host system:

[image]

I've searched through all subfolder from here.
Inside the container, there is no folder named `docker ` inside /var/lib/

Knodd, 2021-01-26:
It's in the overlay2 folder. I have limited experience with docker so I don't know why it's there instead of the containers folder.
I also had a bit of trouble at first, but first found it searching for "deconz" which resulted in lots of hits to go through.
Try a search for "deCONZ" (case sensitive), those where the hits with a "devices" folder in for me

oywino, 2021-01-26:
Still no luck.
I found 5 `deconz` subfolders under `overlay2` but none of them contains a filname remotely close to the name `button_maps` inside (or further down into subfolders). I did a case-insensitiv search so it would find any deconz or deCONZ folders

[image]

None of these deconz folders had a subfolder named `devices`

Knodd, 2021-01-26:
@oywino hmmm thats strange... Then I don't know where to look

@SwoopX I checked another switch and got a few error messages:
14:22:06:242 [INFO] - Button 1002 On/Off ElkoDimmerRemoteZHA
14:22:06:714 ZCL attribute report 0x001788010647B322 for cluster: 0x0406, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
14:22:11:487 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
14:22:11:644 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
14:22:11:667 ZCL attribute report 0x086BD7FFFE1E168B for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:11:673 ZCL attribute report 0x680AE2FFFE1C5BD5 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:11:704 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
14:22:11:758 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
14:22:11:810 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
14:22:11:863 ZCL attribute report 0xEC1BBDFFFEEB194C for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:11:910 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
14:22:11:916 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
14:22:11:922 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
14:22:11:935 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
14:22:11:941 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
14:22:11:989 ZCL attribute report 0xEC1BBDFFFEEB0C3C for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
**14:22:12:263 0x0000000000000000 error APSDE-DATA.confirm: 0xD2 on task
14:22:12:365 0x0000000000000000 error APSDE-DATA.confirm: 0xD2 on task
14:22:12:422 0x0000000000000000 error APSDE-DATA.confirm: 0xD2 on task**
14:22:12:932 ZCL attribute report 0x000D6FFFFE1056DF for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:16:160 Current channel 20
14:22:16:169 Device TTL 6960 s flags: 0x6
14:22:16:673 ZCL attribute report 0x086BD7FFFE1E168B for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:16:680 ZCL attribute report 0x680AE2FFFE1C5BD5 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:16:817 ZCL attribute report 0xEC1BBDFFFEEB194C for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:16:857 ZCL attribute report 0xEC1BBDFFFEEB0C3C for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:19:875 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
14:22:19:896 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
14:22:19:942 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
14:22:19:998 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
14:22:20:068 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
14:22:20:076 [INFO] - Button 3002 Dim down ElkoDimmerRemoteZHA
14:22:20:815 ZCL attribute report 0x000D6FFFFE1056DF for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:21:677 ZCL attribute report 0x086BD7FFFE1E168B for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:21:687 ZCL attribute report 0x680AE2FFFE1C5BD5 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:21:819 ZCL attribute report 0xEC1BBDFFFEEB194C for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:21:861 ZCL attribute report 0xEC1BBDFFFEEB0C3C for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:24:760 ZCL attribute report 0xEC1BBDFFFEEB194C for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:25:214 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
14:22:25:269 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
14:22:25:340 ZCL attribute report 0xEC1BBDFFFEEB0C3C for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:25:361 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
14:22:25:405 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
14:22:25:449 delayed group sending
14:22:25:449 delayed group sending
14:22:25:454 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
14:22:25:493 [INFO] - Button 2002 Dim up ElkoDimmerRemoteZHA
14:22:25:522 delayed group sending
14:22:25:523 delayed group sending
14:22:25:583 delayed group sending
14:22:25:623 delayed group sending
**14:22:25:803 0x0017880103A1623D error APSDE-DATA.confirm: 0xE9 on task**
14:22:25:808 ZCL attribute report 0x000D6FFFFE1056DF for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:26:666 ZCL attribute report 0x086BD7FFFE1E168B for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:26:688 ZCL attribute report 0x680AE2FFFE1C5BD5 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:26:825 ZCL attribute report 0xEC1BBDFFFEEB194C for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
14:22:26:865 ZCL attribute report 0xEC1BBDFFFEEB0C3C for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000

oywino, 2021-01-26:
@SwoopX do you any advice to me where I can locate the file where the `button_maps` is supposed to be entered?
I run HA-Docker v2021.1.5 on a Linux server.

Mimiix, 2021-01-26:
@oywino that question is unrelated to this topic. I suggest asking in the HA git.

@Knodd https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log

oywino, 2021-01-26:
> @oywino that question is unrelated to this topic. I suggest asking in the HA git.
I don't understand - how can it be unrelated if this is what it takes to get the `Elko ESH 316 Endevender RF` to work ?

Mimiix, 2021-01-27:
@oywino Because it's a generic question, not specifically related to this device :)

oywino, 2021-01-27:
> @oywino Because it's a generic question, not specifically related to this device :)
It is apparently necessary to add the data suggested by @SwoopX listed in #3258 (comment) to a file named `button_maps` in order to get this device to operate.  The data is not generic, but specific for this device. I only need to know where to find (or create) the file for HA/deConz to detect and use it.
Anyone else reading this topic will need to know that as well, in order to successfully use this device.

[21 more replies]