dresden-elektronik/deconz-rest-plugin #3260: New firmware versions online.

Mimiix, 2020-09-14:
**Latest firmware files are online now:**

##### ConBee II
http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_ConBeeII_0x26650700.bin.GCF

##### RaspBee II
http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_RaspBeeII_0x26650700.bin.GCF

##### RaspBee I and ConBee I
http://deconz.dresden-elektronik.de/deconz-firmware/deCONZ_Rpi_0x26370500.bin.GCF

* These have source routing enabled with max. 32 source routes, where the oldest entries are replaced automatically by newer ones. The amount will likely be increased in later releases, but should already work well.
* If a source route and a "normal" route entry exists the source route is preferred. This is non-standard but seems to work better.
* The firmware has general improvements for serial protocol robustness.
* ConBee II and RaspBee II got improved frame counter handling, to mitigate problems where after a power-cycle the network could have appeared to be lost until the frame counter hits its old value again.

_Originally posted by @manup in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261#issuecomment-691733972_

jurriaan, 2020-09-14:
Great news! Do we need the new deCONZ version to enable the source routing, or does it work with the current version?

manup, 2020-09-14:
The automatic routing works independent of the deCONZ version. But to create manual source routes the newest deCONZ version is required (currently building as alpha).

manup, 2020-09-15:
@Adminiuga the protocol hasn't changed, but the command parser in the firmware is now much stricter to force correctness and shield against some system services which may probe devices.

For example if the firmware version request doesn't contain the expected number of bytes, the command would be dropped.

pittbull, 2020-09-15:
Will this updated firmware have any positive impact on the boot loop issue that has been reported?

salopette, 2020-09-15:
> The automatic routing works independent of the deCONZ version. But to create manual source routes the newest deCONZ version is required (currently building as alpha).

is alpha only for Jessy or is that also possible with Buster?

Mimiix, 2020-09-15:
@pittbull "Bootloop issue"?

Bootloops often happen when firmware is not correct. I've asked you a few times to create _your own_ issue with _your own_ problem. Could you please stop posting everywhere asking this?

Adminiuga, 2020-09-15:
@manup is format of the version command still as indicated in the docs?
[image]

this is what I'm writing to the serial port, but getting no response
```
3739  09:59:14 write(6, "\xc0\x0a\x02\x00\x08\x00\x01\x00\x22\xc9\xff\xc0", 12) = 12
3739  09:59:14 read(6, "\xc0\x0a\x02\x00\x0a\x00\x03\x00\x22\x0c\x01\xb8\xff\xc0", 1024) = 14
3739  09:59:14 write(6, "\xc0\x0d\x03\x00\x05\x00\xeb\xff\xc0", 9) = 9
3739  09:59:21 write(14, "\xc0\x0a\x02\x00\x08\x00\x01\x00\x22\xc9\xff\xc0", 12) = 12
3739  09:59:21 read(14, "\xc0\x0a\x02\x00\x0a\x00\x03\x00\x22\x0c\x01\xb8\xff\xc0", 1024) = 14
3739  09:59:21 write(14, "\xc0\x0d\x03\x00\x05\x00\xeb\xff\xc0", 9) = 9
3739  09:59:25 --- SIGINT {si_signo=SIGINT, si_code=SI_KERNEL} ---
```

Adminiuga, 2020-09-15:
Is this correct "Read Firmare Version Request" ?

Type | Field | Value
--- | --- | --- |
U8 | Command ID | Version (0x0D)
U8 | Sequence number | 0 -- 255
U8 | Reserved | Shall be set to 0
U16 | Frame length | 0x09
U32 | Reserved for version response | Shall be set to 0x00000000

Hypfer, 2020-09-15:
I'd've really loved a short explaination and/or link to what source routing is exactly.

It's of course always possible to google a term, but since it's really not common knowledge (i think?) that would've been very useful

JBS5, 2020-09-15:
> I'd've really loved a short explaination and/or link to what source routing is exactly.
> 
> It's of course always possible to google a term, but since it's really not common knowledge (i think?) that would've been very useful

This comment could be helpful https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261#issuecomment-687902156

Adminiuga, 2020-09-15:
Would love to know what are the actual protocol commands to set and read the source routes

manup, 2020-09-15:
The updated docs will be available soon.
Basically it's an extension to `APS_DATA_REQUEST (0x12)`.

Flags: `0x02`   (source route included)
And after the `Radius` field at the end.

```
U8   Relay count                  // max. 7
U16 Relays[Relay Count]
```

Note that the array is reversed, where the first item at index 0 is the actual destination.

Manual source routes are only available if the protocol version is greater equal `0x010C` (1.12)

Revorge, 2020-09-16:
I tried flashing the new firmware to my raspbee twice, but that is not reflected in the GUI. I have succesfully flashed firmware before, so not sure whats up. Followed the instructions carefully and double checked that I flashed the correct version.

[image]

manup, 2020-09-16:
Can you please show the command line output of GCFFlasher (note deCONZ must be closed before flashing).

Revorge, 2020-09-16:
> Can you please show the command line output of GCFFlasher (note deCONZ must be closed before flashing).

[image]

manup, 2020-09-16:
Looks like it doesn't do much, please try following command:

`sudo GCFFlasher_internal -x 3 -t 60 -f deCONZ_Rpi_0x26370500.bin.GCF`

This will try harder and print some more debugging info.

Revorge, 2020-09-16:
> Looks like it doesn't do much, please try following command:
> 
> `sudo GCFFlasher_internal -x 3 -t 60 -f deCONZ_Rpi_0x26370500.bin.GCF`
> 
> This will try harder and print some more debugging info.

Output from last retry:

[image]

salopette, 2020-09-16:
You have stop a gui from Deconz, not deconz self.
Maybee try 
Systemctl stop deconz
Systemctl stop deconz-gui

manup, 2020-09-16:
I think that's a bug in GCFFlasher itself, it thinks this is a RaspBee II with a newer bootloader V3.
Working on a Fix now.

manup, 2020-09-16:
Another question, it appears the reset via PIN doesn't work, do you have latest wiringpi 2.52 installed?

`dpkg -l wiringpi`

If not:

```
cd /tmp
wget https://project-downloads.drogon.net/wiringpi-latest.deb
sudo dpkg -i wiringpi-latest.deb
```

tubalainen, 2020-09-16:
@Revorge try to power cycle the dongle (or the raspberry). It worked for me after getting lots of random errors while trying to flash the new fw.

tubalainen, 2020-09-16:
After installing .82 and running the new conbee 1 fw I had to power cycle all my devices before the mesh was properly rebuilt.
Ive got 61 nodes, Mainly IKEA.

[image]

manup, 2020-09-16:
After the initial update in some cases it may take 5–10 minutes until the nodes respond (since the frame counter from the former version wasn't stored in a timely manner). This is fixed in the new version and shouldn't happen in the future anymore.

tubalainen, 2020-09-16:
> After the initial update in some cases it may take 5–10 minutes until the nodes respond (since the frame counter from the former version wasn't stored in a timely manner). This is fixed in the new version and shouldn't happen in the future anymore.

Great to hear!
My hidden IKEA dimmer drivers are a pain to reach :)

manup, 2020-09-16:
Judging from the green lines it seems to be a strong mesh with good connections. Routes should be figured out after a while and should stay healthy.

tubalainen, 2020-09-16:
> Judging from the green lines it seems to be a strong mesh with good connections. Routes should be figured out after a while and should stay healthy.

Will report back - Ive had quite some issues with the light bulbs farthest away from the dongle (2 or 3 mesh jumps). Mainly failing the off commands for some strange reason.

Revorge, 2020-09-16:
> Another question, it appears the reset via PIN doesn't work, do you have latest wiringpi 2.52 installed?
> 
> `dpkg -l wiringpi`

[image]

> @Revorge try to power cycle the dongle (or the raspberry). It worked for me after getting lots of random errors while trying to > flash the new fw.

Thanks, but restarting / power cycling seems to make no difference.

manup, 2020-09-16:
This version contains a new GCFFlasher with a few related fixes:

http://deconz.dresden-elektronik.de/raspbian/alpha/deconz-2.05.82-raspbian-jessie-alpha~git.188b85.3b813a_armhf.deb

Revorge, 2020-09-16:
> This version contains a new GCFFlasher with a few related fixes:
> 
> http://deconz.dresden-elektronik.de/raspbian/alpha/deconz-2.05.82-raspbian-jessie-alpha~git.188b85.3b813a_armhf.deb

That solved it :) Thanks.

[image]

rcmcronny, 2020-09-17:
For me, the Deconz 2.05.82 version causes all my sensors (2 aqara motion, 1 hue motion, 1 aqara multisensor, 1 aquara door/window sensor) to not send statusupdates anymore. The stay, if i press the sensor button, they send an update to deconz, but thats it. Repair and so on, does not help. I am back to 2.5.80 and all works as it should. Took me 2 hours to get the cause. 
I also downgraded the firmware, but that was not the cause, so i am on the latest deCONZ_ConBeeII_0x26650700.bin.GCF firmware for my "Conbee II". 
Does anybody does also have issues with sensors not getting updated anymore with 2.5.82 ?

andriej, 2020-09-17:
@rcmcronny some of mine also stopped working. Randomly, so I don't blame the LQI issues (I have good coverage with routers).
Didn't try downgrading tho.

Mimiix, 2020-09-17:
@rcmcronny Can you open a seperate issue for that issue?

rcmcronny, 2020-09-17:
> @rcmcronny Can you open a seperate issue for that issue?
Done in #3273

virtualducttape, 2020-09-27:
Just wanted to drop a quick note (doesn't seem like it's worth a separate issue... ), for the Firmware Update instructions in Windows, there should be a note to run the CMD prompt as "Administrator", the same way there are "sudo" commands for the other options. Without running as an administrator the flasher just starts and then says "retry, failed" after it times out.

Mimiix, 2020-09-28:
@virtualducttape Do you mean the [wiki page?](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually)  I've edited it just now. good catch!

virtualducttape, 2020-09-28:
@Mimiix That's the one, thanks! 👍

sorryusernameisalreadytaken, 2020-09-30:
> 
> 
> > I'd've really loved a short explaination and/or link to what source routing is exactly.
> > It's of course always possible to google a term, but since it's really not common knowledge (i think?) that would've been very useful
> 
> This comment could be helpful [#1261 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261#issuecomment-687902156)

I didn't get that... :-(
Thought that zigbee's idea is to create a mesh network and source routing seems to be a kind of "routing to the source". Can somebody tell me: what was before the case and what is it now? :D

rasssta, 2020-10-02:
I've had issues with the ConBee II. Had to send pretty much every command twice for it to go through with 26650700 (lights and outlets). Reverted to 26580700 and it works like a charm again.

salopette, 2020-10-02:
Today after the restart of Pi comes this great message!
[image]

Ronoon, 2020-10-03:
Just tryed to update my old conbee 1 USBstick with the latest firmware and got no chance with the actual windows flashers.
So back to the roots and using the avail version 2.10 von dresden-elektroniks and with a little effort got the latest flashupgrade installed. 
In between tryed the versions between 210 and the latest with no sucess.
Also tryed it on two different machines, so no arguments are open to windows failures.
Seems the flasher has some not so nice issues in the windows version since 2.10

Mimiix, 2020-10-07:
Closing this one, There's a new version 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3377

closed 2020-10-07 by Mimiix