dresden-elektronik/deconz-rest-plugin #3265: Scene settings are fluctuating randomly and are not saved

ChrisPrefect, 2020-09-15:
I made a video about the issue:

[Phoscon App - Google Chrome 2020-04-14 23-48-18.zip](https://github.com/dresden-elektronik/phoscon-app-beta/files/4477942/Phoscon.App.-.Google.Chrome.2020-04-14.23-48-18.zip)

It also looks like the connection is really unstable. After only about 60 devices added, commands often get not trough to all lights and some lights are shown as "offline" or they don't react to brightness or temperature changes in the app. 

This is all on one floor and lights are online every few meters, so the mesh network should reach every device without a problem.

ConBee II stick on a Raspi3B+ with USB extension cable. Zigbee channel 25, Unifi-WIFI on channel 1.

ChrisPrefect, 2020-10-11:
This issue still persists. Even with the latest beta version of Deconz and Phoscon.

Does this work for other users or is this a bug that just affects me and/or Tradfri-bulbs?

Mimiix, 2020-10-12:
I've forwarded it ;)

manup, 2020-11-01:
Related: https://github.com/dresden-elektronik/deconz-rest-plugin-v2/issues/5

ChrisPrefect, 2020-11-24:
It's still not fixed, right?

gilbert-grape, 2020-11-26:
The problem still exists. It is very annoying and frustrating. I have never seen a such buggy software.
I have the problem with Ikea Tradfri, but also with Philips and Xioami.

ChrisPrefect, 2020-12-18:
This issue is still not fixed.

iisworks, 2021-02-03:
Just encountered the same issue in Phoscon (Docker Latest, group of Aqara led tunable white bulbs):
2.09.01 / 12/23/2020
26680700

When I try to save 2 scenes, they always end up with the dim level + color temperature of last saved entry. Example:
1) Create scene "Day:" 80% dimlevel + Color 1, save -> OK
2) Add scene "Night": 20% dimlevel + Color 2, save
3) Now both Day and Night have dimlevel 20% + Color 2.

closed 2021-04-03 by github-actions[bot]