dresden-elektronik/deconz-rest-plugin #3266: Release notes for Venus (v2.05.81) states that firmware should be updated manually

fi-sch, 2020-09-15:
## Describe the bug
Not sure if that's a bug in release notes or it is intended but it seem that two points in release notes are inconsistent.

I mean:
`The firmware needs to be installed manually.`
and
`Improved firmware update #3256`

Since #3256 states that `Enable firmware update via Phoscon App again.`, manual update requirement shouldn't be the case.
Am I right?

Mimiix, 2020-09-15:
You are correct,  However:  some users don;t update their deCONZ versions due to various reasons. You can use the latest deCONZ/Phoscon again to firmware update, but manual updating could be done too.

fi-sch, 2020-09-15:
Ok. Thanks for clarifying!

Mimiix, 2020-09-15:
You are most welcome :) @fi-sch

closed 2020-09-15 by Mimiix