dresden-elektronik/deconz-rest-plugin #3268: Conbee 2 - NUC - USB passthrough - USB Stick not stable

Robert83-cmd, 2020-09-16:
hello together,

it is a little bit frustrating because since two days I am trying to find a solution:
I have NUC with proxmox installed and a debian lxc.
conbee 2 usb stick is connected.

problem: seems that the usb connection is not stable (directly connected).
lsusb is showing the device - but after some time the device is not visible any more and the device id is changing, for example:

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 8087:0aaa Intel Corp.
**Bus 001 Device 075: ID 1cf1:0030 Dresden Elektronik**
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@pve:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 8087:0aaa Intel Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@pve:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 8087:0aaa Intel Corp.
**Bus 001 Device 087: ID 1cf1:0030 Dresden Elektronik**
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

now I updated the firmware on windows - version 26650700 - no change.
I had the same issue on a raspberry 3, but that was ok - I used the phoscon GUI, now deconz should run headless with phoscon on port 80.

Furthermore - I tried to set up some rules for /etc/udev/rules.d to make sure passthrough is working. 
But that is almost impossible because the device ID is changing too fast. Example:

lxc.mount.entry: /dev/bus/usb/001/026 dev/bus/usb/001/026 none bind,optional,create=file

Is that a normal behavior? If yes - may someone please explain to setup passthrough and what kind of settings must be done on the lxc?

On the lxc (and proxmox host) I was able to get the serial, for example:
udevadm info -a -n /dev/ttyACM0 | grep '{serial}' | head -n1
.. if I setup a passthrough for ttyACM0 - but GCFFlasher_internal -l is showing me an empty table.
BUT: if the device was not available again - the output changes:
root@pve:~# udevadm info -a -n /dev/ttyACM0 | grep '{serial}' | head -n1
Unknown device "/dev/ttyACM0": No such file or directory

Would be very nice if someone give me a hint!!!

Thanks a best regards
Robert

Smanar, 2020-09-16:
On promox you have perhaps a similar issue (with a solution) here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3001#issuecomment-688898913

Robert83-cmd, 2020-09-17:
Thank you @Smanar for ur reply!
I downgraded proxmox, restarted everything several times - but still - same issue.

But I think u are right. It must be a proxmox problem. Not sure why other usb devices are working and only conbee is reconnecting all the time, maybe I must find another solution. 
I checked the NUC firmware, too - did an update, but no change.

Smanar, 2020-09-17:
Yep normally using windows is a good way to test device (and upgrade the device if you have some issue), realy few issue on windows, if deconz work on windows, lot of chance the problem is from the other machine.

BTW I have forget something important, on proxmox you can set the USB port port, don't use the USB 3.0 but the usb 2.0.

Robert83-cmd, 2020-09-17:
Yes, using usb 2.0.

Can you explain how to set the USB port?

Will do another test with a raspberry soon.

Smanar, 2020-09-17:
Here for exemple

https://dannyda.com/myupload/2020/08/image-18.png?v=1597391307

You have a case to uncheck.

Goldwing1973, 2020-09-17:
Same problem here with the Conbee 2 connected pass thru on a VMWare ESXi 6.7 NUC with a Home Assistant VM.
Every couple of days the stick changes from /dev/ttyACM1 and back to /dev/ttyACM2.
Using the /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2195308-if00 as port in Home Assistant doesn't work correctly either, it resolves to the port before the spontaneous port change.

Robert83-cmd, 2020-09-17:
ok - that is crazy. now I have the same problem with a raspberry. 

`pi@raspberrypi:~ $ lsusb
Bus 001 Device 021: ID 1cf1:0030 Dresden Elektronik
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~ $ lsusb
Bus 001 Device 023: ID 1cf1:0030 Dresden Elektronik
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`
syslog:
`Sep 17 12:19:33 raspberrypi mtp-probe: bus: 1, device: 20 was not an MTP device
Sep 17 12:19:41 raspberrypi kernel: [148278.950902] usb 1-1.2: USB disconnect, device number 20
Sep 17 12:19:41 raspberrypi kernel: [148279.254210] usb 1-1.2: new full-speed USB device number 21 using dwc_otg
Sep 17 12:19:41 raspberrypi kernel: [148279.388623] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep 17 12:19:41 raspberrypi kernel: [148279.388643] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 17 12:19:41 raspberrypi kernel: [148279.388658] usb 1-1.2: Product: ConBee II
Sep 17 12:19:41 raspberrypi kernel: [148279.388672] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep 17 12:19:41 raspberrypi kernel: [148279.388685] usb 1-1.2: SerialNumber: DE1964489
Sep 17 12:19:41 raspberrypi kernel: [148279.391148] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
Sep 17 12:19:41 raspberrypi mtp-probe: checking bus 1, device 21: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2"
Sep 17 12:19:41 raspberrypi mtp-probe: bus: 1, device: 21 was not an MTP device
Sep 17 12:19:41 raspberrypi mtp-probe: checking bus 1, device 21: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2"
Sep 17 12:19:41 raspberrypi mtp-probe: bus: 1, device: 21 was not an MTP device
Sep 17 12:19:42 raspberrypi systemd[1]: Started Session c4 of user pi.
Sep 17 12:22:34 raspberrypi kernel: [148452.518801] usb 1-1.2: USB disconnect, device number 21
Sep 17 12:22:35 raspberrypi kernel: [148452.835317] usb 1-1.2: new full-speed USB device number 22 using dwc_otg
Sep 17 12:22:35 raspberrypi kernel: [148452.969619] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep 17 12:22:35 raspberrypi kernel: [148452.969640] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 17 12:22:35 raspberrypi kernel: [148452.969654] usb 1-1.2: Product: ConBee II
Sep 17 12:22:35 raspberrypi kernel: [148452.969668] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep 17 12:22:35 raspberrypi kernel: [148452.969682] usb 1-1.2: SerialNumber: DE1964489
Sep 17 12:22:35 raspberrypi kernel: [148452.970858] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
Sep 17 12:22:35 raspberrypi mtp-probe: checking bus 1, device 22: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2"
Sep 17 12:22:35 raspberrypi mtp-probe: bus: 1, device: 22 was not an MTP device
Sep 17 12:22:35 raspberrypi mtp-probe: checking bus 1, device 22: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2"
Sep 17 12:22:35 raspberrypi mtp-probe: bus: 1, device: 22 was not an MTP device
Sep 17 12:22:42 raspberrypi kernel: [148460.454796] usb 1-1.2: USB disconnect, device number 22
Sep 17 12:22:43 raspberrypi kernel: [148460.805359] usb 1-1.2: new full-speed USB device number 23 using dwc_otg
Sep 17 12:22:43 raspberrypi kernel: [148460.939742] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep 17 12:22:43 raspberrypi kernel: [148460.939763] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 17 12:22:43 raspberrypi kernel: [148460.939777] usb 1-1.2: Product: ConBee II
Sep 17 12:22:43 raspberrypi kernel: [148460.939791] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep 17 12:22:43 raspberrypi kernel: [148460.939805] usb 1-1.2: SerialNumber: DE1964489
Sep 17 12:22:43 raspberrypi kernel: [148460.942271] cdc_acm 1-1.2:1.0: ttyACM0: USB ACM device
Sep 17 12:22:43 raspberrypi mtp-probe: checking bus 1, device 23: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2"
Sep 17 12:22:43 raspberrypi mtp-probe: bus: 1, device: 23 was not an MTP device
Sep 17 12:22:43 raspberrypi mtp-probe: checking bus 1, device 23: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2"
Sep 17 12:22:43 raspberrypi mtp-probe: bus: 1, device: 23 was not an MTP device
`

Maybe is was on the raspberry all the time like this, too and I did not see it?

Mimiix, 2020-09-17:
Looks like broken firmware.  Did you click the "update firmware" button earlier?

Follow this: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually 

to update the firmware.

Robert83-cmd, 2020-09-17:
Yes, updated the firmware yesterday (windows), now version deCONZ_ConBeeII_0x26650700.bin.GCF is running.

Smanar, 2020-09-17:
And have you tried too use deconz on windows, to see if the issue is from the device or the host ?
You are not using USB 3.0 ?

Robert83-cmd, 2020-09-17:
@Smanar yes.

Windows 10 = all fine! Was able to do to the firmware update, too
Raspberry (fresh Pi OS) = reconnecting problem
NUC with proxmox and vlx (with debian) = reconnection problem

SwoopX, 2020-09-17:
@Robert83-cmd you have mtp probes in your output interfering. This is probably modemmanager not playing nice.

Robert83-cmd, 2020-09-17:
@SwoopX modemmanager is not installed any more. 

I tried to check if there might be a problem with the usb autosuspend - but still no success. Values get reseted after the USB stick reconnects again.

Smanar, 2020-09-17:
And you have a fresh buster install on the raspberry ? (updated ?)

Robert83-cmd, 2020-09-17:
@Smanar correct. fresh buster:

pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
pi@raspberrypi:~ $ uname -r
5.4.51-v7+

Robert83-cmd, 2020-09-17:
Short update:

seems that conbee 2 is running now stable again on the raspberry.
might be my fault - I hope that I did not wast ur time!

after starting deconz service the device id did not change any more. 
if I restart the raspberry - the device is reconnecting all the time - from device id 1 up to 7 - but if I start the deconz service, the device id is not changing any more. 

so the main problem on proxmox seems to be that I cannot install deconz service on the host, only on a lxc.
maybe the usb connection between lxc and the host is a problem or not correct configured.
will check that again tomorrow. 

if it will not work on the lxc - i will keep the conbee connected to the raspberry and connect with the deconz from the lxc to the raspberry. hopefully it will work like that.

but what I don't understand is: might that really be the problem? is the deconz service keeping up the connection to the usb device and because of that the device id cannot change frequently?

thanks!

Smanar, 2020-09-17:
>but what I don't understand is: might that really be the problem? is the deconz service keeping up the connection to the usb device and because of that the device id cannot change frequently?

I don't understand what you want to say, but yes, when you run deconz, it take and keep the connexion to the usb device, no other applications can use it. But it s not normal ?

Robert83-cmd, 2020-09-18:
Yes, maybe that is normal - but I don't have deep knowledge about usb connections.

the problem I have now if I want to switch to proxmox is - the device id keeps changing.
I can mount the usb stick on the host, in the config file for the vlx like:

lxc.cgroup.devices.allow: c 189:* rwm
lxc.mount.entry: /dev/bus/usb/001/002 dev/bus/usb/001/002 none bind,optional,create=file
lxc.cgroup.devices.allow: c 188:* rwm
lxc.mount.entry: /dev/ttyUSB0 dev/ttyUSB0 none bind,optional,create=file

but because of the changing ID the connection is not working (in my opinion).

anyway - I think that is not a conbee problem! 
will check another [article ](https://forum.proxmox.com/threads/usb-device-in-unprivilegierten-lxc-durchreichen.52985/)tomorrow, same problem.

closed 2020-10-25 by stale[bot]

ceesvanegmond, 2021-01-03:
@Robert83-cmd Did you solve this issue? I'm having the same problem with the Conbee stick. Other USB devices are working find though.

Goldwing1973, 2021-01-03:
> @Robert83-cmd Did you solve this issue? I'm having the same problem with the Conbee stick. Other USB devices are working find though.

It is still happening here, but way less frequent.
Now it happens mostly within 24 hours after reloading automations in Node Red.
But if I do a full reboot of Home Assistant after adjusting automations it keeps on working fine for weeks.
And when it stops responding then it doesn't hop ttyACM's anymore, only the Deconz integration loses its connection to the Conbee 2, hitting the restart button in Deconz fixes it.
Today I upgraded Deconz to version: 6.6.2, I hope it brings even more stability.

mbrrg, 2021-01-10:
Having the same problems on a Raspberry Pi 3, Raspbian Buster. Latest ConBee 2 firmware, original RPi power adapter.

```
[ 9926.968976] usb 1-1.2: USB disconnect, device number 114
[ 9927.265061] usb 1-1.2: new full-speed USB device number 115 using dwc_otg
[ 9927.399556] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[ 9927.399584] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9927.399600] usb 1-1.2: Product: ConBee II
[ 9927.399615] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[ 9927.399630] usb 1-1.2: SerialNumber: XXXXXXXXXXX
[ 9942.843125] usb 1-1.2: USB disconnect, device number 115
[ 9943.135065] usb 1-1.2: new full-speed USB device number 116 using dwc_otg
[ 9943.269426] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[ 9943.269449] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9943.269464] usb 1-1.2: Product: ConBee II
[ 9943.269479] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[ 9943.269494] usb 1-1.2: SerialNumber: XXXXXXXXXXX
[ 9946.685177] usb 1-1.2: USB disconnect, device number 116
[ 9946.985096] usb 1-1.2: new full-speed USB device number 117 using dwc_otg
[ 9947.119431] usb 1-1.2: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[ 9947.119454] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9947.119470] usb 1-1.2: Product: ConBee II
[ 9947.119485] usb 1-1.2: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[ 9947.119500] usb 1-1.2: SerialNumber: XXXXXXXXXXX
```