dresden-elektronik/deconz-rest-plugin #3269: Curtain close action not reported by Deconz addon to Home Assistant

bsfaxi, 2020-09-16:
## Describe the bug

I have paired a new acquired curtain zigbee switch with the Raspbee II. It appears correctly in Home Assistant with the 3 action buttons : Up, Stop and Down.
The 2 actions UP and STOP work correctly, but not the DOWN one. It seems even not taken into account as I don't see its state reporting in the log!
[image]

However, using the deConz app (trough VNC), I can control correctly the 3 actions. The binding is also successful.
[image]

But fixing the Current Postion Lift Percentage is not successful. 
[image]

## Steps to reproduce the behavior

## Expected behavior

The close button closes the curtain.

## Screenshots

## Environment

 - Host system: HassOS 4.13 on Raspberry Pi 4 Model B
 - Running method: Home Assistent (Supervisor 242), deCONZ Add-on v6.2.3
 - Firmware version: 26520700
 - deCONZ version: 2.05.79
 - Device: RaspBee II
 - Do you use an USB extension cable: No of course :)

### deCONZ Logs

Please find below an extract of the deconz addon from Home Assistant log file set in DEBUG mode.
We can see that there is no reporting of the close state !

2020-09-16 21:54:29 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":null,"lastseen":"2020-09-16T19:54:29Z","manufacturername":"_TYZB01_dazsid15","modelid":"TS0302","name":"Window covering device 2","swversion":null,"type":"Window covering device","uniqueid":"bc:33:ac:ff:fe:17:05:4e-01"},"e":"changed","id":"2","r":"lights","t":"event","uniqueid":"bc:33:ac:ff:fe:17:05:4e-01"}

-- UP/OPEN
2020-09-16 21:54:31 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': False}" to "172.30.33.1 /lights/2/state"
2020-09-16 21:54:31 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/2/state/open': True}}]

2020-09-16 21:54:31 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":null,"lastseen":"2020-09-16T19:54:31Z","manufacturername":"_TYZB01_dazsid15","modelid":"TS0302","name":"Window covering device 2","swversion":null,"type":"Window covering device","uniqueid":"bc:33:ac:ff:fe:17:05:4e-01"},"e":"changed","id":"2","r":"lights","t":"event","uniqueid":"bc:33:ac:ff:fe:17:05:4e-01"}

2020-09-16 21:54:31 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"2","r":"lights","state":{"bri":254,"lift":100,"on":true,"open":false,"reachable":true},"t":"event","uniqueid":"bc:33:ac:ff:fe:17:05:4e-01"}

-- STOP
2020-09-16 21:54:32 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'bri_inc': 0}" to "172.30.33.1 /lights/2/state"
2020-09-16 21:54:32 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/2/state/stop': True}}]

2020-09-16 21:54:32 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"2","r":"lights","state":{"bri":127,"lift":50,"on":true,"open":true,"reachable":true},"t":"event","uniqueid":"bc:33:ac:ff:fe:17:05:4e-01"}

-- DOWN/CLOSE
2020-09-16 21:54:33 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': True, 'bri': 254}" to "172.30.33.1 /lights/2/state"
2020-09-16 21:54:33 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/2/state/lift': 100}}]

2020-09-16 21:54:33 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"lastannounced":null,"lastseen":"2020-09-16T19:54:33Z","manufacturername":"_TYZB01_dazsid15","modelid":"TS0302","name":"Window covering device 2","swversion":null,"type":"Window covering device","uniqueid":"bc:33:ac:ff:fe:17:05:4e-01"},"e":"changed","id":"2","r":"lights","t":"event","uniqueid":"bc:33:ac:ff:fe:17:05:4e-01"}

## Additional context

Here are screenshots of the device's info:

[image]

[image]

Mimiix, 2020-09-16:
Hi, this is not something we manage or maintain. You should create a issue in the HA integration git. 

Closing this one.

Kane610, 2020-09-16:
This is an issue with deconz not reporting change in state and the integration exposes the issue

bsfaxi, 2020-09-16:
Sincerely, I'm quite lost! :)
Could you please point me where to post this issue?
I'm a newbie in this new world and trying to learn. Please bear with me.
Thanks.

Mimiix, 2020-09-17:
@Kane610 I have had this for a while. The curtains not always work with the arrows but do with the set cover.

@bsfaxi Could you use the https://www.home-assistant.io/integrations/cover/#service-coverset_cover_position 

Service? See what happens if you put it on 0 or 100?

Kane610, 2020-09-17:
> @Kane610 I have had this for a while. The curtains not always work with the arrows but do with the set cover.
> 
> @bsfaxi Could you use the https://www.home-assistant.io/integrations/cover/#service-coverset_cover_position
> 
> Service? See what happens if you put it on 0 or 100?

I would expect sending commands to deconz to work. What is not working is the state change signalling from deconz to hass

Smanar, 2020-09-17:
Ha yep it s missing
>2020-09-16 21:54:31 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"2","r":"lights","state":{"bri":254,"lift":100,"on":true,"open":false,"reachable":true},"t":"event","uniqueid":"bc:33:ac:ff:fe:17:05:4e-01"}

For the down/close

bsfaxi, 2020-09-17:
Thanks to all for your your precious support.
Here are the tests I've just did:

Service cover.set_cover_position
- putting 100 opens the blinds
- putting 0 does nothing
- putting 50 does nothing too

Service cover.set_cover_tilt_position
- putting 100 does nothing
- putting 0 does nothing
- putting 50 or 30 does nothing too

Service cover.open_cover opens the blinds.
Service cover.stop_cover stops the blinds.
Service cover.close_cover does nothing.

Service cover.open_cover_tilt does nothing.
Service cover.close_cover_tilt does nothing.

I've attached the log file.
[home-assistant (2).log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5241410/home-assistant.2.log)

SwoopX, 2020-09-17:
Haven't we had the same issue with a different device a couple of months back? That somehow rings a bell...

Shouldn't pydeconz/deconz send a "on: true" only, for closure?

Mimiix, 2020-09-18:
Re opening until we know for sure.

bsfaxi, 2020-09-19:
> Re opening until we know for sure.
> 
> 

Thanks for your understanding.
Please let me know if you need me to do some actions.

Smanar, 2020-09-19:
>The 2 actions UP and STOP work correctly, but not the DOWN one. It seems even not taken into account as I don't see its state reporting in the log!

It s just the reporting is missing, or the curtain don't move at all ? Like when you have used cover.close_cover ?

bsfaxi, 2020-09-21:
> And BTW you have too python error in the homeassistant core.
> 
> >

Hi,
Yes, the curtain doesn't move at all like when I have used cover.close_cover.

Concerning the python error, I have it in the beginning each time when I re-paired the curtain with the Raspbee. It takes some minutes. And then, HA re-sync everything with the integration. The error disappear. So, this error lasts only a while and is not problematic.

With deconz, I tried the command "go to lift". I had no response from the device.

Smanar, 2020-09-21:
>With deconz, I tried the command "go to lift". I had no response from the device.

Can explain the problem.
Do you think your device support Set postion command ? (From documentation)

You have tried lift with value et percentage ? Make try with titlt (perhaps it s reversed)

But if it don't work with deconz, it's normal it  don't work with the API.

BTW some covering NEED calibration for the set position feature work.

bsfaxi, 2020-09-21:
> Can explain the problem.
> Do you think your device support Set postion command ? (From documentation)
>
> You have tried lift with value et percentage ? Make try with titlt (perhaps it s reversed)
>
> But [...]

The device doesn't support the "set position command" at all. in Deconz, I've tried the "Lift" with both "value" and "Percentage" options and also the "Tilt" one, with no success. The curtain switch doesn't respond to those commands, even if it shows "success" in Deconz !
When I use the device with the "Smart Life" app, which is designed for originally, I can see that it only supports UP, DOWN and STOP commands.

I hope it's clear. :-)

bsfaxi, 2020-09-21:
I just found that there is a similar issue for the same device here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2999.

Smanar, 2020-09-21:
Ok so 2 solutions.
- The third app need to use the good command according to device
- Use a list in the code to convert "lift" to "open" or some device.

bsfaxi, 2020-09-22:
Thanks Smanar .
So, is it something to handle in the Deconz code ? Or in the Integration (Deconz Addon) ? Or do you mean in Home Assistant (the third app) ?
In the Phoscon app, the curtain is displayed as a light and it doesn't respond to any command. In my humble opinion, it should be supported originally by the Zigbee GW provider and thus Deconz, and then exposed with the correct API to third parties.
How could I help?

Smanar, 2020-09-22:
It s a 4 lines code, nothing hard for deconz.

But there s a little problem now, there is a major change in the actual deconz code, so
- do you know how to compile code to test ?
- do you prefer use the last code, but need some more manipulation to be usable, or a file based on Version 80 (just for test) ?

bsfaxi, 2020-09-22:
I'm not aware of any method to apply the latest version (80) in my actual configuration (HassOS, Deconz addon and a Raspbee II). Maybe, I'll have to build another environment on a different SD card. If you point me to the correct procedure, I'll do it. 😀

Smanar, 2020-09-22:
So you have a PI, but only one ?
So you need to shut down all your home automation for tests ? (If you have Xiaomi device, they don't like that)

Other thing, to test we need to use the API, so not possible with phoscon, if you haven't HA for test we will need to use command line (easy, on clean install just copy / paste).

To resume, 10 mn of work for me 2/3h for you ^^

If you are still agree I will put the procedure tomorrow with a modified 80 version for test.
I can make it in blind mode if you are not hurried (the code is realy easy), but will take more than 1 month if I m wrong to have it in HA.

bsfaxi, 2020-09-22:
Yes, I have only one PI (for now).
And I'm using it for testing right now with multiple devices and brands in order to build my future production. 😃 
So, I can play with it. Let's give it a try.

Smanar, 2020-09-23:
Ok So I have used an old version for test, faster.

- So you need a normal OS, buster for exemple, take with desktop, can be usefull later.
- Update all the OS
- Install deconz on it.
- Install the dev package and compile the code (from https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only)
```
sudo apt install deconz-dev
git clone --branch curtain https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```
- Run deconz
- Include the device again
- Try using the bugged command with the API, you can use HA if you have it on another computer or with the API (from https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/)

```
curl -H 'Content-Type: application/json' -X PUT -d '{"lift": 100}' http://IP_DECONZ:80/api/API_KEY/lights/ID/state
curl -H 'Content-Type: application/json' -X PUT -d '{"lift": 0}' http://IP_DECONZ:80/api/API_KEY/lights/ID/state
```

IP_DECONZ is the machine IP
ID the device ID (I think it will be 1)

If you can have an API_KEY the easy way, on the local machine (I m using local host and don't need to authenticate app)
`curl http://127.0.0.1:80/api/ -X POST -d '{"devicetype": "TestKey"}'`

The modification I have done
```
    //Some device don't support lift, but third app can use it
    if (hasLift)
    {
        if (taskRef.lightNode->manufacturer() == QLatin1String("_TYZB01_dazsid15"))
        {
            hasLift = false;
            hasOpen = true;
            if (targetLiftZigBee > 0)
            {
                targetOpen = false;
            }
            else
            {
                targetOpen = true;
            }
        }
    }
```

If you have error message, you can stop direcly, I am often wrong.

bsfaxi, 2020-09-25:
Hi @Smanar ,
I created a new whole DEV environment as you requested. It took me more than 4 hours (it was finally a HDMI port problem). 😆 
Anyway, I can confirm that the new code works perfectly using the curl commands and also using the integration from Home Assistant. So, Well done. 😃
What is the next step?

Smanar, 2020-09-25:
Just wait for PR validation, I have just made the PR, with exactly the same code.

bsfaxi, 2020-09-25:
Many thanks to everyone.

closed 2020-10-01 by manup