dresden-elektronik/deconz-rest-plugin #3271: Heavy disk (sdcard) usage since middle August

tnowak, 2020-09-17:
Hi. I'm using deConz/Phoscon 2.05.80 / 14.08.2020 in a docker container (marthoc/deconz:latest) on RPi with microSD card.
Lately I've noticed substantial IOPS increase in writes to my sdcard.
Using iotop -ao I've narrowed it down to (most probably) deconz process writing frequently to /opt/deconz/zll.db
Please see my IOPS graphs:

[image]

So it seems it started mid August and that would coincide with new deConz version released on Aug 14th.
I would really like to keep my sdcard writes low, so my questions are:

1. Is it really by something that changed in deConz software? can someone confirm my observations?
2. Can I somehow tune / define deConz disk write interval?

Smanar, 2020-09-21:
Ha, it s perhaps the new filed added in device information `lastseen` ?

tnowak, 2020-09-22:
Might be. I have quite a lot of devices (~50) so it might be an escalating factor.

Smanar, 2020-09-22:
And IDK if it s usefull to memorize them, I think they are reseted after a reboot, but not sure.

closed 2020-10-25 by stale[bot]

arthurgeek, 2020-12-23:
I also see consistent IOPS from my this running on docker. Any advices?

Smanar, 2020-12-24:
Perhaps we can skip the write for "last seen" modifications ? Or ignore them for database (they don't need to be stored) ?

manup, 2021-01-09:
> Perhaps we can skip the write for "last seen" modifications ? Or ignore them for database RW (they don't need to be stored) ?

Yes, saving and restoring them through database is ok, but setting the attribute shouldn't trigger a write database.