dresden-elektronik/deconz-rest-plugin #3273: sensors not working with 2.5.82 (only manuell button press sends the status to deconz)

rcmcronny, 2020-09-17:
As mentioned in issue #3260 ( https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3260#issuecomment-694121865 )

Version 2.5.82 (beta)
Firmware: 0x26650700 (also testet with earlier firmware, same)

After starting the sensors are grey and "not reachable" or have the last state open, closed whatever it was), no motion is visible, no reaction to events. Pressing the button, sends the state to deconz and it is shown. Only a button press updates the state, it d

Comment in release issue was as followed:

`For me, the Deconz 2.05.82 version causes all my sensors (2 aqara motion, 1 hue motion, 1 aqara multisensor, 1 aquara door/window sensor) to not send statusupdates anymore. The stay, if i press the sensor button, they send an update to deconz, but thats it. Repair and so on, does not help. I am back to 2.5.80 and all works as it should. Took me 2 hours to get the cause.
I also downgraded the firmware, but that was not the cause, so i am on the latest deCONZ_ConBeeII_0x26650700.bin.GCF firmware for my "Conbee II".
Does anybody does also have issues with sensors not getting updated anymore with 2.5.82 ?
`
No specific logs, but did not create debug logs.
Switches and Lamps were working, only sensors not.

andriej, 2020-09-17:
some of mine Aqara sensors also stopped working. Randomly, so I don't blame the LQI issues (I have good coverage with routers).
Did the downgrade few minutes ago, they are still unavailable but will check if it changes.

trewbene, 2020-09-17:
I have the same problem.
Since version >=2.05.80 my sensors (5x Philips Hue Motion) do not work anymore (lamps and switches work without problems).
After the downgrade to 2.05.80 they work again.

rcmcronny, 2020-09-17:
@trewbene 
> I have the same problem.
> Since version >=2.05.80 my sensors (5x Philips Hue Motion) do not work anymore (lamps and switches work without problems).
> After the downgrade to 2.05.80 they work again.

Sorry to say that, but I am happy, to be not alone ;) That rules out a mistake on my side 👍

dvbavel, 2020-09-18:
I can confirm seeing the exact same thing on 2.5.82 and latest FW. Especially the Aqara sensors are flaky here.

manup, 2020-09-18:
Can you please check that the endpoint cluster list in deCONZ GUI expands and shows the name of the clusters as in:

[image]

Expect the FFFF the clusters should not show as "Unknown".

rcmcronny, 2020-09-18:
I have headless, is there a way to get it headless ? 
Or can another user with this issue and running the gui can check for the needed info ?

andriej, 2020-09-18:
Connect with VNC

rcmcronny, 2020-09-18:
> Connect with VNC
I know, that i can setup this, but atm it is headless and there is no x server, nor vnc server installed, so this need to be setup and configured.

dvbavel, 2020-09-18:
[image]

manup, 2020-09-18:
On headless you can check if the ZCLDB is the correct one:

```
$ md5sum /usr/share/deCONZ/zcl/general.xml 
c129f1ffb1e1d77d569bf08bde94e985  /usr/share/deCONZ/zcl/general.xml
```

We had for a brief amount of time a version of .82 online which had a bug in this file, it was recompiled quickly but may have been installed in some setups: https://github.com/dresden-elektronik/deconz-rest-plugin/commit/3b813ad770b9932508de6fb704600e89467a723b

I'm not sure if that's the case here, but the checksum should be the above one.

manup, 2020-09-18:
@dvbavel that looks ok, so it might be a different issue.
Does the problem also happen with .81?

irekzielinski, 2020-09-18:
Same issue - latest deconz (82) updated 1h ago, docker on synology.
State of the motion sensors is visible in Phoscon, but not passed to the home assistant (only button presses are).

Mimiix, 2020-09-18:
@irekzielinski  Some devices are not "devices" in ha but need to use the Events to trigger a automation.

rcmcronny, 2020-09-18:
For me, i used deCONZ to check for status updates (motion, open,closed etc) and the are not send, and there is motion or the door open/closed. The are only updating with a button press on the sensors.So i do not think its really important, what Automation is in use ?!

Mimiix, 2020-09-18:
@rcmcronny 

Please follow the instruction as per Manuel:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3273#issuecomment-694733732

rcmcronny, 2020-09-18:
> 
> 
> @rcmcronny
> 
> Please follow the instruction as per Manuel:
> [#3273 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3273#issuecomment-694733732)

~#  md5sum /usr/share/deCONZ/zcl/general.xml
c129f1ffb1e1d77d569bf08bde94e985  /usr/share/deCONZ/zcl/general.xml

dvbavel, 2020-09-19:
Did rebuild my container today with just 2.5.82 again, seems al is fine again. So LGTM. Didn't check for the md5sum but guess I was also having the wrong version. 

Case closed to me.

trewbene, 2020-09-26:
After downgrading to 2.5.80 and upgrading to 2.5.82 again, everythings seems to work correctly. So i think i had the version with the bug.

rcmcronny, 2020-09-26:
So i close this issue, seems to have really been the defective file and now all is fine.

closed 2020-09-26 by rcmcronny