dresden-elektronik/deconz-rest-plugin #3274: Retrieve End Dvices consumption

Rania-M, 2020-09-17:
Hi, i am implementing a home assistant application and i would like to know if there is an (official) way to retrieve power consumption from zigbee end devices via the deconz API , i searched for it but mostly found only userss requests about this subject  (hte devices do retrieve the consumption information but it seems that the API doesn't consider it)

- Environment : version : 2.05.78
 - Host system: Raspberry Pi 
 - Running method: Raspbian 
 - Device: RaspBee II

Thank you in advance

Smanar, 2020-09-17:
I think it s not possible having it. The consumption depend of usage. End device use hibernating working mode, so if you don't disturb it, it will have different consumption than another one that is trigger every 10mn.

SwoopX, 2020-09-17:
Was is that question supposed to mean? Either a device offers measurement capabilities, then it is exposed though the API, or not. Simple as that.

Smanar, 2020-09-17:
He don't sepak about power consumption mesured by a device but the device consumption itself (end device never provide power measurement)

SwoopX, 2020-09-17:
None does afaik

Rania-M, 2020-09-23:
the device i need to retreive the information from is develco smart plug, in the product presentation, it is clearly mentionned that the device does return the consumed power , i woul like to know if there is a way to get this information using the deconz api, thank you

Smanar, 2020-09-23:
Ok so perhaps the device is not supported, so you can have light entries, for on/off but not the sensor with consumption.

You have the model id of the device ? (visible in deconz)

SwoopX, 2020-09-23:
Which zigbee modelID does the device have? I know that a Develco smart plug is supported (SPLZB-131).

Rania-M, 2020-09-24:
well its the same (SPLZB-131)
The device is supported, i can ,turn  it on and off using the API, the problem is retrieving the power consumption information

SwoopX, 2020-09-24:
How do you determine the availability of the power and consumption resources? Have you queried the API directly?

Eventually, you need to enter sensor search in Phoscon again and switch the plun on while a load is applied. Alternatively, read the attributes of the basic cluster with active search.

Rania-M, 2020-09-24:
Well i actually just send a get lights request to the API, i should also mention that the plug is detected as a light, not as a sensor.
When i send this request i get the state information, which i can usé to switch it on and off but there is no consumption value anywhere, maybe im not singers the right method

Smanar, 2020-09-24:
The on/off feature is an a "light" (always created)
The consumption is in a "sensor" (only crated if device is supported)

Rania-M, 2020-09-24:
You mean that one device can be represented as a light and sensor at the same time ?

Smanar, 2020-09-24:
Yep (or more)

Rania-M, 2020-09-24:
Ok i will check this out , thanks

Mimiix, 2020-09-25:
@Rania-M Can i close this?

closed 2020-10-19 by Mimiix