dresden-elektronik/deconz-rest-plugin #3275: ROBB Rotary dimmer and EcoDim Rotary dimmers state updates

aryelevin, 2020-09-17:
When I'm using my ROBB Rotary dimmer and EcoDim Rotary dimmers their state is not reflected either on phoscon or homebridge-hue.

Only after few minutes (about 10 minutes or so).

What could be the problem?

## Environment
deCONZ v2.05.81 on Debian 10.

 - Host system: Debian 10.
 - Running method: VirtualBOX.
 - Firmware version: (26650700)
 - deCONZ version: (2.05.81)
 - Device: ConBee II
 - Do you use an USB extension cable: yes

I have some ubisys D1 units and their state is reflected immediately.

SwoopX, 2020-09-19:
Which device? Which state? I'm afraid your input is a little rare here.

aryelevin, 2020-09-20:
OK, so here you have a links to both devices with this issue.
1. https://en.robbshop.nl/draaidimmer-zigbee-ecodim
2. https://en.robbshop.nl/hue-rotary-dimmer-zigbee-2-wire

So the problem is that their state is not being updated in real-time, only after few minutes it changes state on the apps.
Steps to reproduce:
1. Open App/phoscon and navigate to the "light" on the group is it in or inside "lights" list.
2. Toggle power manually on the dimmer itself or change brightness.
3. No changes on the App/phoscon.

Expected behavior: Changes made locally should be reflected on the App/phoscon immediately.

Thanks for everyone.

SwoopX, 2020-09-20:
Ok, now the picture becomes clearer. It seems both devices are "not fully supported", which means: Typically, lights and wired switches are supported out of the box. However, some details might be missing, which I suspect in your case being the bindings for attribute reporting. That would explain why the attributes are polled like around every 10 mins (and only then updated), whereas the update should be instant with active reporting.

Can you please provide a screenshot from the node info panel as well as from the basic cluster with the attributes being read for both devices?

aryelevin, 2020-09-21:
OK, here you have it.
[image]
[image]
[image]
[image]

aryelevin, 2020-09-22:
@SwoopX Any idea on how to solve it?
Can I configure the bindings myself? (even by updating the db file for example?)

Thanks a lot

timovos, 2020-09-24:
I'm also having the Robb rotary dimmer. Sometimes the dimmer has to be pushed twice to turn it on. Possibly this issue is related. Following!

aryelevin, 2020-09-25:
@timovos  I don't think it could be related, since the local control should not be interfering from the wireless control.
It might possible that you dim you lights so low that it appears to be off, and then when you trying to switch them on, your first push is actually turns it off and the second push turns it on at some higher level as configured by defaults.
If this is the case, you should configure the Min. Brightness level to match your bulbs (look on the owners manual came with the dimmer), and then it will be solved.

aryelevin, 2020-09-25:
For everyone else, I used the "Bind Dropbox" pane to do the Binding myself, simply but dragging the On/Off cluster and the Level Control cluster as source (each at a time of-course) and the Coordinator (Its "Configuration Tool" service) as Destination and clicked Bind Button.
Then it works perfectly.

Thanks for everyone

closed 2020-09-25 by aryelevin