dresden-elektronik/deconz-rest-plugin #3277: Hue motion sensor light detection not respected

max3-2, 2020-09-17:
## Describe the question or issue you are having
Paired a Hue motion sensor with Phoscon. This is running in a room with a ceiling light. **I want to only turn on the light when the ceiling light is off**, e.g. the room is dark.
I adjusted `tholddark` and `tholdoffset` so that the sensor shows `dark : false` and `daylight : true` If the ceiling light is on. Even though., the sensor triggers (icon in Phoscon) and the connected group lights up and starts the motion sensor mode.

- Do I need to update / reboot the sensor in any way to apply the new values? 
- What exactly does the sensitive (currently on 2, max. Value) influence if light levels are set elsewhere (see above)

## Environment
Vanilla updated deconz (service), phoscon, RPi, RaspBee II

max3-2, 2020-09-19:
Attaching my GET sensors with the three elements form the sensor in a state where it **should not trigger due to ambient light** but it does...Also screenshots from the setup in phoscon. Hoping anyone can help or at least hast the same issue or setup??

Also, why can I click and activate the motion symbol with switches, see image 4...?
[image]
[image]
[image]
[image]

```
"15": {
    "config": {
      "alert": "none",
      "battery": 100,
      "delay": 62,
      "ledindication": false,
      "on": true,
      "pending": [
      ],
      "reachable": true,
      "sensitivity": 2,
      "sensitivitymax": 2,
      "usertest": false
    },
    "ep": 2,
    "etag": "7b6102904a4f49d47613b829aeef019e",
    "lastseen": "2020-09-19T09:17Z",
    "manufacturername": "Philips",
    "modelid": "SML001",
    "name": "Bewegungsmelder Bad",
    "state": {
      "lastupdated": "2020-09-19T09:14:22.553",
      "presence": true
    },
    "swversion": "6.1.0.18912",
    "type": "ZHAPresence",
    "uniqueid": "00:17:88:01:08:65:94:f5-02-0406"
  },
  "16": {
    "config": {
      "alert": "none",
      "battery": 100,
      "ledindication": false,
      "offset": 0,
      "on": true,
      "pending": [
      ],
      "reachable": true,
      "usertest": false
    },
    "ep": 2,
    "etag": "3e952097b0861e9f148357ada6c85f0a",
    "lastseen": "2020-09-19T09:17Z",
    "manufacturername": "Philips",
    "modelid": "SML001",
    "name": "Bewegungsmelder Bad",
    "state": {
      "lastupdated": "2020-09-19T09:13:57.878",
      "temperature": 2184
    },
    "swversion": "6.1.0.18912",
    "type": "ZHATemperature",
    "uniqueid": "00:17:88:01:08:65:94:f5-02-0402"
  },
  "17": {
    "config": {
      "alert": "none",
      "battery": 100,
      "ledindication": false,
      "on": true,
      "pending": [
      ],
      "reachable": true,
      "tholddark": 7500,
      "tholdoffset": 3000,
      "usertest": false
    },
    "ep": 2,
    "etag": "306f4c1ef7a87fe96a11da075271a3cf",
    "lastseen": "2020-09-19T09:17Z",
    "manufacturername": "Philips",
    "modelid": "SML001",
    "name": "Bewegungsmelder Bad",
    "state": {
      "dark": false,
      "daylight": true,
      "lastupdated": "2020-09-19T09:17:07.872",
      "lightlevel": 22521,
      "lux": 179
    },
    "swversion": "6.1.0.18912",
    "type": "ZHALightLevel",
    "uniqueid": "00:17:88:01:08:65:94:f5-02-0400"
  },
```

max3-2, 2020-09-21:
A last bump - I cant be the only one running Hue motion sensors .... still not finding any way forward. The sensor triggers as above but phoscon turns the group on even though daylight is true...

Mimiix, 2020-09-22:
This is a Phoscon issue(mostly), not a deCONZ issue. Moving the issue.