dresden-elektronik/deconz-rest-plugin #3278: Inconsistent fade in/up behavior

stevenwfoley, 2020-09-18:
Different fading in/up behavior when turning on a light via the group and the individual light.
Light: 3A Smart Home DE - LXT56-LS27LX1.7

## Steps to reproduce the behavior
  1. Go to a light group and turn on or off the light group. The light respects the "OnOff Transition Time" attribute and smoothly fades in and out for the respective on and off actions.
  2. Go to a light group and individually turn on one of the lights in the group. The light will turn on to a low dim, pause for a second, and then start fading up. The fade out of the off action works as expected.

## Expected behavior
The fading behavior should be the same between these two methods of turning on/off a light. Ideally the group method would be the proper on/off behavior because its the smoothest.

## Environment
 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26370500
 - deCONZ version: 2.05.82
 - Device: RaspBee I

stevenwfoley, 2020-09-23:
I've done a bit of sniffing with Wireshark and captured some revealing data.

This is what happens when I turn on a light at the Group level:
Command: On (0x01)

This is what happens when I turn on an individual light within the same Group:
Command: Move to Level with OnOff (0x04)
Payload:
     Level: 2
     Transition Time: 0.0 seconds
Command: Move to Level (0x00)
Payload:
     Level: 254
     Transition Time: 0.9 seconds

So the behavior I'm experiencing seems deliberate and intended. But then the question becomes: why such an elaborate way of turning on an individual light instead of just issuing the simple On command as is done at the Group level?

ebaauw, 2020-10-17:
> So the behavior I'm experiencing seems deliberate and intended.

Yes, see https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2553.  This is the logic by the Hue bridge, but it seems it doesn't work well for non-Hue lights.  I wanted to change the code, so it applies this logic only to Hue lights, but they screwed up the code, by adding logic for shutters and locks in the light handling.  I'm sorry, I'm not touching this hell anymore.

closed 2020-11-24 by stale[bot]