dresden-elektronik/deconz-rest-plugin #3280: Sengled E1D-G73 sensor not working

stevenwfoley, 2020-09-18:
## Describe the question or issue you are having
Hello,
I have a couple of the Sengled E1D-G73 Smart Window & Door Sensors that are not working in deconz v2.05.82. The node shows up in the deconz gui app, but is not reading/populating the clusters. The dot for the drop-down cluster menu is not even showing on the node object in the gui.
Issue #2870 indicates that this device should be supported as of v2.05.78.

## Screenshots
[image]

## Environment
 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26370500
 - deCONZ version: 2.05.82
 - Device: RaspBee I

stevenwfoley, 2020-09-20:
I was finally able to get this device to display properly in the deconz gui by repeatedly resetting and repairing. It's also showing up in phoscon, but its not working. None of the values in phoscon are being updated, like battery and open/close status, etc...

Smanar, 2020-09-20:
The inclusion procedure in the API is probably not finished.
Something you can try, as you have deconz and the device already included
- Set phosocn in permit join (add new sensor)
- Read again the basic attribute (cluster "basix 0x0000, press the "read" button with using the device to wake it up)

It will trigger a new device inclusion, but easier for the API.

stevenwfoley, 2020-09-22:
Thanks for the help, but I'm giving up on this sensor. I've tried countless attempts to get anything from this device, but its not giving me a thing, and I can find numerous comparable alternatives.

closed 2020-09-22 by stevenwfoley