dresden-elektronik/deconz-rest-plugin #3282: I can't figure out if the Aqara D1 (WXKG06LM / WXKG07LM) are supported or not?

fribse, 2020-09-19:
## Describe the question or issue you are having

I have just gotten a few buttons of WXKG06LM and one WXKG07LM.
It seems that I can pair it, it shows up as three topics, like described elsewhere.
It doesn't show up as a light or sensor in phoscon.
if I go to the API and just do a GET of /APIKEY/ it doesn't show up in the list if I search for the MAC address..
If I go to read values in the Attribute Editor it fails (just says 'reading').
If I try to listen for deconz_event in Home Assistant I don't see any messages.

I looked at the https://zigbee.blakadder.com/remotes.html which is usually accurate before buying them, and there they are listed, so I'm a bit puzzled?

## Screenshots

[image]
[image]

## Environment

 - Host system: Intel NUC running PVE, with Home Assistant
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26580700)
 - deCONZ version: (2.05.79)
 - Device: ConBee II
 - Do you use an USB extension cable: yes - 1m USB3

Smanar, 2020-09-19:
This device was included in deconz V 80 > https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_80

fribse, 2020-09-19:
Ok, so it will be supported soon for me :-) Thankyou very much for the precise answer @Smanar !

zray3dvn, 2020-09-23:
just updated the latest deconz 2.05.81 and firmware 26650700, but I cannot add the  Aqara D1 WXKG07LM.

Smanar, 2020-09-23:
Can you include it in deconz direclty ? without using phoscon ?

zray3dvn, 2020-09-24:
> Can you include it in deconz direclty ? without using phoscon ?

Honestly, I do not know how to include it in deconz directly, I always use Phoscon to add devices. Could you send me the guide to do that, thanks

fribse, 2020-09-24:
Ok, so the 2.05.81 is released for HA now, I'm going to re-test this.

@zray3dvn I've actually added it as a sensor in phoscon (though it doesn't show up there) but it does show up in deconz, and if you find it there, you can press the right-most dot on it, that will open the cluster list, then click on cluster info and click on basic and then press read for the values to be read.
That will show if it's communicating, then it's a question of moniting deconz_event in the developer tools events, and figuring out the values to react to.

zray3dvn, 2020-09-24:
update: I have 2 wireless switches, I've just added one by phoscon without any problem, I will try to change the battery to check the other tonight.

Nanohacs, 2020-09-26:
hello, I have WXKG07LM and WXKG06LM switches. WXKG07LM (2 gangs) works great but for WXKG06LM (1 gang) I can't use any button automation in home assitant It only show battery sensor. Do you have this problem too @zray3dvn ?

zray3dvn, 2020-09-27:
> hello, I have WXKG07LM and WXKG06LM switches. WXKG07LM (2 gangs) works great but for WXKG06LM (1 gang) I can't use any button automation in home assitant It only show battery sensor. Do you have this problem too @zray3dvn ?

sorry, I do not have the WXKG06LM. You should try to use the event tool like @fribse told above.

Smanar, 2020-09-27:
@Nanohacs the 1 gang is hte model is "lumi.remote.b186acn02" ?
Can you show the device JSON ?

fribse, 2020-09-27:
With the 2.05.81 I see the switches created correctly in Deconz/phoscon.

crabmanX, 2020-09-29:
I have a WXKG07LM. It is displayed in phoscon (Version: 2.05.82, Firmware 26580700), but i cannot add it to any groups. In home-assistant, the switch is displayed as "lumi.remote.b286acn02" but only has a battery entity. Listening to  deconz_event gives the following output when pressing both buttons consecutively:

`
{
    "event_type": "deconz_event",
    "data": {
        "id": "switch_8",
        "unique_id": "00:15:8d:00:04:4e:78:98",
        "event": 2002
    },
    "origin": "LOCAL",
    "time_fired": "2020-09-29T19:06:05.280966+00:00",
    "context": {
        "id": "d2f0f3ea028611ebba0b8d96c26d3987",
        "parent_id": null,
        "user_id": null
    }
}
{
    "event_type": "deconz_event",
    "data": {
        "id": "switch_8",
        "unique_id": "00:15:8d:00:04:4e:78:98",
        "event": 1002
    },
    "origin": "LOCAL",
    "time_fired": "2020-09-29T19:06:01.490364+00:00",
    "context": {
        "id": "d0ae8c7b028611eb8b44372052550605",
        "parent_id": null,
        "user_id": null
    }
}
`

Is this a bug or is the device not yet supported? Should i open a new issue?

Smanar, 2020-09-29:
Have you tried with the old web app ? in phoscon / help / old web app

crabmanX, 2020-09-29:
Yes, it is not displayed at all. Neither in devices nor in groups

fribse, 2020-10-04:
@crabmanX I'm sorry, but I don't understand what the issue is, the event data looks perfectly fine, you have the id and the event, that's what you need?

crabmanX, 2020-10-04:
@fribse : sorry, the problem is that while the switch is correctly listed in the phoscon switches UI, the REST api and home assistant, i cant add it to a group:

```
{
  ...
  "2": {
    "config": {
      "battery": 100,
      "group": "2",
      "on": true,
      "reachable": true
    },
    "ep": 2,
    "etag": "bb158a4a60ed1f0c3644a447eae89cb1",
    "lastseen": "2020-10-04T08:53Z",
    "manufacturername": "Philips",
    "mode": 1,
    "modelid": "RWL021",
    "name": "WZ_dimmer",
    "state": {
      "buttonevent": 4002,
      "eventduration": 2,
      "lastupdated": "2020-10-03T21:12:29.821"
    },
    "swversion": "5.45.1.17846",
    "type": "ZHASwitch",
    "uniqueid": "00:17:88:01:10:5e:5e:0c-02-fc00"
  },
  "3": {
    "config": {
      "battery": 62,
      "group": "4",
      "on": true,
      "reachable": true"5": {
    "config": {
      "battery": 100,
      "duration": 122,
      "on": true,
      "pending": [],
      "reachable": true
    },
    "ep": 1,
    "etag": "42cd350ba263eaf92a95aab0fe5b54ed",
    "lastseen": "2020-10-04T08:51Z",
    "manufacturername": "eWeLink",
    "modelid": "MS01",
    "name": "KU_Bewegungsmelder",
    "state": {
      "lastupdated": "2020-10-04T08:37:53.308",
      "lowbattery": false,
      "presence": false,
      "tampered": false
    },
    "swversion": "20191107",
    "type": "ZHAPresence",
    "uniqueid": "00:12:4b:00:22:28:b1:e4-01-0500"
  },
    },
    "ep": 2,
    "etag": "37d04d4effd10377798c1c9a3c8210c0",
    "lastseen": "2020-10-04T08:54Z",
    "manufacturername": "Philips",
    "mode": 1,
    "modelid": "RWL021",
    "name": "SZ_dimmer",
    "state": {
      "buttonevent": 1002,
      "eventduration": 1,
      "lastupdated": "2020-09-30T16:49:23.447"
    },
    "swversion": "6.1.1.28573",
    "type": "ZHASwitch",
    "uniqueid": "00:17:88:01:10:5e:5e:28-02-fc00"
  },
  ...
  "8": {
    "config": {
      "battery": 100,
      "on": true,
      "reachable": true,
      "temperature": 2700
    },
    "ep": 1,
    "etag": "ac6454ac575d7db6ec65f2badc210c82",
    "lastseen": "2020-10-04T08:22Z",
    "manufacturername": "LUMI",
    "mode": 1,
    "modelid": "lumi.remote.b286acn02",
    "name": "SZ_schalter_1",
    "state": {
      "buttonevent": 1002,
      "lastupdated": "2020-10-03T05:35:14.857"
    },
    "swversion": "20190329",
    "type": "ZHASwitch",
    "uniqueid": "00:15:8d:00:04:4e:78:98-01-0012"
  }
}
```

[image]

vs.:

[image]

fribse, 2020-10-04:
Ok, so a COMPLETELY different issue than the original, you should create a seperate issue for it. I doubt it will be seen here.

crabmanX, 2020-10-04:
@fribse OK, thanks. I created a new issue in the phoscon-app-beta repo: https://github.com/dresden-elektronik/phoscon-app-beta/issues/221 since this seems to be unrelated to the REST api.

closed 2020-10-05 by fribse