dresden-elektronik/deconz-rest-plugin #3283: Possible bug with the 0x26480700.bin.GCF firmware on conbee II

alekslyse, 2020-09-19:
I updated conbee II to deCONZ_ConBeeII_0x26480700.bin.GCF from 0x26580700, but it seems like it does not like unraid (built on slackware) with this firmware. Tried the stick in windows, with no issues, reflashed, same issue. Flashed the 0x26580700 and it works again in unraid. Seems like the OS cant identify the stick with 0x26480700.

Anyone experienced this?

[2061555.860531] usb 3-6: unable to read config index 0 descriptor/all
[2061555.860533] usb 3-6: can't read configurations, error -110
[2061555.860557] usb usb3-port6: unable to enumerate USB device

SwoopX, 2020-09-19:
Erm, why are you desperately trying to downgrade the firmware? And you know that your OS is not exactly supported?

alekslyse, 2020-09-19:
No issue downgrading, just saying the latest had issues. I think it’s important as unraid is used by a lot with usb passthrough to vm and docker.

SwoopX, 2020-09-19:
> I updated conbee II to deCONZ_ConBeeII_0x26480700.bin.GCF from 0x26580700

That is a downgrade!

> Flashed the 0x26580700 and it works again in unraid.

That's a fairly new firmware where some issues got ironed out, so not too surprising.

> Seems like the OS cant identify the stick with 0x26480700

Presumably because that firmware is almost 2 years old and improvements are not included. Basically, your stating that 0x26480700 is old. What should I say... Yes.

alekslyse, 2020-09-20:
Im not sure why I wrote that version number. I meant of course trying to upgrade from deCONZ_ConBeeII_0x26580700.bin.GCF to deCONZ_ConBeeII_0x26650700.bin.GCF Im just as confused why I wrote dead old firmware versions, but still. The 26580700 works, but 26650700 does not.

Mimiix, 2020-10-12:
@alekslyse Wasnt this fixed with the latest firmware?

alekslyse, 2020-10-14:
> @alekslyse Wasnt this fixed with the latest firmware?

Yes for me it was solved with the latest. No issues with the latest

Mimiix, 2020-10-15:
Closing this one!

closed 2020-10-15 by Mimiix