dresden-elektronik/deconz-rest-plugin #3286: Is is possible to run in the background on Windows?

mdolnik, 2020-09-20:
I could not find any information anywhere else regarding this.

I am wanting to run Deconz on my regular windows machine, but not have the GUI always open.

There are means to run as headless on linux, but I cannot seem to figure out how to run on Windows in the background.

Is there a command line argument or anything for this on Windows?

Smanar, 2020-09-20:
I don't think it s possible on deconz side.
But take a look on windows side, for exemple, on a shortcut propertie, you can configure it to be run with minimzed window.

What is exactly your problem with the GUI ?

mdolnik, 2020-09-20:
> What is exactly your problem with the GUI ?

No issues with the GUI, but I am wanting to use my main PC as my main zigbee hub, but do not want to have the deCONZ application always showing in the taskbar.

I have tried various ways to open up the deCONZ app, involving batch scripts with custom arguments and creating scheduled tasks:
I've tried a couple of tactics, mainly:
- https://superuser.com/a/1168592
- https://serverfault.com/questions/31608

The closest I got was with a scheduled task to run as the "system" user, which opened up deCONZ without the GUI, but the Phoscon app was only half responsive, I was able to navigate through various pages, but had no control of any lights. I'm assuming when running as the system user, the program does not have access to everything it normally does.

I've seen a couple of third-party programs that promise the ability to open GUI programs in the background, but I don't really want to go down that route.

Does anyone use their windows machine as their main host? Or is a Raspberry Pi the main way to go about this. I always have my PC on, so I don't want to spend the money on a Pi if I don't have to.

Mimiix, 2020-09-21:
@mdolnik How about running it on a other user account?

morfei1, 2020-09-21:
Its not even possible in linux (raspbian) for example, as far as I know.

Smanar, 2020-09-21:
I think the better way is using third app, I now you want to avoid this way, but some of them seem realy easy to use.

mdolnik, 2020-09-22:
> @mdolnik How about running it on a other user account?

I just tried this. It works if I manually log into the account, and open deCONZ, then switch back to my main account.

It does NOT work when I create a scheduled task to start deCONZ in that user's account on computer startup. When deCONZ starts up it seems to start in the same half-responsive way that I mentioned earlier, where Phoscon will load, but is unusable.

> I think the better way is using third app, I know you want to avoid this way, but some of them seem realy easy to use.
By this you mean a Raspberry Pi or similar dedicated computer for this sort of thing?

Regardless, I think I will have to bite the bullet and get a Raspberry Pi for running deCONZ on.

Thanks for your suggestions though.

closed 2020-09-22 by mdolnik

Mimiix, 2020-09-22:
@mdolnik That is expected: USer accounts work in such way that they start after they get signed into. 

Maybe a service?

Smanar, 2020-09-22:
>I think the better way is using third app, I know you want to avoid this way, but some of them seem realy easy to use.

No I want to say there is so much application on windows to hide an applications.

On my side I m using too a raspberry for domotic, (5 Euros electricity by years), but my "Media center" is on windows. If you realy like windows, there is some little machine that can host windows without problem.