dresden-elektronik/deconz-rest-plugin #3287: Sengled Smart Switch E1E-G7F

stevenwfoley, 2020-09-20:
## Describe the question or issue you are having
Hello,
I have a Sengled E1E-G7F Smart Switch that is not working in deconz v2.05.82. The node shows up in the deconz gui app, but is not reading/populating the clusters. The dot for the drop-down cluster menu is not even showing on the node object in the gui.

## Screenshots
[image]

## Environment
 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26370500
 - deCONZ version: 2.05.82
 - Device: RaspBee I

Mimiix, 2020-09-20:
Hi,

Seems like it's not paired correctly. 

You should re pair a few times.

stevenwfoley, 2020-09-20:
I've must have tried (re)pairing the device like a dozen times, but I guess it just needed that one last 13th repairing attempt. Its now showing up in deconz gui properly but not in phoscon. Is this suppose to be a supported device? Should I open a device request issue?

Mimiix, 2020-09-21:
@stevenwfoley 

Phoscon is a different application. If the device works fine with the rest-api / ha /homebridge, you need to request support in the Phoscon repo.

SwoopX, 2020-09-21:
I cannot recall this devices being supported...

stevenwfoley, 2020-09-22:
Thanks all, I'll submit a new device request.

closed 2020-09-22 by stevenwfoley