dresden-elektronik/deconz-rest-plugin #3288: Permanently different port even after update

cjk2, 2020-09-20:
Hello,
it is no problem to change the default port from 80 to 81. Even after a restart everything is fine. But after a firmware update the configuration is changed back to port 80 (in deconz.service).
Is it possible (and how) to change the port permanently - even after an update?

Smanar, 2020-09-20:
You can set it in the command line or edit a configuration file
CF https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1699 for exemple

cjk2, 2020-09-20:
I know that you can change the port in the configuration. This goes so far. But after an update from Dresden Elektronik (last on 14.9., version 2.05.81) this change will be reset to port 80 in my case!
Maybe the whole thing would be much easier if you could set the port in the GUI ;-)

Smanar, 2020-09-20:
The update have changed the configuration file ? Else the only reason for deconz use another port is the port is already used.

cjk2, 2020-09-20:
Yes, indeed. It's really amazing. A new start is possible. But as soon as an update is installed ... Already Port 80 is in there again. 
And the port (81 for deCONZ in my case) is not in use otherwise.  But port 80 is actually used by OpenHub.

Smanar, 2020-09-20:
You are using the config file edition to change the port ?
So you are editing the deconz-gui.service file or using  a override.conf file ? I can understand the first is modified, but the second ....

SwoopX, 2020-09-20:
You need to use drop in files for systemd. They can basically overwrite defined setting to ensure persistence, even on upgrade.

cjk2, 2020-09-20:
@SwoopX 
drop in files? What do you meen?

@Smanar 
Yes! I edited the deconz.service file. Or weich file do you meen with „override.conf“

SwoopX, 2020-09-20:
@cjk2 google is your friend here.

cjk2, 2020-09-21:
@SwoopX 
Thank you. Google is 1. not my friend, but a search engine and 2. I was hoping that someone might be able to tell me exactly how to do that. 
Sorry for this assumption. Not everybody has Google as a friend and cuddles with Linux ;-)

Smanar, 2020-09-21:
This issue is realy old but more explicit https://github.com/dresden-elektronik/phoscon-app-beta/issues/26

user897943, 2021-02-16:
This is extremely aggravating because my reverse proxy runs on port 80 and 443, so having deconz "steal" port 80 is absolutely infuriating, everything stops working and I have dozens of services the most important my home automation, my lights stop working, my alarm, my heating and ventilation, my doorbell, my cameras, my sensors, both my door locks which cannot be used with keys, so basically deconz turns off my entire house because it thinks it can override what I say and change back to port 80. Disgusting to be honest, if I was the programmer I'd be ashamed. I should be able to permanently say what port to use and it should obey my setting forever, without making me hack into Linux.

Backes89, 2021-11-19:
+1

It's very unmotivating if you have to change the port every then and now because there was an update. Can't there be just an option at install oder in the web-interface to change the port?

That the configuration file get's overwritten is a bug in my opinion and needs to be adressed. Why am I creating this change in the configuration if it get's overwritten?

Smanar, 2021-11-19:
But you are using override file ?

msandres13, 2022-06-04:
+1 also did run into this problem and my deconz server failed while I was on vacation due to an automatic update.
I did not expect that the service file is simply overwritten.

I guess a good solution would be adding a comment in the services file which says:
DO NOT EDIT this file it gets overwritten during updates.
Instead do the following
sudo mkdir -p /etc/systemd/system/deconz.service.d/
sudo cp -R  /usr/lib/systemd/system/deconz-gui.service /etc/systemd/system/deconz.service.d/override.conf
sudo vi /etc/systemd/system/deconz.service.d/override.conf

karlquint, 2023-01-11:
+1 
Had this issue after the latest firmware update. Quickly figured out the port was reset to 80 because another service using port 80 didn't work anymore. The user chosen port should remain the same even after an update.

Smanar, 2023-01-12:
I m right too.
IDK what is better, make a wiki page or make a  "deconz suggest feature" on forum ?