dresden-elektronik/deconz-rest-plugin #3289: Iluminize wall switch 511.524 (same as Sunricher)

PGU79, 2020-09-20:
## Device

- Product name: 4 Group CCT Zigbee Touch Remote controller
- Manufacturer: Ilulminize
- Model identifier: ZGRC-TEUR-003
- Device type : Switch

## Screenshots

[image]
[image]
[image]

PGU79, 2020-09-20:
deCONZ is 2.05.81 with conbee II.

PGU79, 2020-09-26:
I tried to grab some debug info running deCONZ with this option :  -dbg-aps=2 . Below details of what I got when pressing touch buttons : 

**Scene buttons (short press)**

> 13:46:43:031    asdu: 013705000001
> 13:47:24:375 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 95, rssi: -82
> 13:47:24:376    asdu: 013805000002
> 13:47:24:385 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 95, rssi: -82
> 13:47:24:392    asdu: 013805000002
> 13:47:24:408 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 95, rssi: -82
> 13:47:24:408    asdu: 013805000002
> 13:47:26:100 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 79, rssi: -84
> 13:47:26:101    asdu: 013905000003
> 13:47:26:133 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 79, rssi: -84
> 13:47:26:133    asdu: 013905000003
> 13:47:26:443 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 87, rssi: -83
> 13:47:26:444    asdu: 013905000003
> 13:47:26:450 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 87, rssi: -83
> 13:47:26:450    asdu: 013905000003
> 13:47:27:118 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 87, rssi: -83
> 13:47:27:118    asdu: 013a05000004
> 13:47:27:125 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 87, rssi: -83
> 13:47:27:126    asdu: 013a05000004
> 13:47:27:136 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 87, rssi: -83
> 13:47:27:136    asdu: 013a05000004
> 13:47:27:476 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 95, rssi: -82
> 13:47:27:477    asdu: 013a05000004
> 

**Top buttons with long press (corresponding to groups when linked manually to zigbee lamps)** 

> 3:50:02:284    asdu: 014000
> 13:50:04:201 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x03 dstAddrMode: 1, profile: 0x0104, cluster: 0x0006, lqi: 87, rssi: -83
> 13:50:04:202    asdu: 014101
> 13:50:06:272 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0006, lqi: 79, rssi: -84
> 13:50:06:273    asdu: 014201
> 13:50:07:481 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8038, lqi: 95, rssi: -82

**Dimming touch buttons**

> 3:51:58:236    asdu: 014c0600200000
> 13:51:58:589 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 95, rssi: -82
> 13:51:58:590    asdu: 014d0600200000
> 13:51:59:877 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 79, rssi: -84
> 13:51:59:879    asdu: 014e0600200000
> 13:52:02:417 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 95, rssi: -82

**Warm to Cold light touchpad**

> 3:52:08:116    asdu: 01520abf0008000000
> 13:52:30:146 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0300, lqi: 87, rssi: -83
> 13:52:30:148    asdu: 01530abf0000000000
> 13:52:30:930 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0300, lqi: 58, rssi: -83
> 13:52:30:930    asdu: 01540a2f0108000000
> 13:52:32:652 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0300, lqi: 95, rssi: -82
> 13:52:32:652    asdu: 01570ac20100000000
> 13:52:32:864 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0300, lqi: 95, rssi: -82
> 13:52:32:864    asdu: 01560ac20108000000
> 13:52:32:999 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0300, lqi: 87, rssi: -83
> 13:52:32:000    asdu: 01580ac60008000000
> 13:52:33:426 APS-DATA.indication srcAddr: 0xf3af, srcEp: 0x04 dstAddrMode: 1, profile: 0x0104, cluster: 0x0300, lqi: 95, rssi: -82
> 
>

PGU79, 2020-09-26:
I don't what to do more to help ?

SwoopX, 2020-10-24:
Can you please provide screenshots so that all available endpoints are visible? Some of them are cut of in the given ones above. Thanks!

PGU79, 2020-10-24:
Hello SwoopX

Here are some screenshots. Hope it is OK for you.

[image]
[image]

I will collect information asap.

PGU79, 2020-10-25:
I don't see Info message when I press a button. I may be because the switch is already known by deCONZ Gateway (I did some bindings using deCONZ-gui. I will try to reset the switch and test again).

PGU79, 2020-10-25:
I reset the switch, blue was off whatever area (button) I Pressed. I then attach to deCONZ again. Blue led is on when i touch button area but i still can't see any INFO message from sdout.

Switch is working fine after bindings are set using deconz-gui. ON/OFF, "Level Control", and "Color" (warm/cold). Even group bindings are OK.

SwoopX, 2020-10-25:
Are you on 2.05.86?

PGU79, 2020-10-25:
Yes version is 2.05.86

You are right it is not too urgent as I learned hoot manually link switch to groups and all touch buttons are OK.

It may be difficult for people not used to REST API to find group Ids and use bindings in deCONZ gui. Using Phoscon APP would be much easier, REST API would a bonus for other apps (jeedom, homebridge, ...)

PGU79, 2020-10-25:
[image]

SwoopX, 2020-10-25:
Ah, I see. Well, good to see that you got it working "the technical way". However, that should all work automatically once support.

Will add some code within the next few days. Would you have the chance to compile and test it?

PGU79, 2020-10-25:
OK I can try. 

I only run deCONZ on a very small VM running on an old mac mini. I miss development packages. Will check what I can do this week.

closed 2020-10-25 by PGU79

PGU79, 2020-10-31:
Hello Swoopx

Sorry the delay. I had no internet access this week. Fiber was cut somewhere !

I was able to checkout version provided and compile it (there was a bug in de_web_plugin.cpp, missing || preceding line 
sensor->modelId().startsWith(QLatin1String("ZGRC-TEUR-")))         // iluminize wall switch 511.524)

Here is INFO lines output when button are pressed.

It seems Button labels 1 to 4 are not sent on short press. They change other buttons target (on/off, level control....).

[debug_INFO_SunRicher.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5468891/debug_INFO_SunRicher.txt)

SwoopX, 2020-10-31:
Cool, thanks. I'll prepare a map (in case we don't already have a fitting one) so you can try it on your end then.

SwoopX, 2020-11-08:
Sorry for the late feedback. Unfortunately, that info is a little bit random and incomplete.

So, I understand button 1-4 are just for selecting the corresponding endpoint. Apart from that, I canot know how you pressed the remaining buttons.

I'd assume the following:
- A short press on the power button sends on, the next one sends off. Would that be correct and what about a long or double press?
- For the level control buttons (left and right), I do not know how you pressed them short, long, double
- Pretty much same for the sliders: did you move down or up?
- What makes me wonder for the scenes buttons is that I only see a `recall scene` command, but never a `store scene`. Typically, the latter is a long press

I'm afraid you need to note down what you exactly did and then add the commands seen in the debug log to it to be able to create a reasonable map here. Maybe you can give it another try when you got some time. Also, doing it for one endpoint (buttons 1-4) consistenly would be enough.

PGU79, 2020-11-09:
It's OK for me as switch is working fine after manual binding of groups of lights using deCONZ gui.

I will give another try with better debug info sometime this week.

In the mean time, here is  how I use it :

- short press on power button can be used to power on/off all group of lights known by the switch.
- long press to button 1 to 4 can be used to power on/off lights bound to each button (using power on/off, level control, color control ...  bindings)
- short press to button 1 to 4 "select" group of lights : then "dimming up/down" or "warm/cold white" areas change levels for corresponding lights or group of lights.
- short press / long press work as expected to dim up or down or to change white temperature
- sliding on cold/white area seems to change white temperature while your finger move on the area
- It seems double press is doing nothing more than 2 short press.
- scene buttons ; you are right long press store scene

PGU79, 2020-11-11:
Another try.

- Button 1 short press : no info

- Button 1 long press  twice.

> 12:04:49:878 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: ONOFF (0x0006) command: OFF (0x00) payload[0]: 000
> 12:04:54:269 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: ONOFF (0x0006) command: ON (0x01) payload[0]: 000
> 
> 

- short press level contro, left button first : reduce level , right button : raise level

> 10:48:03:026 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STEP_WITH_ON_OFF (0x06) payload[0]: 001
> 10:48:09:079 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STEP_WITH_ON_OFF (0x06) payload[0]: 000
> 

- long press level control : 1st message when finger stay pressed, 2nd message when finger is removed. Again left button first

> 10:48:47:531 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_WITH_ON_OFF (0x05) payload[0]: 001
> 10:48:49:969 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STOP_WITH_ON_OFF (0x07) payload[0]: 000

> 10:48:55:523 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_WITH_ON_OFF (0x05) payload[0]: 000
> 10:48:57:173 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STOP_WITH_ON_OFF (0x07) payload[0]: 000
> 

- short press color area (warm/cold temperature) multiple press on the whole area. 2 messages for each press ?

> 10:49:21:487 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0A3
> 10:49:21:727 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0A3
> 10:49:23:315 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 03D
> 10:49:23:539 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 039
> 10:49:24:253 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0CA
> 10:49:24:495 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0CA

- sliding color area 

> No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 09E
> 11:00:20:069 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0A9
> 11:00:20:908 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0BB
> 11:00:21:739 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0D0
> 11:00:22:579 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0EA
> 11:00:22:988 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0FE
> 11:00:24:055 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 00F
> 11:00:24:899 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 03D
> 11:00:25:724 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 086
> 11:00:26:559 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0AF
> 11:00:27:089 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0B8

- long press color area : sending same message while finger stays pressed

> 11:01:12:179 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0BF
> 11:01:13:018 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0BF
> 11:01:13:855 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0BF
> 11:01:14:694 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0BF
> 11:01:15:518 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: MOVE_TO_COLOR_TEMPERATURE (0x0A) payload[0]: 0BF
[... cut ...]
> 10:56:39:398 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x01 cluster: SCENES (0x0005) command: STORE_SCENE (0x04) payload[0]: 000
> 
> 

On tests run last week It seems storing scenes using scene buttons messed bindings and removed lights from groups bound to buttons 1-4. ATM, these buttons are useless as I don't know how to use scenes defined using phoscon app.

BTW I have short documentation from Sunricher (English) and Illuminize (german) if needed. 

One last question : I know how to set bindings but not how to read them, how do I get bindings setup using deconz gui ?

SwoopX, 2020-11-11:
Thanks for that. Just having a glimspe here real quick and it's looking promising...

Will have a closer one in the evening.

SwoopX, 2020-11-12:
Alright, so if I've put the pieces together correctly, the button map for the first endpoint should look like this:

```
"iluminizeMap": {
    "modelids": ["ZGRC-TEUR-003"],
    "map": [
        [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
        [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
        [1, "0x01", "LEVEL_CONTROL", "STEP_WITH_ON_OFF", "1", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Step up (with on/off)"],
        [1, "0x01", "LEVEL_CONTROL", "STEP_WITH_ON_OFF", "0", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Step down (with on/off)"],
        [1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "1", "S_BUTTON_3", "S_BUTTON_ACTION_HOLD", "Dimm up (with on/off)"],
        [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_3", "S_BUTTON_ACTION_LONG_RELEASED", "Dimm up stop"],
        [1, "0x01", "LEVEL_CONTROL", "MOVE_WITH_ON_OFF", "0", "S_BUTTON_3", "S_BUTTON_ACTION_HOLD", "Dimm down (with on/off)"],
        [1, "0x01", "LEVEL_CONTROL", "STOP_WITH_ON_OFF", "0", "S_BUTTON_4", "S_BUTTON_ACTION_LONG_RELEASED", "Dimm down stop"],
        [1, "0x01", "COLOR_CONTROL", "MOVE_TO_COLOR_TEMPERATURE", "0", "S_BUTTON_5", "S_BUTTON_ACTION_SHORT_RELEASED", "Move to color temperature"],
        [1, "0x01", "SCENES", "RECALL_SCENE", "0", "S_BUTTON_6", "S_BUTTON_ACTION_SHORT_RELEASED", "Recall scene 1"],
        [1, "0x01", "SCENES", "STORE_SCENE ", "0", "S_BUTTON_6", "S_BUTTON_ACTION_SHORT_RELEASED", "Store scene 1"]
    ]
},
```

It would also require a small code change to interpret the dim events right. I'll see that I get it into the next beta. Would appreciate if you could test it then and share your feedback.

PGU79, 2020-11-22:
I have downloaded latest beta to test how switch is seen. Not sure what to do, I reset the switch following manufacturer instruction, opened network using phoscon app. rest API returns 4 new switch. "Hue essentials" app only sees one (id 10)

    "9": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "17ca09a329e5db18a64bdc6c1d477cb1",
        "lastseen": "2020-11-22T16:01Z",
        "manufacturername": "iluminize",
        "mode": 1,
        "modelid": "ZGRC-TEUR-003",
        "name": "ZGRC-TEUR-003 9",
        "state": {
            "buttonevent": null,
            "lastupdated": "none"
        },
        "swversion": "2.3.1_r30",
        "type": "ZHASwitch",
        "uniqueid": "00:0d:6f:00:13:f0:07:36-01-1000"
    },
    "10": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 2,
        "etag": "17ca09a329e5db18a64bdc6c1d477cb1",
        "lastseen": "2020-11-22T16:01Z",
        "manufacturername": "iluminize",
        "mode": 1,
        "modelid": "ZGRC-TEUR-003",
        "name": "ZGRC-TEUR-003 10",
        "state": {
            "buttonevent": null,
            "lastupdated": "none"
        },
        "swversion": "2.3.1_r30",
        "type": "ZHASwitch",
        "uniqueid": "00:0d:6f:00:13:f0:07:36-02-1000"
    },
    "11": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 3,
        "etag": "17ca09a329e5db18a64bdc6c1d477cb1",
        "lastseen": "2020-11-22T16:01Z",
        "manufacturername": "iluminize",
        "mode": 1,
        "modelid": "ZGRC-TEUR-003",
        "name": "ZGRC-TEUR-003 11",
        "state": {
            "buttonevent": null,
            "lastupdated": "none"
        },
        "swversion": "2.3.1_r30",
        "type": "ZHASwitch",
        "uniqueid": "00:0d:6f:00:13:f0:07:36-03-1000"
    },
    "12": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 4,
        "etag": "17ca09a329e5db18a64bdc6c1d477cb1",
        "lastseen": "2020-11-22T16:01Z",
        "manufacturername": "iluminize",
        "mode": 1,
        "modelid": "ZGRC-TEUR-003",
        "name": "ZGRC-TEUR-003 12",
        "state": {
            "buttonevent": null,
            "lastupdated": "none"
        },
        "swversion": "2.3.1_r30",
        "type": "ZHASwitch",
        "uniqueid": "00:0d:6f:00:13:f0:07:36-04-1000"
    }

PGU79, 2020-11-22:
Concerning INFO in logfile 

> - Button 1 short press then ON/OFF short press
> 
> 7:22:31:705 [INFO] - Button 2002 Off ZGRC-TEUR-003
> 17:22:31:729 [INFO] - Button 2002 Off ZGRC-TEUR-003
> 17:22:31:730 [INFO] - Button 2002 Off, discard too fast event (dt = 21) ZGRC-TEUR-003
> 17:22:31:750 [INFO] - Button 2002 Off ZGRC-TEUR-003
> 17:22:31:751 [INFO] - Button 2002 Off, discard too fast event (dt = 42) ZGRC-TEUR-003
> 17:22:31:764 [INFO] - Button 2002 Off ZGRC-TEUR-003
> 17:22:31:766 [INFO] - Button 2002 Off, discard too fast event (dt = 57) ZGRC-TEUR-003
> 
> - Button 2 short press then ON/OFF short press
> 
> 7:23:16:128 [INFO] - Button 7002 On ZGRC-TEUR-003
> 17:23:16:160 [INFO] - Button 7002 On ZGRC-TEUR-003
> 17:23:16:160 [INFO] - Button 7002 On, discard too fast event (dt = 30) ZGRC-TEUR-003
> 17:23:16:183 [INFO] - Button 7002 On ZGRC-TEUR-003
> 17:23:16:184 [INFO] - Button 7002 On, discard too fast event (dt = 54) ZGRC-TEUR-003
> 17:23:16:210 [INFO] - Button 7002 On ZGRC-TEUR-003
> 17:23:16:211 [INFO] - Button 7002 On, discard too fast event (dt = 81) ZGRC-TEUR-003
> 
> - Button 3 short press then ON/OFF short press
> 
> 17:23:22:688 [INFO] - Button 14002 Off ZGRC-TEUR-003
> 17:23:22:713 [INFO] - Button 14002 Off ZGRC-TEUR-003
> 17:23:22:717 [INFO] - Button 14002 Off, discard too fast event (dt = 23) ZGRC-TEUR-003
> 17:23:22:749 [INFO] - Button 14002 Off ZGRC-TEUR-003
> 17:23:22:749 [INFO] - Button 14002 Off, discard too fast event (dt = 55) ZGRC-TEUR-003
> 17:23:22:769 [INFO] - Button 14002 Off ZGRC-TEUR-003
> 17:23:22:769 [INFO] - Button 14002 Off, discard too fast event (dt = 75) ZGRC-TEUR-003
> 
> - Button 4 short press then ON/OFF short press
> 
> 17:23:27:522 [INFO] - Button 19002 On ZGRC-TEUR-003
> 17:23:27:534 [INFO] - Button 19002 On ZGRC-TEUR-003
> 17:23:27:542 [INFO] - Button 19002 On, discard too fast event (dt = 19) ZGRC-TEUR-003
> 17:23:27:559 [INFO] - Button 19002 On ZGRC-TEUR-003
> 17:23:27:560 [INFO] - Button 19002 On, discard too fast event (dt = 37) ZGRC-TEUR-003
> 17:23:28:020 [INFO] - Button 19002 On ZGRC-TEUR-003
> 17:23:28:021 [INFO] - Button 19002 On, discard too fast event (dt = 498) ZGRC-TEUR-003
> 
> 
> - Button 1 long press twice
> 
> 17:24:10:217 [INFO] - Button 2002 Off ZGRC-TEUR-003
> 17:24:14:451 [INFO] - Button 1002 On ZGRC-TEUR-003
> 
> - Button 2 long press twice
> 
> 17:24:50:120 [INFO] - Button 8002 Off ZGRC-TEUR-003
> 17:24:58:259 [INFO] - Button 7002 On ZGRC-TEUR-003
> 
> - Button 3 long press twice
> 
> 17:25:01:966 [INFO] - Button 14002 Off ZGRC-TEUR-003
> 17:25:09:154 [INFO] - Button 13002 On ZGRC-TEUR-003
> 
> - Button 4 long press twice
> 
> 17:25:05:689 [INFO] - Button 20002 Off ZGRC-TEUR-003
> 17:25:14:425 [INFO] - Button 19002 On ZGRC-TEUR-003
> 
> 
> - scene buttons 1 to 4 : seems to log same info (endpoint 0x04 as button 4 was last pressed)
> 
> 7:26:47:532 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x04 cluster: SCENES (0x0005) command: RECALL_SCENE (0x05) payload[0]: 000
> 17:26:47:548 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x04 cluster: SCENES (0x0005) command: RECALL_SCENE (0x05) payload[0]: 000
> 17:26:47:560 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x04 cluster: SCENES (0x0005) command: RECALL_SCENE (0x05) payload[0]: 000
> 17:26:47:579 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x04 cluster: SCENES (0x0005) command: RECALL_SCENE (0x05) payload[0]: 000
[... 46 lines cut ...]
> - touching warm/cold area (button 1 was pressed before)
> 
> 17:30:04:956 [INFO] - Button 5002 Move to color temperature ZGRC-TEUR-003
> 17:30:05:193 [INFO] - Button 5002 Move to color temperature ZGRC-TEUR-003
> 17:30:05:195 [INFO] - Button 5002 Move to color temperature, discard too fast event (dt = 236) ZGRC-TEUR-003
> 17:30:07:963 [INFO] - Button 5002 Move to color temperature ZGRC-TEUR-003
> 17:30:08:199 [INFO] - Button 5002 Move to color temperature ZGRC-TEUR-003
> 17:30:08:200 [INFO] - Button 5002 Move to color temperature, discard too fast event (dt = 231) ZGRC-TEUR-003
> 17:30:12:353 [INFO] - Button 5002 Move to color temperature ZGRC-TEUR-003
> 17:30:12:590 [INFO] - Button 5002 Move to color temperature ZGRC-TEUR-003
> 17:30:12:592 [INFO] - Button 5002 Move to color temperature, discard too fast event (dt = 233) ZGRC-TEUR-003
> 17:30:13:903 [INFO] - Button 5002 Move to color temperature ZGRC-TEUR-003
> 17:30:14:130 [INFO] - Button 5002 Move to color temperature ZGRC-TEUR-003
> 17:30:14:130 [INFO] - Button 5002 Move to color temperature, discard too fast event (dt = 222) ZGRC-TEUR-003
> 17:30:14:726 [INFO] - Button 5002 Move to color temperature ZGRC-TEUR-003
> 17:30:14:958 [INFO] - Button 5002 Move to color temperature ZGRC-TEUR-003
> 17:30:14:959 [INFO] - Button 5002 Move to color temperature, discard too fast event (dt = 229) ZGRC-TEUR-003
> 
> - one long press on scene button 
> 
> 17:34:34:081 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x02 cluster: SCENES (0x0005) command: STORE_SCENE (0x04) payload[0]: 000
> 17:34:34:100 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x02 cluster: SCENES (0x0005) command: STORE_SCENE (0x04) payload[0]: 000
> 17:34:34:115 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x02 cluster: SCENES (0x0005) command: STORE_SCENE (0x04) payload[0]: 000
> 17:34:34:125 [INFO] - No button handler for: ZGRC-TEUR-003 endpoint: 0x02 cluster: SCENES (0x0005) command: STORE_SCENE (0x04) payload[0]: 000
>