dresden-elektronik/deconz-rest-plugin #3291: Sonoff BASICZBR3 not switching when used in scenes

etique57, 2020-09-21:
## Describe the bug
I have one Sonoff BASICZBR3 switch. I added it to Deconz, then Phoscon using the trick that is described in one issue. So far so good.

Using it with the "toggle" command from my wall switch it works perfectly.

But if I use it in a scene, the behavior is the following
- the switch appears to be "on" (that's how the REST sees it I believe), in Phoscon, Hassio, etc.
- but the relay is actually not switched on (so the light is not on)

## Steps to reproduce the behavior
1. add SONOFF BASICZBR3 to Deconz & then Phoscon
2. Assign BASICZBR3 to a room
3. Create a new scene ON
4. Select the BASICZBR3 and set it to 100%
5. Create a new scene OFF
6. Select the BASICZBR3 and set it to 0%
7. Save
8. Activate the scene ON, then OFF

BUG: the BASICZBR3 is not switching accordingly, but the state appears to be fine.

## Expected behavior
The BASICZBR3 should switch properly when the scene is activated.

## Screenshots
N/A

## Environment

 - Host system: Tinkerboard
 - Running method: Marthoc Docker container
 - Firmware version: 260B0500
 - deCONZ version: 2.05.82
 - Device: ConBee I 
 - Do you use an USB extension cable: no

### deCONZ Logs
None

## Additional context
N/A

morfei1, 2020-09-21:
I think the problem is more related to the phoscon git than to the deCONZ git.

Why I think that....

When I first added my basiczbr3, I have to use the trick the switch to appear in the phoscon.
Then, if I manually switch it on in the phoscon it started to on/off/on/off/on/off......every second and it never ends.
If do the same but in HA the switch is working OK.

Now its working with nodered automation switch on and off based on my bathroom humidity and it does not have problems.
So, I think the rest-api is OK, because its ok in HA and NodeRed, but something is breaking it in phoscon.... maybe thats why we need a trick to make the switch appear in the phoscon.... the integration is not fully compleated or who knows....

etique57, 2020-09-21:
@morfei1 
You're probably right, I should have opened it to Phoscon. Maybe an admin can move it there?

I'm thinking about something related to the scene setting. Maybe it shows "100%" while it's actually "99.9%" or such and in a On-Off switch it decides to stay at Off?

Thanks for chiming in anyway!

morfei1, 2020-09-21:
@Mimiix can you move the issue in the phoscon repo? Thanks :) !

Mimiix, 2020-09-21:
_Your wish is my command!_