dresden-elektronik/deconz-rest-plugin #3292: Aqara Opple Switch integration - after join all lights are operate by switch when pressing a button

easybeat, 2020-09-21:
## Describe the bug
I received an Aqara Opple Switch 6-Button and after some problems I was able to join it in phoscon app. It is showing correctly but when I press one off the top buttons ALL lights on my system switch on or off! I did check if there was a group with all lights assigned to the button but no lights or group is assigned to the switch as I can see in phoscon.

## Steps to reproduce the behavior
1. Connect Aqara Opple switch using phoscon app to deconz
2. Switch is showing correctly in Switch list in phoscon app
3. press top left button on switch
4. All lights turning on or off

## Expected behavior
No lights should turn on or off without adding lights to the switch

## Screenshots
no screenshots

## Environment
Deconz version: 2.5.82
Host system:  Raspberry Pi
Running method: Raspbian
 Firmware version: 26370500
 deCONZ version: (2.5.82)
 Device: RaspBee I
 Do you use an USB extension cable: no

### deCONZ Logs
will deliver logs when necessary

## Additional context
--

Mimiix, 2020-09-21:
**Moderation note: Thanks for opening another issue, this one seems right ;)**

I had this too. However, this seems to be a multicast from the device and opple specific. After i paired it, i never had it again.

It scared me, as it turned everything off LOL

easybeat, 2020-09-21:
It didn't go away for me after I paired it.

easybeat, 2020-09-21:
Do you mean delete the switch and pair it again?

I will try again later tonight.

Smanar, 2020-09-21:
I don't thing deleting is usefull, I think it s better don't delete it, in case of deconz use old value.

easybeat, 2020-09-21:
Hi

Ok. I removed the switch in phoscon first. Then joined the switch again and immediately pressed button 1 (top left). All lights went off but from that moment no buttons didn't switch on ligths again. Also Button 1 did not turn off anymore lights. So this did work!!

After that I did the same with an Opple 4 Button and it did work on first time.

So I imagine is to press top left Button immediately after the switch shows fully joined in phoscon.

I did also check in ioBroker as I use deconz in there and all button events from both switches came through properly!

So not sure if you someone is going to do some more investigation or if we just close this bug report?

Anyway thank you very much for your help.

Kind regards

Beat

Mimiix, 2020-09-22:
I think this is intended behaviour by design of the switch. 

I'll just close it, if more comes in we can always re-open.

closed 2020-09-22 by Mimiix

helipus, 2020-12-29:
Hi @Mimiix,
I'm having exactly the same problem. But the solution is not working for me.
I tried:
1. deleting the switch in the Phoscon app
2. pressing the reset button 13s
3. start to add a new switch device
4. the Philipps lights blink
5. The Phoscon App says the switch is connected
6. pressing the top left button once short

I tried step 6 also after step 4.

Both versions are not working. The switch keeps controlling all lights as described by @easybeat.

helipus, 2020-12-29:
Ok the solution is to press the reset button short directly after the switch is connected. :-)