dresden-elektronik/deconz-rest-plugin #3293: Conbee 2 working for months started to bootloop

artkrz, 2020-09-21:
## Describe the question or issue you are having

Last week I moved to z2m from deconz, it was all good and sudenly after HA restart without any other apparent reason (i did restart the VM, and HassOS multiple times durning this period) it started to bootloop.

I tried to flash 2 latst fw couple of times but it does not help. I also connected it to other rpi directly without any of above and it's same. 

## Screenshots

Flashed before creating this issue:
[image]

and logs immediately after plugging it directly to rpi that never had the stick connected before:

[image]

## Environment
 - Host system: Raspberry Pi 
 - Running method: rpi with usbipd => proxmox => hassos 
 - Firmware version: deCONZ_ConBeeII_0x26650700.bin.GCF and deCONZ_ConBeeII_0x26580700.bin.GCF
 - deCONZ version: (2.xx.yy)
 - Device: ConBee II
 - Do you use an USB extension cable: yes

SwoopX, 2020-09-21:
The MTP probes shouldn't be there. Do you have this file `/etc/udev/rules.d/69-conbee.rules` and if yes, what's its content?

artkrz, 2020-09-21:
I don't have it on the test pi but on the system that issue occurred first I have this:

```
cat /etc/udev/rules.d/88-conbeeii.rules
# ConBee II usb stick rules
KERNEL=="ttyACM[0-9]*", SUBSYSTEM=="tty", ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", ENV{MTP_NO_PROBE}="1", SYMLINK="ttyCONBEEII"
```

SwoopX, 2020-09-21:
Hm, when you install deconz on a bare Raspbian, it looks like this:

```
# ConBee II disbale modem manager and mtp probing
ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", ENV{ID_MM_DEVICE_IGNORE}="1"
ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", ENV{MTP_NO_PROBE}="1"
# KERNEL=="ttyACM?", SUBSYSTEMS=="usb", DRIVERS=="usb", ACTION=="add", ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", SYMLINK+="ttyConBee2", GROUP="dialout", MODE="0660"
```

No clue where this comes from and also not too sure if MTP supression works on symlinks.

artkrz, 2020-09-21:
It helped to get rid off MTP but still bootlooping.

[image]

SwoopX, 2020-09-21:
Can you try that with the 0x26650700?

Was your setup previously running and if yes, since how long?

artkrz, 2020-09-21:
I bought Conbee II in Jan 2020. I was using it with Deconz. Last weekend I migrated to zigbee2mqtt due to TRVs that aren't supported in Deconz. As described I used usbipd between rpi and my proxmox machine that is located in a rack and can't have it connected directly. This setup was working fine till today. I re-paired some last sensors, rebooted VM and that's about it.

Btw, 0x26650700 disconnects as well.

Mimiix, 2020-09-22:
Bootloops are mostly caused due to broken firmware.  Please note that you did use Powershell and not CMD.

Proxmox seem to cause issues lately, no clue why.

artkrz, 2020-09-22:
@Mimiix Yeah, Proxmox causes problems but this also happens on vanilla Rasbian. I'll flash it from CMD and report back.

artkrz, 2020-09-22:
CMD makes no difference here:

[image]

Mimiix, 2020-09-22:
So wait, does it still bootloop on a RPI3B+?

artkrz, 2020-09-22:
Yes, same behaviour.

Mimiix, 2020-09-22:
What PSU does the RPi use?

artkrz, 2020-09-22:
It's powered by official Raspberry Pi PSU 5.1V, 2.5A. Why would that matter if it was powered by the same PSU for months?

Mimiix, 2020-09-22:
If it doesn't get enough power, this could happen too .

Can you try with a diff PSU?

Also, does it work properly on the windows PC (Native, no proxmox or any other VM involved?)

artkrz, 2020-09-22:
Tried with Anker PSU, same. On my PC I seem like it has same problem as I hear usb connect/disconnect sound every minute or so. Not sure how else I can monitor this as I'm not familiar with Windows that much.

artkrz, 2020-09-22:
I also tried flashing from Rasbian:

```
pi@raspberrypi:~ $ GCFFlasher_internal -d /dev/ttyACM0 -t 60 -f deCONZ_ConBeeII_0x26650700.bin.GCF -x 3
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.utf8)
GCFFlasher V3_15 (c) dresden elektronik ingenieurtechnik gmbh
14:08:01:857 using firmware file: deCONZ_ConBeeII_0x26650700.bin.GCF
14:08:02:216 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn:
14:08:02:219 ls dev: /dev/ttyACM0 (0x1CF1/0x0030) sn: DE2147999
Reboot device /dev/ttyACM0 (ConBee II)
14:08:03:252 query bootloader v1 ID after 1006 ms
14:08:03:754 close serial after 1508
14:08:04:766 query bootloader v3 ID after 2520 ms
14:08:04:767 TX c081027dffc0
14:08:05:268 close serial after 3022
14:08:05:278 query deCONZ firmware version
14:08:05:284 SLIP RX frame length: 9
deCONZ firmware version 26580700
14:08:05:295 set watchdog ttl status: 0x00
14:08:09:418 query bootloader v1 ID after 2014 ms
14:08:09:434 RX 60 bytes ASCII
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
, 08:48:53
 after 2031 ms
14:08:09:435 bootloader start after 2031 ms
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
14:08:09:439 GCF_ResetDeviceDone
14:08:09:452 bootloader v1 update firmware
flashing 164090 bytes: |==============================|
verify: .
SUCCESS
Wait 10 seconds until application starts
14:08:28:942 verify application is running
14:08:28:955 query deCONZ firmware version
14:08:28:958 SLIP RX frame length: 9
deCONZ firmware version 26650700
14:08:28:964 ok, application is running
```
no dice:
```
Sep 22 14:10:19 raspberrypi kernel: [ 1496.763165] usb 1-1.4: USB disconnect, device number 36
Sep 22 14:10:20 raspberrypi kernel: [ 1497.122762] usb 1-1.4: new full-speed USB device number 37 using dwc_otg
Sep 22 14:10:20 raspberrypi kernel: [ 1497.258970] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep 22 14:10:20 raspberrypi kernel: [ 1497.259005] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 22 14:10:20 raspberrypi kernel: [ 1497.259026] usb 1-1.4: Product: ConBee II
Sep 22 14:10:20 raspberrypi kernel: [ 1497.259045] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep 22 14:10:20 raspberrypi kernel: [ 1497.259065] usb 1-1.4: SerialNumber: DE2147999
Sep 22 14:10:20 raspberrypi kernel: [ 1497.276913] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
Sep 22 14:10:23 raspberrypi kernel: [ 1500.603174] usb 1-1.4: USB disconnect, device number 37
Sep 22 14:10:23 raspberrypi kernel: [ 1500.972782] usb 1-1.4: new full-speed USB device number 38 using dwc_otg
Sep 22 14:10:24 raspberrypi kernel: [ 1501.108856] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep 22 14:10:24 raspberrypi kernel: [ 1501.108891] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 22 14:10:24 raspberrypi kernel: [ 1501.108911] usb 1-1.4: Product: ConBee II
Sep 22 14:10:24 raspberrypi kernel: [ 1501.108930] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep 22 14:10:24 raspberrypi kernel: [ 1501.108949] usb 1-1.4: SerialNumber: DE2147999
Sep 22 14:10:24 raspberrypi kernel: [ 1501.123795] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
Sep 22 14:10:43 raspberrypi kernel: [ 1520.827199] usb 1-1.4: USB disconnect, device number 38
Sep 22 14:10:44 raspberrypi kernel: [ 1521.142972] usb 1-1.4: new full-speed USB device number 39 using dwc_otg
Sep 22 14:10:44 raspberrypi kernel: [ 1521.279035] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep 22 14:10:44 raspberrypi kernel: [ 1521.279072] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 22 14:10:44 raspberrypi kernel: [ 1521.279093] usb 1-1.4: Product: ConBee II
Sep 22 14:10:44 raspberrypi kernel: [ 1521.279113] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep 22 14:10:44 raspberrypi kernel: [ 1521.279131] usb 1-1.4: SerialNumber: DE2147999
Sep 22 14:10:44 raspberrypi kernel: [ 1521.294974] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
Sep 22 14:10:47 raspberrypi kernel: [ 1524.667194] usb 1-1.4: USB disconnect, device number 39
Sep 22 14:10:47 raspberrypi kernel: [ 1524.972901] usb 1-1.4: new full-speed USB device number 40 using dwc_otg
[... 48 lines cut ...]
Sep 22 14:12:19 raspberrypi kernel: [ 1616.259270] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep 22 14:12:19 raspberrypi kernel: [ 1616.259305] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 22 14:12:19 raspberrypi kernel: [ 1616.259325] usb 1-1.4: Product: ConBee II
Sep 22 14:12:19 raspberrypi kernel: [ 1616.259345] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep 22 14:12:19 raspberrypi kernel: [ 1616.259365] usb 1-1.4: SerialNumber: DE2147999
Sep 22 14:12:19 raspberrypi kernel: [ 1616.280824] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
Sep 22 14:12:38 raspberrypi kernel: [ 1636.027430] usb 1-1.4: USB disconnect, device number 46
Sep 22 14:12:39 raspberrypi kernel: [ 1636.343480] usb 1-1.4: new full-speed USB device number 47 using dwc_otg
Sep 22 14:12:39 raspberrypi kernel: [ 1636.479305] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Sep 22 14:12:39 raspberrypi kernel: [ 1636.479337] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 22 14:12:39 raspberrypi kernel: [ 1636.479358] usb 1-1.4: Product: ConBee II
[... cut ...]
Sep 22 14:12:43 raspberrypi kernel: [ 1640.329368] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 22 14:12:43 raspberrypi kernel: [ 1640.329388] usb 1-1.4: Product: ConBee II
Sep 22 14:12:43 raspberrypi kernel: [ 1640.329408] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Sep 22 14:12:43 raspberrypi kernel: [ 1640.329429] usb 1-1.4: SerialNumber: DE2147999
Sep 22 14:12:43 raspberrypi kernel: [ 1640.347080] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
```

btw, the firmware files are from http://deconz.dresden-elektronik.de/deconz-firmware/

Smanar, 2020-09-22:
Your firmware updates on windows was on real PC or virtual one ? Captures are stranges.

artkrz, 2020-09-22:
@Smanar all physical hardware pi and windows. What do You need from the network settings window?

Smanar, 2020-09-22:
Just compare with this one 

[image]

artkrz, 2020-09-22:
[image]
 this is it

Smanar, 2020-09-22:
You haven't this message in log ?
>16:30:59:109 apsUseExtPanid is 0xDDDDDDDDDDDDDDDD but should be 0, start reconfiguration

IDK if it can change something to your configuration, but try to put good value for EXT PAN ID

artkrz, 2020-09-22:
Where's the log and what is good value for EXT PAN ID? Sorry I never used Deconz to do this, only used Phoscon when it worked.

Smanar, 2020-09-22:
Same than the capture

- 0x00212 ..... (it s the mac adress)
- 0x0000000000000000

artkrz, 2020-09-22:
Still don;t get what @Smanar is asking for but I gave it another shot with reconnecting it via usbipd when it was showing up on pi and strangely after couple of tries I managed to hook it up in the way I used it when it was working. pi => usbipd => proxmox => hasos vm. It's not getting disconnected but it also doesn't work. Shows up in `lsusb` on proxmox host, is passed to VM but doesn't respond to any commands from zigbee2mqtt or update any sensors. I'll leave it for the night and see did anything change in the morning.

Smanar, 2020-09-23:
If you have no more the disconnection bug, you can too run deconz in debug mode.

artkrz, 2020-09-23:
I still do, it's just not happening if I'm fast enough to connect z2m to it. Doesn't really matter as it still doesn't work. Even when I'm technically connected to the stick.

artkrz, 2020-09-24:
SO new stick arrives, it's stable but FW is 0x264a0700. When I flashed it to 0x26580700 it bootloops! This is so strange. Anyway, looks like I have a spare now or I broke 2nd one.

artkrz, 2020-09-24:
I'm happy to report I got it stable, sort of, here's what I did:

1. On the pi: Start USBIP host
2. On the client (Prooxmox) start USBIP client
3. Start z2m quickly so it grabs conbee2 via usb pass trough before it gets disconnected.

No more disconnects in the log for now.

I mentioned this 2 comments earlier that it connects but does not work, turns out it does work but I have to go around and reset all devices, no idea why. Once i do, they're working fine. Will see for how long. I think combining Proxmox pass-trough and USBIP is a major PITA here as it needs perfect timing to grab the stick on host, then client needs to connect to it and then Proxmox needs to pass it to VM which has to actually use it. If disconnection happens anytime in this process it all goes downhill :( 

So for now it's working but I have no idea will my network work after another restart though. Question is, should the disconnects happen at all? That's not normal for a USB device.

If there's anything else I can do to troubleshoot this I'm happy to help as I have a spare Conbee2 now.

Smanar, 2020-09-24:
Just by curisoity, you have configured the USB port as 2.0 and not 3.0 (Frequent issue on Promox)

artkrz, 2020-09-24:
Sure, obviously it’s not simple. Btw, yes, proxmox has usb 3 ticked. But lets take that whole proxmox, usbip etc out of the picture for a moment. How is it possible both sticks disconnect when connected to a different raspberry pi that doesn’t even have usbip installed. Just plain rasbian with deconz installed but mot started at all?

Smanar, 2020-09-24:
You need to have it NOT ticked.
If you are using Raspberry 4, it s same, don't use USB 3.0.

artkrz, 2020-09-24:
I’m not using rpi 4. Again, happens on different pi with Rasbian which was never connected in any way with proxmox as well. Proxmox has nothing to do with it. At least on that board.

closed 2020-10-25 by stale[bot]

dvbit, 2021-01-03:
please reopen.
Issue is not solved.

Mimiix, 2021-01-03:
@dvbit quit replying on old issues. I've already told you what to do.