dresden-elektronik/deconz-rest-plugin #3294: Xiaomi (mija) door\window sensor Not reachable after RaspBee reboot

Nazarbayevv, 2020-09-21:
RaspBee
Version 2.05.77/22.05.2020
Firmware 26370500

When I add Xiaomi (mija) door\window sensor it's work fine till I reboot my Raspberry PI. After reboot I can see in Phoscon, that sensor Not reachable
[image]
and this sensor disappear from GUI

I tried previosly firmwares - the same issue. Other devices, including Aqara door\window sensor works fine after reboot.

SwoopX, 2020-09-21:
I leave this issue open if you provide the remaining info requested from the template, as this is requested for the reason to better understand your environment and what might be going on.

Devices can only disappear from deconz GUI if they cannot be, due to whatever reason, saved to the database. Do you have other devices and are they working ok? As a sidenote, this should not be related to the firmware.

Why did you update the firmware, but not the software?

Nazarbayevv, 2020-09-21:
I will add adition information.

I use with my RaspBee: a lot of Aqara wall switches, few aqara windows/door sensors, aqara motion sensors, xiaomi motion sensors and there are work fine. I bought few xiaomi windows/door sensors and thay work not correctly.
I can easy add xiaomi windows/door sensor via Phoscon. And I can see it in GUI:
  
[image]
[image]
[image]

I have already update version to 2.05.81/14.09.2020, but withot changes. I deleted sensor and added again. I can see status close, but it does not react to open/close and i can't find sensor in deconz GUI.

Nazarbayevv, 2020-09-23:
I reinstalled my sistem (Rasbian) yesterday and tried to add this door/window sensor again and i reboted system after that. Sensor still works. But than i added other xiaomi door/window sensor and I saw previously issues (I added sensor via Phoscon, I saw open/close status, but after reboot sensor disappeared from GUI). I can't see Manufacturer code 
[image]

closed 2020-10-25 by stale[bot]