dresden-elektronik/deconz-rest-plugin #3295: Device name

asgothian, 2020-09-21:
I would like to request support for the new Sonoff Occupancy sensors (They are marketed as Motion Sensors, but i am 99% certain they sense occupancy, not motion)_

[image]

## Device

- Product name: Sonoff SNZB-03
- Manufacturer: eWeLink
- Model identifier: MSO1.
- Device type : Occupancy Sensor
   - Sensor
  

## Screenshots

[image]
[image]
[image]
[image]
[image]

Mimiix, 2020-09-22:
Already integrated into deCONZ. 

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2991

SwoopX, 2020-09-22:
Tapped almost in the same trap, different model ID here 😉

Mimiix, 2020-09-22:
Really?

Good catch!

asgothian, 2020-09-22:
Which version has this integrated ? I am running 2.05.80 / 8/14/2020, and the sensors are not recognized.

A.

SwoopX, 2020-09-22:
Please use the search function to find out. You can't miss it.

closed 2020-09-25 by manup