dresden-elektronik/deconz-rest-plugin #3296: updating firmware RaspBee to 0x26650700 does not work

wvuyk, 2020-09-22:
## Describe the question or issue you are having

I am trying to manually flash my RaspBee following instructions at https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually, so I can continue with the latest version of the firmware (0x26650700).
I have updated the WiringPi package to the latest version. There were no errors during this update. Still it seems there is something missing when trying to flash the firmwafre.

Running version raspbian:

NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch

When I run the command `sudo GCFFlasher_internal -t 60 -f deCONZ_Rpi_0x26370500.bin.GCF` I get an error
`/usr/bin/GCFFlasher_internal.bin: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by /usr/lib/libwiringPi.so)`

## Environment

 - Host system: Raspberry Pi 
 - Running method: Raspbian 
 - Firmware version: 26350500
 - deCONZ version: 2.05.79
 - Device:  RaspBee I 

### deCONZ Logs

## Additional context

I am not really good with Linux here, how do I proceed with the update from here?

Wim

SwoopX, 2020-09-22:
You could try `sudo apt install -f` to see if any dependencies were missing. The command should install them.

wvuyk, 2020-09-22:
I tried this command, but it seems the system is already up to date, I use this command when upgrading the deCONZ software most of the time indeed.

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```
Did some more searching and found a similar error on the domotics forum [here](https://github.com/domoticz/domoticz/issues/3880), where someone mentions this will not work on raspbian stretch and you need buster for this? Can this be true in this case as well?

Smanar, 2020-09-22:
Its the command you have used to update wingripi ?

```
 cd /tmp
 wget https://project-downloads.drogon.net/wiringpi-latest.deb
 sudo dpkg -i wiringpi-latest.deb
```

SwoopX, 2020-09-22:
> Did some more searching and found a similar error on the domotics forum here, where someone mentions this will not work on raspbian stretch and you need buster for this? Can this be true in this case as well?

My production environment is on stretch, so no.

wvuyk, 2020-09-22:
@Smanar  Yes, I did copy these commands from the wiki page https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually to make sure I would get the package. If I repeat the commands, in installs it again, with no difference. 
@SwoopX clear... feels like I am in trouble now... is it possible the wiringpi is having dependencies that are not found yet?

morfei1, 2020-09-24:
On Buster it working for me.

wvuyk, 2020-09-25:
I have upgraded to buster here to test, but the error does not change, still getting the same issue here. I bought the device from dresden electronics a few years back, had to replace the old image with the stretch version at some point, Now I did the upgrade to buster, but this error is keeping me from updating to the latest firmware.

Is there a way of setting up this libwiringpi package so that it finds or installs the LIBC_2.28?

czm97176, 2020-09-25:
1. Run bash in docker

```docker run -it --rm --entrypoint /bin/bash --privileged --cap-add=ALL -v /dev:/dev -v /lib/modules:/lib/modules -v /sys:/sys marthoc/deconz```

2. Add "deb http://ftp.de.debian.org/debian sid main" to /etc/apt/sources.list

```echo "deb http://ftp.de.debian.org/debian sid main" > /etc/apt/sources.list```

3. Execute apt-install

```apt-get update && apt-get upgrade libc6```

4. Execute /firmware-update.sh

```./firmware-update.sh```

i was facing the similar issue found it from https://github.com/marthoc/docker-deconz/issues/153#issuecomment-538689683

wvuyk, 2020-09-25:
I am not using docker here on the PI I think? Am afraid I do not understand how to proceed on this, sorry. 
Will starting on the 2nd step do any good?

wvuyk, 2020-09-25:
@czm97176 
Got to thank you here ! I performed the second part of the 3rd step as wild guess on my RPI:
`apt-get upgrade libc6`

A lot was happening here, but I saw the GLIBC_2.28 passing by.
After a reboot, succes! Firmware is upgraded.

So, in short, the upgrade of WiringPi can be not enough for older systems, a `apt-get upgrade libc6` can be needed too?

Smanar, 2020-09-25:
Yep, probably, usefull to know.

wvuyk, 2020-09-27:
One more update on this. One of the users of my plugin had the same issue with the Raspbee 1. On Stretch as well and getting the same GLIB_2.28 error when trying to upgrade the firmware. Only performing the `apt-get upgrade libc6` did not help.
He (and me too during the process) would then get a message saying ' libc6 is already newest version (2.24-11+deb9u4)'

It would only work after an upgrade to buster AND after that doing a `apt-get upgrade libc6`

manup, 2020-09-27:
With GCFFlasher 3.15 (included since deCONZ 2.05.81) this issue should be fixed. In this version wiringPi is loaded dynamically and isn't linked against a specific GLIB version.

Note wiringPi needs to be compiled as described in https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

wvuyk, 2020-09-27:
@manup 
Thanks for this info. When I checked the upgrade of the firmware I was still on 2.0.5.80 and did not think of first upgrading to the new deCONZ version.... Should have save a lot of time. Will update my forum with this advise as well!

Mimiix, 2020-10-19:
Closing as seems solved. @wvuyk If it needs to be re-opened, let me know ;)

closed 2020-10-19 by Mimiix