dresden-elektronik/deconz-rest-plugin #3299: Philips Hue Motion Sensor: Cannot configure daylight parameter in phoscon App

jwich71, 2020-09-22:
Where can I configure the daylight parameter for Hue Motion Sensor in phoscon App? The sensor is already connected and shows the current light level in lux. Now I want to use this feature to configure a light level and using it in my motion sensor control settings. But there is no option for activating motion sensor on light level (lux) basis.
How can I do this?

Mimiix, 2020-09-23:
Closing this issue. Not using any template.

closed 2020-09-23 by Mimiix

jwich71, 2020-09-24:
Hmm, I'm new to this forum. Can you help me asking my question in the right way? I don't know, what you mean with "Not using any template" ...

Mimiix, 2020-09-25:
If you click "New issue" you are presented with 4 options.  Select user question.