dresden-elektronik/deconz-rest-plugin #3301: #define VENDOR_CENTRALITE   0x104E // wrong name?

7wells, 2020-09-22:
Hello, I have an Osram Smart+ Motion Sensor. Deconz shows the manufacturer ID 0x110c.

However, this is defined differently:
`#define VENDOR_CENTRALITE   0x104E // wrong name?`

Could it be that the manufacturer ID 0x104E is (as well as 0x110c) also for Osram?

Smanar, 2020-09-22:
No it s ok for me.
0x110c is Osram so no problem for manufacture number
Your device have "CentraLite" as manufacture name, but this value is written on the firmware (String value, not a number for here)

Mimiix, 2020-09-23:
@7wells Closing this issue. Not using the template.

closed 2020-09-23 by Mimiix

7wells, 2020-09-23:
??

Mimiix, 2020-09-23:
Please read: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3199

7wells, 2020-09-23:
Thanks for reminding me.