dresden-elektronik/deconz-rest-plugin #3304: Moes Tuya Thermostat BTH-002

d-litvinov, 2020-09-23:
## Device

- Product name: Moes thermostat BHT-002
- Manufacturer: _TZE200_aoclfnxz
- Model identifier: TS0601
- Device type :
  - Thermostat

## Screenshots
deconz version 2.05.81
[image]

[image]

[image]

[image]
[image]

d-litvinov, 2020-09-23:
Those are cheap and pretty decent floor heating thermostats from Aliexpress, so I think they are going to be pretty popular.

Smanar, 2020-10-02:
Hello, do you have an Unix machine to compile code to test  the code ?

Jut for information we already have some tuya thermostat working, but values can be different, so can take some time looking in log if your device don't work at first try.

d-litvinov, 2020-10-02:
Yes, I have a raspberry. And I'll be happy to help.
What should I test?

Smanar, 2020-10-03:
Here you have the procedure https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

Just replace the step 1 by
`git clone --branch tuya https://github.com/Smanar/deconz-rest-plugin.git`
and you can skip the step 
`git checkout -b mybranch HEAD`

The last step replace your actual file, so better to close deconz before, or deconz will be closed itself by the change.

After that you just need to re-include the device, will be invisible in phoscon but visible in API, third applications.

d-litvinov, 2020-10-03:
It works! It adds to rest api as a smart plug
`  {
      "etag": "e142589e806633ad22ee7215ec013336",
      "hascolor": false,
      "lastannounced": "2020-10-03T21:42:03Z",
      "lastseen": "2020-10-03T21:51Z",
      "manufacturername": "_TZE200_aoclfnxz",
      "modelid": "TS0601",
      "name": "Smart plug 2",
      "state": {
        "alert": "none",
        "on": false,
        "reachable": true
      },
      "swversion": null,
      "type": "Smart plug",
      "uniqueid": "bc:33:ac:ff:fe:5e:c7:d7-01"
    }

And I can switch it on/off from the HomeAssistant.
But setting temperature isn't avaliable
`

Smanar, 2020-10-04:
Ha ?
Normally I need to disable it from the "light" section, because this device will create a ZHAthermostat entry in "sensor"
When you switch it on/off in HA it change something ? (Else  I will disable the smart plug device)

You haven't another device created in "sensor" part ? try to add it with "add new sensor" in phoscon (don't delete the previous one), just try to add it again.

d-litvinov, 2020-10-04:
> When you switch it on/off in HA it change something ? 
No. But it updates the switch state, when I switch it on/off by the button on the thermostat.

>try to add it with "add new sensor" in phoscon (don't delete the previous one), just try to add it again.
I tried that with no success. 

But when I were trying to readd it by removing from the mesh and adding it through Phoscon, I got that sensor for a second:
`     "6": {
      "config": {
        "heatsetpoint": null,
        "locked": null,
        "mode": null,
        "offset": 0,
        "on": true,
        "preset": null,
        "reachable": true,
        "schedule": {},
        "schedule_on": null,
        "windowopen_set": null
      },
      "ep": 1,
      "etag": "ca04f5916fb6944912f01a427a7a06d5",
      "lastseen": "2020-10-04T13:24Z",
      "manufacturername": "_TZE200_aoclfnxz",
      "modelid": "TS0601",
      "name": "Thermostat 6",
      "state": {
        "lastupdated": "none",
        "lowbattery": false,
        "on": null,
        "temperature": null,
        "valve": null
      },
      "type": "ZHAThermostat",
      "uniqueid": "bc:33:ac:ff:fe:5e:c7:d7-01-0201"
    }`

But then it was replaced by the smart plug. It's weird.

Smanar, 2020-10-04:
Ok so I have added your device in a blacklist to prevent the plug device creation.

But I don't understand why It was replaced, they can't have the same uniqueid ....

ogkita, 2020-10-15:
Hello,

The first thing I want to say is thank you for the support this forum gives me, I am very new to this technology.

I'm very interested in this thermostat. @d-litvinov, did the @Smanar solution finally work?

Thank you!

d-litvinov, 2020-10-15:
@ogkita, no I haven’t tested the last version. Sorry, I’m on vacation at the moment, and can’t test it. I’ll check it in 2 weeks.

ftp27, 2020-10-16:
I've tried to build the latest commit in @Smanar 's branch `tuya`. So it was recognized as one light sensor
```json
{
    "etag": "454432e75e25912c36b1438401483348",
    "hascolor": false,
    "lastannounced": "2020-10-16T11:03:40Z",
    "lastseen": "2020-10-16T11:04Z",
    "manufacturername": "Heiman",
    "modelid": "TS0601",
    "name": "Smart plug 4",
    "state": {
        "alert": "none",
        "on": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Smart plug",
    "uniqueid": "bc:33:ac:ff:fe:5e:c8:82-01"
}
```
maybe I do something wrong. I do it the very first time. So what can I do to help?

Smanar, 2020-10-16:
Yep there is a problem, you have "Heiman" as manufacturename, the device can't work with that, all the code is based on the good manufacture name.

You haven't the GUI to read the values again ?

ftp27, 2020-10-16:
@Smanar sorry for the quality, it's VNC 
[image]
[image]

**Basic**
[image]
[image]
**Groups**
[image]
**Scenes**
[image]

Smanar, 2020-10-16:
Nice, better.
Can you check in "sensor" if you haven't a new one too ?

ftp27, 2020-10-16:
It can be this one. But I'm not sure
```json
{
    "config": {
        "battery": 100,
        "on": true,
        "reachable": true,
        "temperature": 2700
    },
    "ep": 1,
    "etag": "b5b9217ac54feb4253732a625b760a95",
    "lastseen": "2020-10-16T20:30Z",
    "manufacturername": "LUMI",
    "mode": 1,
    "modelid": "lumi.remote.b1acn01",
    "name": "Smart Switch",
    "state": {
        "buttonevent": 1002,
        "lastupdated": "2020-10-16T19:39:54.020"
    },
    "swversion": "20180525",
    "type": "ZHASwitch",
    "uniqueid": "00:15:8d:00:04:01:92:e1-01-0012"
}
```

Smanar, 2020-10-16:
Ok So I have updated the code, on the base of version 85 beta.

Now it force the device detection even without tuya cluster for this device

If you don't want to include the device again, just set phoscon in permit join "add new sensor" and read the basic attribute again (cluster 0x000, press the buton "read")

ogkita, 2020-10-23:
Hello all,

Any progress on this issue?

I expect to receive my thermostat in the next few weeks. Then I can check it out.

Thank you!

ogkita, 2020-10-30:
Hi @Smanar 

I have here the thermostat and i am doing test for include in the deconz network.

When i link the termostat:

1. I can only see a lights in phoscon web application.
2. I can not see any more device (no sensor, no switch).

In /lights? api call i get this json:
{
  "1": {
    "etag": "eeebb8c23b7fb7eb40163ee292b7ab05",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2020-10-30T17:53Z",
    "manufacturername": "dresden elektronik",
    "modelid": "RaspBee II",
    "name": "Configuration tool 1",
    "state": {
      "reachable": true
    },
    "swversion": null,
    "type": "Configuration tool",
    "uniqueid": "00:21:2e:ff:ff:06:1b:54-01"
  },
  "2": {
    "etag": "d50809778874780320e29153d1c842b0",
    "hascolor": false,
    "lastannounced": "2020-10-30T17:30:38Z",
    "lastseen": "2020-10-30T17:53Z",
    "manufacturername": "_TZE200_aoclfnxz",
    "modelid": "TS0601",
    "name": "Smart plug 2",
    "state": {
      "alert": "none",
      "on": false,
      "reachable": true
    },
    "swversion": "0xFFFFFFFF",
    "type": "Smart plug",
    "uniqueid": "60:a4:23:ff:fe:9f:4f:a7-01"
  }
}

In /sensors? api call i get this json:
{
  "1": {
    "config": {
      "configured": true,
      "on": true,
      "sunriseoffset": 30,
      "sunsetoffset": -30
    },
    "etag": "b8b1fa9f2a3ebc440a03f3af6ccfc7eb",
    "manufacturername": "Philips",
    "modelid": "PHDL00",
    "name": "Daylight",
    "state": {
      "dark": true,
      "daylight": false,
      "lastupdated": "2020-10-30T17:42:22.617",
      "status": 210,
      "sunrise": "2020-10-30T06:42:31",
      "sunset": "2020-10-30T17:14:05"
    },
    "swversion": "1.0",
    "type": "Daylight",
    "uniqueid": "00:21:2e:ff:ff:06:1b:54-01"
  },
  "2": {
    "config": {
      "heatsetpoint": null,
      "locked": null,
      "mode": "off",
      "offset": 0,
      "on": true,
      "preset": null,
      "reachable": true,
      "schedule": {
        "W127": [
          {
            "heatsetpoint": 40,
            "localtime": "T06:00"
          },
          {
            "heatsetpoint": 42,
            "localtime": "T11:30"
          },
          {
            "heatsetpoint": 44,
            "localtime": "T13:30"
          },
          {
            "heatsetpoint": 46,
            "localtime": "T17:30"
          },
          {
            "heatsetpoint": 48,
            "localtime": "T06:00"
          },
          {
            "heatsetpoint": 46,
            "localtime": "T12:00"
          },
          {
            "heatsetpoint": 44,
            "localtime": "T14:30"
          },
          {
            "heatsetpoint": 42,
            "localtime": "T17:30"
          },
          {
            "heatsetpoint": 38,
            "localtime": "T06:00"
          },
          {
            "heatsetpoint": 40,
            "localtime": "T12:30"
          },
          {
            "heatsetpoint": 42,
            "localtime": "T14:30"
          },
          {
            "heatsetpoint": 40,
            "localtime": "T18:30"
          }
        ]
      },
      "schedule_on": null,
      "setvalve": null,
      "windowopen_set": null
    },
    "ep": 1,
    "etag": "1bbf09191538b1facfbbf770d32fda31",
    "lastseen": "2020-10-30T17:53Z",
    "manufacturername": "_TZE200_aoclfnxz",
    "modelid": "TS0601",
    "name": "TS0601",
    "state": {
      "lastupdated": "none",
      "lowbattery": null,
      "on": null,
      "temperature": null,
      "valve": null
    },
    "type": "ZHAThermostat",
    "uniqueid": "60:a4:23:ff:fe:9f:4f:a7-01-0201"
  }
}

I try to add de thermostat in the openhab and i get 3 devices:
   Daylight (Philips), Daylight Sensor
[image]

   Smart plug 2 (_TZE200_aoclfnxz), On/Off Light
[image]

   TS0601 (_TZE200_aoclfnxz), Thermostat
[image]

I can only manage the "Smart plug 2", on/off, When I press the on button, the thermostat turns on the screen, off button turns off the screen.

I hope this information help you.

Thank you

ogkita, 2020-10-30:
Hi @Smanar 

One more thing: when I restarted my Raspberry, the two lights in the endpoint /lights? are disappeared.

Thank you.

ogkita, 2020-10-30:
Hi @Smanar 

After rebboting the raspberry, if I flash the raspbee with this command:
	sudo GCFFlasher_internal -d /dev/ttyS0 -t 10 -f deCONZ_RaspBeeII_0x26660700.bin.GCF

Then i can see the smart plug 2 in /lights? endpoint

{
  "1": {
    "etag": "fe4d98e1b44d378727afef008d3cfb9f",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2020-10-30T20:12Z",
    "manufacturername": "dresden elektronik",
    "modelid": "RaspBee II",
    "name": "Configuration tool 1",
    "state": {
      "reachable": true
    },
    "swversion": null,
    "type": "Configuration tool",
    "uniqueid": "00:21:2e:ff:ff:06:1b:54-01"
  },
  "2": {
    "etag": "fe4d98e1b44d378727afef008d3cfb9f",
    "hascolor": false,
    "lastannounced": "2020-10-30T17:30:38Z",
    "lastseen": "2020-10-30T20:12Z",
    "manufacturername": "_TZE200_aoclfnxz",
    "modelid": "TS0601",
    "name": "Smart plug 2",
    "state": {
      "alert": "none",
      "on": false,
      "reachable": true
    },
    "swversion": "0xFFFFFFFF",
    "type": "Smart plug",
    "uniqueid": "60:a4:23:ff:fe:9f:4f:a7-01"
  }
}

And i can see de version in the web application:
[image]

If i reboot and dont flash, the raspbee ii does not show lights and version in web app
[image]

Thank you

Smanar, 2020-10-30:
Hello, 

So first when you have "not connected" at firmware, it s not good, it mean deconz don't see the gateway, so better to use the other version.

>I can only manage the "Smart plug 2", on/off, When I press the on button, the thermostat turns on the screen, off button turns off the screen.

Realy nice information, first time I see that, and the screen sta on, or turn off itself after a time ?

In reality I need to disable the device you are seen in "light", for you have only the one in "sensor", but I did know the last feature, now it s another question ...

But more important, need to check what is working, what is not working, and what the device don't have.

- Temperature is not working
- heatsetpoint is not working
- Schedule is working, it s the first tuya it work natively, do you have a way to compare values ?
- all options are not working, windowsopen, lock, battery, .....

I have updated the code according to your device, you will have more features working.
To update code

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

I haven't tested the code on my side, so if you have error message during compilation (except warning), stop the procedure

Edit:
ok, i have found for the display on/off, it because they are using same value than for switch

ogkita, 2020-10-31:
Hi @Smanar 

I updated de git, compile without errors, delete device, and start again. this is that i have:

/lights?
{
  "1": {
    "etag": "088b17c4a2e088c243752ad9405eda10",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2020-10-31T06:39Z",
    "manufacturername": "dresden elektronik",
    "modelid": "RaspBee II",
    "name": "Configuration tool 1",
    "state": {
      "reachable": true
    },
    "swversion": null,
    "type": "Configuration tool",
    "uniqueid": "00:21:2e:ff:ff:06:1b:54-01"
  },
  "3": {
    "etag": "4bde907c04111e40f5e463a6492b6737",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2020-10-31T06:40Z",
    "manufacturername": "Unknown",
    "modelid": "TS0601",
    "name": "Smart plug 3",
    "state": {
      "alert": "none",
      "on": false,
      "reachable": true
    },
    "swversion": null,
    "type": "Smart plug",
    "uniqueid": "60:a4:23:ff:fe:9f:4f:a7-01"
  }
}

/sensors?
{
  "1": {
    "config": {
      "configured": true,
      "on": true,
      "sunriseoffset": 30,
      "sunsetoffset": -30
    },
    "etag": "c86d23c1dd5624a5ccb797ddd4442f55",
    "manufacturername": "Philips",
    "modelid": "PHDL00",
    "name": "Daylight",
    "state": {
      "dark": true,
      "daylight": false,
      "lastupdated": "2020-10-31T06:38:44.807",
      "status": 130,
      "sunrise": "2020-10-31T06:43:41",
      "sunset": "2020-10-31T17:12:52"
    },
    "swversion": "1.0",
    "type": "Daylight",
    "uniqueid": "00:21:2e:ff:ff:06:1b:54-01"
  },
  "2": {
    "config": {
      "heatsetpoint": null,
      "locked": null,
      "mode": "off",
      "offset": 0,
      "on": true,
      "preset": null,
      "reachable": true,
      "schedule": {
        "W127": [
          {
            "heatsetpoint": 40,
            "localtime": "T06:00"
          },
          {
            "heatsetpoint": 42,
            "localtime": "T11:30"
          },
          {
            "heatsetpoint": 44,
            "localtime": "T13:30"
          },
          {
            "heatsetpoint": 46,
            "localtime": "T17:30"
          },
          {
            "heatsetpoint": 48,
            "localtime": "T06:00"
          },
          {
            "heatsetpoint": 46,
            "localtime": "T12:00"
          },
          {
            "heatsetpoint": 44,
            "localtime": "T14:30"
          },
          {
            "heatsetpoint": 42,
            "localtime": "T17:30"
          },
          {
            "heatsetpoint": 38,
            "localtime": "T06:00"
          },
          {
            "heatsetpoint": 40,
            "localtime": "T12:30"
          },
          {
            "heatsetpoint": 42,
            "localtime": "T14:30"
          },
          {
            "heatsetpoint": 40,
            "localtime": "T18:30"
          }
        ]
      },
      "schedule_on": null,
      "setvalve": null,
      "windowopen_set": null
    },
    "ep": 1,
    "etag": "8babac84e4da63a596029dcfabb15a01",
    "lastseen": "2020-10-31T06:40Z",
    "manufacturername": "_TZE200_aoclfnxz",
    "modelid": "TS0601",
    "name": "TS0601",
    "state": {
      "lastupdated": "none",
      "on": null,
      "temperature": null
    },
    "type": "ZHAThermostat",
    "uniqueid": "60:a4:23:ff:fe:9f:4f:a7-01-0201"
  }
}

in the openhab pannel adding devices:
[image]

In the openhab pannel things:
[image]
[image]

In the openhab control pannel:
[image]
[image]
[image]

Now the difference is that in the things panel in openhab the thermostat appears online.

If I turn the display on the thermostat physically, the smart plug 3 device is updated on the openhab panel to on. After a few seconds when the screen on the thermostat is physically turned off, on the openhab panel, the smart plug 3 device is turned off.

I hope this information help you.

Thank you

Smanar, 2020-10-31:
After some time, you still have no value for "state/temperature" and "config/heatsetpoint" ?

I have take the tuya value from z2m project and it s the same device, strange you haven't the values ...

If you still haven't no value can you run deconz in debug mode
>deCONZ --dbg-error=2 --dbg-info=2 --dbg-zdp=0 --dbg-zcl=0 --db-aps=0 --dbg-http=0 --dbg-ota=0

In log, every time your thermostat is sending information you will have debug line like

> Tuya debug 4 : xxxxxxxxxxxxxxx
> Tuya debug 5 : xxxxxxxxxxxxxxxxxxxxxxx

Im using them to find tuya value.

ogkita, 2020-10-31:
Hi @Smanar 

[Smanar] After some time, you still have no value for "state/temperature" and "config/heatsetpoint" ?
[Oscar No, any value

Here you have some debug data:
11:40:03:323 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06c1100200040000000e
11:40:03:323 Tuya debug 5 : Status: 6 Transid: 193 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:40:03:344 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06c2100200040000000e
11:40:03:344 Tuya debug 5 : Status: 6 Transid: 194 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:40:03:365 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06c3100200040000000e
11:40:03:365 Tuya debug 5 : Status: 6 Transid: 195 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:40:32:682 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06bc18020004000000b9
11:40:32:682 Tuya debug 5 : Status: 6 Transid: 188 Dp: 536 (0x02,0x18) Fn: 0 Data 185
11:40:32:881 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06bc18020004000000b9
11:40:32:881 Tuya debug 5 : Status: 6 Transid: 188 Dp: 536 (0x02,0x18) Fn: 0 Data 185
11:40:33:082 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06bc18020004000000b9
11:40:33:083 Tuya debug 5 : Status: 6 Transid: 188 Dp: 536 (0x02,0x18) Fn: 0 Data 185
11:40:33:290 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06bc18020004000000b9
11:40:33:290 Tuya debug 5 : Status: 6 Transid: 188 Dp: 536 (0x02,0x18) Fn: 0 Data 185
11:41:32:298 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06bd18020004000000b9
11:41:32:298 Tuya debug 5 : Status: 6 Transid: 189 Dp: 536 (0x02,0x18) Fn: 0 Data 185
11:41:32:499 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06bd18020004000000b9
11:41:32:499 Tuya debug 5 : Status: 6 Transid: 189 Dp: 536 (0x02,0x18) Fn: 0 Data 185
11:41:32:701 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06bd18020004000000b9
11:41:32:701 Tuya debug 5 : Status: 6 Transid: 189 Dp: 536 (0x02,0x18) Fn: 0 Data 185
11:41:32:904 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06bd18020004000000b9
11:41:32:904 Tuya debug 5 : Status: 6 Transid: 189 Dp: 536 (0x02,0x18) Fn: 0 Data 185
11:41:35:101 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06be100200040000000e
11:41:35:101 Tuya debug 5 : Status: 6 Transid: 190 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:35:315 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06be100200040000000e
11:41:35:315 Tuya debug 5 : Status: 6 Transid: 190 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:35:504 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06be100200040000000e
11:41:35:504 Tuya debug 5 : Status: 6 Transid: 190 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:35:706 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06be100200040000000e
11:41:35:706 Tuya debug 5 : Status: 6 Transid: 190 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:35:906 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06be100200040000000e
11:41:35:906 Tuya debug 5 : Status: 6 Transid: 190 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:36:311 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06bf100200040000000e
11:41:36:311 Tuya debug 5 : Status: 6 Transid: 191 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:36:715 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06c0100200040000000e
11:41:36:715 Tuya debug 5 : Status: 6 Transid: 192 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:37:118 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06c1100200040000000e
11:41:37:118 Tuya debug 5 : Status: 6 Transid: 193 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:37:522 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06c2100200040000000e
11:41:37:522 Tuya debug 5 : Status: 6 Transid: 194 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:37:926 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06c3100200040000000e
11:41:37:926 Tuya debug 5 : Status: 6 Transid: 195 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:38:337 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06c4100200040000000e
11:41:38:337 Tuya debug 5 : Status: 6 Transid: 196 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:38:734 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06c5100200040000000e
11:41:38:734 Tuya debug 5 : Status: 6 Transid: 197 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:39:136 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06c6100200040000000e
11:41:39:136 Tuya debug 5 : Status: 6 Transid: 198 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:39:540 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06c7100200040000000e
11:41:39:540 Tuya debug 5 : Status: 6 Transid: 199 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:39:943 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06c8100200040000000e
11:41:39:943 Tuya debug 5 : Status: 6 Transid: 200 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:40:347 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06c9100200040000000e
11:41:40:347 Tuya debug 5 : Status: 6 Transid: 201 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:40:751 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06ca100200040000000e
11:41:40:751 Tuya debug 5 : Status: 6 Transid: 202 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:41:157 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06cb100200040000000e
11:41:41:157 Tuya debug 5 : Status: 6 Transid: 203 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:41:557 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06cc100200040000000e
11:41:41:557 Tuya debug 5 : Status: 6 Transid: 204 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:41:963 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06cd100200040000000e
11:41:41:963 Tuya debug 5 : Status: 6 Transid: 205 Dp: 528 (0x02,0x10) Fn: 0 Data 14
11:41:42:365 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06ce100200040000000e
[... cut ...]
11:42:32:124 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 000618020004000000be
11:42:32:124 Tuya debug 5 : Status: 0 Transid: 6 Dp: 536 (0x02,0x18) Fn: 0 Data 190
11:42:32:310 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 000618020004000000be
11:42:32:310 Tuya debug 5 : Status: 0 Transid: 6 Dp: 536 (0x02,0x18) Fn: 0 Data 190
11:42:32:513 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 000618020004000000be
11:42:32:513 Tuya debug 5 : Status: 0 Transid: 6 Dp: 536 (0x02,0x18) Fn: 0 Data 190

Any test do you need more only let me know it.

Thank you

ogkita, 2020-10-31:
Hi Again @Smanar 

Here you have a complete log for some time.

Thank you

[20201031 1352 deconz.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5469094/20201031.1352.deconz.log)

Smanar, 2020-10-31:
Ok I have found the problem ^^
It s because of the light device, the code use only the "light" part and ignore the sensor part, if you delete the "light" sensor, the code surely work.

I have updated the code, I think it will be better, it will prevent the light creation and can ignore it later.

ogkita, 2020-10-31:
Hi @Smanar 

No changes,

When you say "I have updated the code", do i need update my git and compile library again?

Sorry, maybe i doing something wrong

Thank you.

Smanar, 2020-10-31:
Yep, but faster this time

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

ogkita, 2020-10-31:
Hi @Smanar 

Now it is diferent ;-)

/lights?
{
  "1": {
    "etag": "b354aaa7ad21e67fcc6bee0122d163a3",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2020-10-31T17:06Z",
    "manufacturername": "dresden elektronik",
    "modelid": "RaspBee II",
    "name": "Configuration tool 1",
    "state": {
      "reachable": true
    },
    "swversion": null,
    "type": "Configuration tool",
    "uniqueid": "00:21:2e:ff:ff:06:1b:54-01"
  }
}

/switch?
[
  {
    "error": {
      "address": "/switch",
      "description": "method, GET, not available for resource, /switch",
      "type": 4
    }
  }
]

/sensors?
{
  "1": {
    "config": {
      "configured": true,
      "on": true,
      "sunriseoffset": 30,
      "sunsetoffset": -30
    },
    "etag": "7f194c898b2054c3c94694e7bef7f906",
    "manufacturername": "Philips",
    "modelid": "PHDL00",
    "name": "Daylight",
    "state": {
      "dark": false,
      "daylight": false,
      "lastupdated": "2020-10-31T16:53:02.811",
      "status": 180,
      "sunrise": "2020-10-31T06:43:41",
      "sunset": "2020-10-31T17:12:52"
    },
    "swversion": "1.0",
    "type": "Daylight",
    "uniqueid": "00:21:2e:ff:ff:06:1b:54-01"
  },
  "2": {
    "config": {
      "heatsetpoint": 2200,
      "locked": true,
      "mode": "auto",
      "offset": 50,
      "on": true,
      "preset": null,
      "reachable": false,
      "schedule": {
        "W127": [
          {
            "heatsetpoint": 40,
            "localtime": "T06:00"
          },
          {
            "heatsetpoint": 42,
            "localtime": "T11:30"
          },
          {
            "heatsetpoint": 44,
            "localtime": "T13:30"
          },
          {
            "heatsetpoint": 46,
            "localtime": "T17:30"
          },
          {
            "heatsetpoint": 48,
            "localtime": "T06:00"
          },
          {
            "heatsetpoint": 46,
            "localtime": "T12:00"
          },
          {
            "heatsetpoint": 44,
            "localtime": "T14:30"
          },
          {
            "heatsetpoint": 42,
            "localtime": "T17:30"
          },
          {
            "heatsetpoint": 38,
            "localtime": "T06:00"
          },
          {
            "heatsetpoint": 40,
            "localtime": "T12:30"
          },
          {
            "heatsetpoint": 42,
            "localtime": "T14:30"
          },
          {
            "heatsetpoint": 40,
            "localtime": "T18:30"
          }
        ]
      },
      "schedule_on": null,
      "setvalve": null,
      "windowopen_set": null
    },
    "ep": 1,
    "etag": "73f0c1567b0dd3f1ce08a39f0e9995cb",
    "lastseen": "2020-10-31T16:56Z",
    "manufacturername": "_TZE200_aoclfnxz",
    "modelid": "TS0601",
    "name": "TS0601",
    "state": {
      "lastupdated": "none",
      "on": null,
      "temperature": 1800
    },
    "type": "ZHAThermostat",
    "uniqueid": "60:a4:23:ff:fe:9f:4f:a7-01-0201"
  }
}

In openhab things pannel i have this devices:
[image]
[image]

In openhab control pannel i have this data in the thermostat:
[image]

I can see:
   - Temperature, but this value is not correct, in display i have 20 grades.
   - Target temperature, but this value is not correct, i have set the thermostat on 16 grades.
   - Mode, It the same, I change this value and nothing happend.
   - Offset, I don't know what it means

Thank you

Smanar, 2020-10-31:
The offset is a value to add to have the correct value, it can exlain why you have a difference of 4 degré on all your value (but this value is 50, so need to check), perhaps you need some minuts to have the correct value, this kind of device is relay not reacrive.

For the mode, I need to check to device documentation to see if you have one, most of TRV are using "preset" instead of "mode"

ogkita, 2020-10-31:
Hi @Smanar 

In wep application i can not see any sensor:
[image]

is it normal?

I am trying to start again other time ;-)

Thank you

Smanar, 2020-10-31:
Yep, it s normal, thermostat are not visible in phoscon, you need a third app, or using direclty the api..

ogkita, 2020-10-31:
Hi @Smanar 

Yep, more info...

When i set Target temperature:

13 grades in openhab pannel -> in the thermostat is setted to 2 grades
13,5 grades in openhab pannel -> in the thermostat is setted to 7 grades
14 grades in openhab pannel -> in the thermostat is setted to 12 grades
14,5 grades in openhab pannel -> in the thermostat is setted to 17 grades
15 grades in openhab pannel -> in the thermostat is setted to 22 grades
15,5 grades in openhab pannel -> in the thermostat is setted to 27 grades

Thank you

Smanar, 2020-10-31:
Ok, the offset don't seem to be used by the device (or I haven't the value) Can you try to set this value to 0 in openHab ?

Ok so new version to try:
- have removed windowopen
- perhaps correction for thermostat temperature (not sure, the correction is strange)
- Activate "mode", worj with value "off" and "heat", it turn the thermostat on/off
- Enable "lock feature"
- Enable "preset" that work with "auto" and "program"

ogkita, 2020-10-31:
HI @Smanar 

I have an error compiling:

qmake && make -j2
make -f Makefile.Release
make[1]: Entering directory '/opt/deconz/deconz-rest-plugin'
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -std=gnu++11 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DUSE_WEBSOCKETS -DHAS_SQLITE3 -DHAS_OPENSSL -DGW_SW_VERSION=\"2.05.86\" -DGW_SW_DATE=1604169112 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"116a6fbddd8c1f43ad867aeb6cbd264266fe5063\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26390500 -DGW_MIN_R21_FW_VERSION=0x26660700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_SERIALPORT_LIB -DQT_WEBSOCKETS_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. -I../.. -I../../common -isystem /usr/include/arm-linux-gnueabihf/qt5 -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWidgets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtGui -isystem /usr/include/arm-linux-gnueabihf/qt5/QtSerialPort -isystem /usr/include/arm-linux-gnueabihf/qt5/QtWebSockets -isystem /usr/include/arm-linux-gnueabihf/qt5/QtNetwork -isystem /usr/include/arm-linux-gnueabihf/qt5/QtCore -Irelease -I. -I/usr/lib/arm-linux-gnueabihf/qt5/mkspecs/linux-g++ -o release/rest_sensors.o rest_sensors.cpp
rest_sensors.cpp: In member function ‘int DeRestPluginPrivate::changeSensorConfig(const ApiRequest&, ApiResponse&)’:
rest_sensors.cpp:1205:33: error: expected initializer before ‘modeSet’
                         QString modeSet = map[pi.key()].toString();
                                 ^~~~~~~
rest_sensors.cpp:1206:29: error: ‘modeSet’ was not declared in this scope
                         if (modeSet == "heat") { data = QByteArray("\x00", 1); }
                             ^~~~~~~
rest_sensors.cpp:1206:29: note: suggested alternative: ‘mode_t’
                         if (modeSet == "heat") { data = QByteArray("\x00", 1); }
                             ^~~~~~~
                             mode_t
make[1]: *** [Makefile.Release:1099: release/rest_sensors.o] Error 1
make[1]: Leaving directory '/opt/deconz/deconz-rest-plugin'
make: *** [Makefile:40: release] Error 2

Smanar, 2020-10-31:
Corrected.

ogkita, 2020-10-31:
Hi @Smanar 

Temperature: Working OK

Target temperature: Change in different scale:
13 grades in openhab pannel -> in the thermostat is setted to 2 grades
13,5 grades in openhab pannel -> in the thermostat is setted to 7 grades
14 grades in openhab pannel -> in the thermostat is setted to 12 grades
14,5 grades in openhab pannel -> in the thermostat is setted to 17 grades
15 grades in openhab pannel -> in the thermostat is setted to 22 grades
15,5 grades in openhab pannel -> in the thermostat is setted to 27 grades

Auto: turn off display
Heat: turn on display
Off: turn off display.

Thank you

ogkita, 2020-10-31:
Hi @Smanar 

More info... 

When i set the temperature (Target temperature) in the thermostat, The new Target temperature is not update in the openhab pannel.

Thank you

Smanar, 2020-10-31:
>11:41:48:420 Tuya debug 4 : Address 0x60A423FFFE9F4FA7 Payload 06dd100200040000000e
>11:41:48:420 Tuya debug 5 : Status: 6 Transid: 221 Dp: 528 (0x02,0x10) Fn: 0 Data 14

New try for heatsetpoint, but it seem the devie only support full value, 14 is ok but not 14,5.

On "mode" "auto do nothing on this device.
This command can only turn it on or off, if it is "on", it will only work in "auto" mode,you need to use the "preset" command to set "normal auto" or "programmed auto"

And a corrective for schedule, this device use 12 settings, 4 for the week and 8 for the Week end.

ogkita, 2020-10-31:
Hi @Smanar 

Temperture is working ok
Target temperature is work better:
   Set 18.0 grades -> set 18 grades in the thermostat.
   Set 18.5 grades -> set 18 grades in the thermostat.
   Set 19.0 grades -> set 19 grades in the thermostat.
   Set 19.5 grades -> set 19 grades in the thermostat.

Maybe work with 18.5 instead of the 18,5. "." instead of the ","??

Mode only turn on and turn off display.

Thank you

Smanar, 2020-10-31:
Nope, the device only use the 2 lasts bytes, it use 4 bytes for temperature, but only 2 for setpoint.

I will make the PR tomorrow because I need to prepare it, but you have time to test if you miss something or the schedule if you need it.

ogkita, 2020-10-31:
Ok @Smanar 

Thank you

ogkita, 2020-11-01:
Hi @Smanar 

Always i restart my raspberry i need flash the raspbee ii device and add the thermostat again.

What do you think it is happeiing?

Thank you

Smanar, 2020-11-02:
You need to flash ?
You had a firmware problem ? or want to update it ?
If you just reboot the thermostat is loose again ? (wait for at least some minut after inclusion)

ogkita, 2020-11-03:
Hi @Smanar 

After reboot the raspberry i have:

[image]

[image]

When i stop deconz service and start app with UI:

   sudo systemctl stop deconz
   /usr/bin/deCONZ --http-port=80

[image]

Always is connecting to device and never get connection.

Here you have a log.
[20201103 1922 deconz.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5483246/20201103.1922.deconz.txt)

I waited 10 minutes and nothing chaned

If i flash the raspbee ii with the command "sudo GCFFlasher_internal -f deCONZ_RaspBeeII_0x26660700.bin.GCF" all work fine.

Thabnk you

ogkita, 2020-11-03:
Hi again @Smanar 

More info

After flash raspbee ii, when all is workking, i reset the termostat and it continue working fine. Then I restart deconz service and everything works correctly.

Thank you.

Smanar, 2020-11-03:
Ok so first I think you have a problem with your deconz version

> 17:22:58:072 [INFO] - Found file containing button maps. Parsing data...
> 17:22:58:074 [ERROR] - Expected 'LDSRemoteMap' in JSON file to be an object, but it isn't. Skip loading button map...
> 17:22:58:074 [ERROR] - Expected 'aqaraOpple6Map' in JSON file to be an object, but it isn't. Skip loading button map...
> 17:22:58:074 [ERROR] - Expected 'bitronRemoteMap' in JSON file to be an object, but it isn't. Skip loading button map...

There is transition of the working mode for button, and there is a problem on your .
I can't explain it, it seem you have installed the 86 version so with the update. Do you have the same error message with the official lib ?

For the second problem, after every restart you need to re-upload the firmware ?
If yes, it s critic for me as bug, for me it can't be from the api lib. But same for previous bug, if you can roll back to previous lib file to be sure.

ogkita, 2020-11-25:
Hi @Smanar 

I am sorry for the delay in my response.

I have reinstalled everything and now everything is working correctly. Thank you very much for your support.

Just one more question, this thermostat has a dry contact. Is there a possibility to have the status of this dry contact in openhab?

Thank you.

Smanar, 2020-11-25:
IDK, possible, try to enable log (now possible in deconz) and use the contact to see is something is send.

ogkita, 2020-11-27:
HI @Smanar 

Nothing change when dry contact change :(.

Did you have time to check this https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3304#issuecomment-719986690?

Thank you.

Smanar, 2020-11-27:
Nothing in log starting with "tuya debug" ?
What do I need to check in previous comment ?

ogkita, 2020-12-01:
Hi @Smanar 

About Target temperature:
Set 18.0 grades -> set 18 grades in the thermostat.
Set 18.5 grades -> set 18 grades in the thermostat.
Set 19.0 grades -> set 19 grades in the thermostat.
Set 19.5 grades -> set 19 grades in the thermostat.

Maybe work with 18.5 instead of the 18,5. "." instead of the ","??

Thank you

Smanar, 2020-12-01:
ha, ok, I understand, and no I don't thnk you can use decimal, this device use only full value
>Payload 06dd100200 04 00 00 00 0e

0x0e = 14 not 140

k3mpaxl, 2020-12-05:
Hi there, I‘ve got four BHT-002 devices connected to my conbee II, and all of them seem to not get the current time. Every day they are ahead for multiple minutes. Do you know what the problem here could  be?

Smanar, 2020-12-06:
Ha yes, this device have time cluster.
I will search some information about that.

Smanar, 2020-12-06:
Ok so tuya use it s how timer synchronisation system.
Not sure this device use it, But I will have soon a new code to test.

k3mpaxl, 2020-12-07:
Would be super cool, if we could find our how it works. Next week, I'll receive my tuya gateway. I will try sniffing the traffic, if that would be helpful.
If deconz would be the open replacement for tuya, that would really be wonderful :-)

Smanar, 2020-12-07:
Ha nice you know zigbee ^^.
And yes can be usefull, the code is finished, but it s realy do in blind mode with informations I have find on google.

You have it here 

```
sudo apt install deconz-dev
git clone --branch tuya https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

To explain what I m trying to do
- this device use the command 0x24 to do that for some device, not sure for yours
- if the device use this command, you will have in log `Tuya debug 1 : Time sync Request`
- then after a log starting with `Send Tuya Command`
- The payload is composed by 8 bytes, 4 for the UTC time and 4 for the local time.

Smanar, 2020-12-12:
@k3mpaxl do you have some return ?

Similar issues https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3407

k3mpaxl, 2020-12-12:
I had problems with compiling the code, because I'm using deconz in a docker container environment. If you have a compiled 64bit amd64 version for me, it would be great.
If not, I'll try my best next week.

Viele Grüße
Gregor

[118 more replies]

closed 2023-03-10 by github-actions[bot]