dresden-elektronik/deconz-rest-plugin #3305: #define VENDOR_CENTRALITE 0x104E // wrong name?

7wells, 2020-09-23:
## Describe the bug
I have an Osram Smart+ Motion Sensor. deCONZ shows the manufacturer ID `0x104e`. However, this is defined in `deconz-rest-plugin` with another vendor/manufacturer ("CENTRALITE", not "OSRAM"):
`#define VENDOR_CENTRALITE 0x104E // wrong name?`
The "wrong name?" comment already indicates that it is not sure that this name is correct.

## Prerequisites
Have an Osram Smart+ Motion Sensor paired to the gateway.

## Steps to reproduce the behavior
1. Go to deCONZ.
2. Click on motion sensor.
3. Look into "Note Info" field => "Manufacturer Code".
4. It shows 0x104e.
5. Go to Phoscon.
6. Go to Devices => Sensors.
7. Look under "Connected sensors".
8. The motion sensor is shown with the vendor "CentraLite" and with version "����������������".

## Expected behavior
It should show "OSRAM" (not "CENTRALITE") as vendor/manufacturer in Phoscon.
It should show a correct version in Phoscon.

## Screenshots
None

## Environment
- Version: 14.9.2020
- Host system: Raspberry Pi 2
- Running method: Raspbian
- Firmware version: 26650700
- deCONZ version: 2.05.82
- Device: RaspBee II
- Do you use an USB extension cable: NA

### deCONZ Logs
None

## Additional context
None

Smanar, 2020-09-23:
No it s ok for me.
0x110c is Osram so no problem for the manufacture number
Your device have "CentraLite" as manufacture name, but this value is written on the firmware (String value, not a number for here)

It s strange but believe me, it s worst for some chinese stuff.

7wells, 2020-09-23:
But the device is clearly from Osram, so I wonder if this should not be corrected.

And then there is the remaining problem with the version not correctly shown. Nothing major, of course.

If this is not possible to amend or not worth the effort, I can fully understand this, so no worries. ;)

Smanar, 2020-09-23:
It can be corrected, but for me it s not necessary as the device works fine like this, the correction will display a "bad" information if we correct it, I m using lot of hack in the code to correct thing like this, but I m trying avoid them when it s not necessary.

For the version, IDK what to do, it s not a bad reading, you realy have this string composed by the same value "\ufffd", so I can't find a god value to use.

SwoopX, 2020-09-23:
This is nothing oof any relevance so don't worry. You could also name it "VENDOR_BANANA" and it wouldn't matter as the variable value is all that counts for use. Also take note that the hardware manufacturer must not equal the device vendor and that's one of the reasons the model on the packaging doesn't really matter to us.
With regard to the version information, there's nothing we can do about it as the device is just sending what the firmware dictates.

closed 2020-09-23 by SwoopX