dresden-elektronik/deconz-rest-plugin #3306: New device support request: Linkind Motion Sensor / ZS110040078

fluffyironboots, 2020-09-23:
Hi,

I just bought some new motion sensors from Linkind.
https://www.linkind.com/product/pir-motion-sensor-zigbee/
https://www.amazon.de/Linkind-Bewegungssensor-intelligenter-Bewegungsmelder-Tageslichtsensor/dp/B07YFCHYYK/ref=sr_1_1?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=linkind+motion&qid=1600869875&sr=8-1

The GU10 CCT lamps from linkind are also not on the official support list but they work flawlessly in Phoscon so i thought i just try the sensor.

Data and Screenshots:
Model ZS110040078
FCC ID: 2ASHKZS110040078
[image]
[image]

Let me know if you need more info.

Thanks,
Joe

Mimiix, 2020-09-24:
@SwoopX Could you verify the above is enough? If not, please close this issue.

@fluffyironboots **If Swoop says this is enough to go on, i'll add the device Request label. Otherwise it gets closed. Following the Issue template is mandatory.**

fluffyironboots, 2020-09-24:
Just had time to test some more.
Got the Basic and PowerConfiguration Cluster Info to successfully read.

[image]
[image]

SwoopX, 2020-09-24:
Info provided is sufficient.

Mimiix, 2020-09-25:
@fluffyironboots I'll keep it open and added the label. For the next request, make sure to use the template.

fluffyironboots, 2020-09-25:
@Mimiix Thank you for your help. I don't even now how i opened the issue without the template. I clicked on new issue and got right to an empty input without the option to choose a template. Next time i'll do better.

closed 2020-11-14 by manup