dresden-elektronik/deconz-rest-plugin #3307: Aqara Plug power reporting false

Jey-Cee, 2020-09-23:
## Describe the bug
The aqara plug reports power on two different websocket messages, but one says all the time 0.

## Expected behavior
On both messages the value hast to be the same or it hast to be clarified for which purposes the two different values are.

## Log for Websocket
* (3445) Websocket message: {"e":"changed","id":"44","r":"sensors","state":{"consumption":144200,"lastupdated":"2020-09-15T11:59:55.427","power":0},"t":"event","uniqueid":"xx:xx..."}
* (3445) Websocket message: {"e":"changed","id":"45","r":"sensors","state":{"current":0,"lastupdated":"2020-09-15T12:01:16.117","power":124,"voltage":0},"t":"event","uniqueid":"xx:xx"}

## Environment
- deConz version 2.05.80 & 2.05.82
-  Aqara Plug SP-EUC01

## Additional context
The same Websocket messages for a Bitron, Xiaomi Smart Plug and Blitzwolf SHP13 deliveres the right values.
[Originial Issue (German)](https://github.com/iobroker-community-adapters/ioBroker.deconz/issues/168)

Smanar, 2020-09-23:
Hello, I think the UniqueID will be usefull ^^, at least the last number, to know the used cluster.

basty2, 2020-09-23:
Hi, here we have the entire lines: 
Websocket message: {"e":"changed","id":"44","r":"sensors","state":{"consumption":156616,"lastupdated":"2020-09-22T22:00:45.439","power":0},"t":"event","uniqueid":"xx:xx:xx:xx:3c:76:46:58-01-0702"}
Websocket message: {"e":"changed","id":"45","r":"sensors","state":{"current":0,"lastupdated":"2020-09-22T22:00:49.243","power":157,"voltage":0},"t":"event","uniqueid":"xx:xx:xx:xx:3c:76:46:58-01-0b04"}

SwoopX, 2020-09-23:
That is the simple metering cluster reporting 0. This happens if the corresponding attribute is not supported. Which device is this? We don't need the vendor model, but the zigbee modelID.

basty2, 2020-09-23:
{"attr":{"lastannounced":"2020-09-15T14:00:55Z","lastseen":"2020-09-23T01:47Z","manufacturername":"LUMI","modelid":"lumi.plug.maeu01","name":"Aqara Plug1","swversion":"09-10-2019","type":"Smart plug","uniqueid":"xx:xx:xx:xx:3c:76:46:58-01"},"e":"changed","id":"3","r":"lights","t":"event","uniqueid":"xx:xx:xx:xx:3c:76:46:58-01"}

or do you need the whole uniqueid?

SwoopX, 2020-09-24:
That's already enough, thanks. As I thought, the device is currently not blacklisted. Lemme take care of that.

ysoudy, 2020-09-25:
Same issue for me.
I use Jeedom v4 with d
Conbee II + deConz Version 2.5.81 & Firmware 0x26650700

I noticed the false report is launch every 50min...
Look my scenario log :
[2020-09-24 20:43:46][SCENARIO] Start
[2020-09-24 21:33:53][SCENARIO] Start
[2020-09-24 22:24:01][SCENARIO] Start
[2020-09-24 23:14:04][SCENARIO] Start
[2020-09-25 00:04:11][SCENARIO] Start
[2020-09-25 00:54:18][SCENARIO] Start
[2020-09-25 01:44:25][SCENARIO] Start
[2020-09-25 02:34:32][SCENARIO] Start
[2020-09-25 03:24:38][SCENARIO] Start
[2020-09-25 04:14:45][SCENARIO] Start
[2020-09-25 05:04:50][SCENARIO] Start
[2020-09-25 05:54:54][SCENARIO] Start
[2020-09-25 06:45:00][SCENARIO] Start
[2020-09-25 07:35:07][SCENARIO] Start
[2020-09-25 08:25:14][SCENARIO] Start
[2020-09-25 09:15:19][SCENARIO] Start
[2020-09-25 10:05:23][SCENARIO] Start

SwoopX, 2020-09-25:
@ysoudy You do not have the same issue. This is about a power plug having an attribute not blacklisted, so please do not hijack. If you feel you have an issue, feel free to raise a seperate issue.

Smanar, 2020-09-26:
@ysoudy IDK what is the event you are monitoring, but Xiaomi device send a notification every hours, so if you just checking for that, the reaction is normal.

closed 2020-10-01 by manup

basty2, 2020-10-08:
@SwoopX Unfortunately, the bug isn't fixed yet. I just installed the 84 Version.

SwoopX, 2020-10-08:
Of course not as the fix is not included and therefore mentioned in the release notes.