dresden-elektronik/deconz-rest-plugin #3308: Sengled Smart Light Switch

stevenwfoley, 2020-09-24:
## Device
- Product name: Sengled Smart Light Switch
- Manufacturer: sengled
- Model identifier: E1E-G7F
- Device type : Switch

## Screenshots
[image]
[image]
[image]
[image]

closed 2020-11-15 by manup

SwoopX, 2020-12-17:
@stevenwfoley We would still need the button events to create a fitting button map. See here for reference
#3397

stevenwfoley, 2020-12-17:
Is this the data you're looking for?
Top (ON) button - No button map for: E1E-G7F endpoint: 0x01 cluster: 0xFC10 command: 0x00 payload[0]: 001
Second from top (Dim Up) button - No button map for: E1E-G7F endpoint: 0x01 cluster: 0xFC10 command: 0x00 payload[0]: 002
Third from top (Dim Down) buton - No button map for: E1E-G7F endpoint: 0x01 cluster: 0xFC10 command: 0x00 payload[0]: 003
Bottom (OFF) button - No button map for: E1E-G7F endpoint: 0x01 cluster: 0xFC10 command: 0x00 payload[0]: 004

SwoopX, 2020-12-17:
Hm, interesting. I would have rather expected to see comparable messages from on/off and level control cluster instead of FC10. On/off and level control are typically used to control regular lights and also should allow to used zigbee groups. Maybe the FCC10 commands are used to communicate differently with Sengled lights...
Could you please double check?

stevenwfoley, 2020-12-17:
I don't think the switch supports the zigbee group cluster as its greyed out in the deconz gui. I reran the debug and received the same results. Is there something else you want to test or provide?

SwoopX, 2020-12-18:
It just feels a bit strange that this switch seems to behave significantly differently than all others, that's what I'm trying to understand here. Could also be that this is something device specific, which can be configured via the unknown FC clusters. However, a map based on that is easily doable...

Since you were asking regarding additional info: I'd like to see a full debug output for a timeframe 10 secs before and after a button press. You can now do this via deconz GUI (_Help -> Debug view_) in case you didn't knew already- Please only use those level for that
[image]

Additionally, it would be great to have another screenshot from the node itself with the expanded clusters and ideally, the REST API output for this particular switch. Thanks!

stevenwfoley, 2020-12-18:
I'm running a production zigbee environment, so my debug log is a bit noisy, but here it is. This is about 10 seconds before and after of pressing all 4 buttons in sequence from top to bottom.

20:52:43:737 schedule 4 diff 7637, Thu Dec 17 23:00:00 2020
20:52:43:738 enqueue event config/localtime for /config/
20:52:43:739 poll node 00:17:88:01:02:22:0c:93-0b
20:52:43:740 Poll light node Master Bedroom 1
20:52:43:741 Idle timer triggered
20:52:43:742 Force binding of attribute reporting for node Living Room Lamp 2
20:52:43:743 Force read attributes for ZHALightLevel SensorNode Motion Sensor
20:52:43:769 rule event /config/localtime: 20:52:42.737 -> 20:52:43.738 (1001ms)
20:52:44:182 read attributes of 0x0017880102220C93 cluster: 0x0006: [ 
20:52:44:184 0x0000 
20:52:44:184 ]
20:52:44:185 add task 197 type 19 to 0x0017880102220C93 cluster 0x0006 req.id 52
20:52:44:186 Poll APS request 52 to 0x0017880102220C93 cluster: 0x0006
20:52:44:195 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 7, node: 0xADDA
20:52:44:284 Poll APS confirm 52 status: 0x00
20:52:44:285 Erase task req-id: 52, type: 19 zcl seqno: 41 send time 0, profileId: 0x0104, clusterId: 0x0006
20:52:44:316 Node data 0x0017880102220c93 profileId: 0x0104, clusterId: 0x0006
20:52:44:318 0x0017880102220C93: update ZCL value 0x0B/0x0006/0x0000 after 0 s
20:52:44:319 enqueue event event/checkgroupanyon for /groups/25424
20:52:44:320 enqueue event event/checkgroupanyon for /groups/5
20:52:44:321 enqueue event event/checkgroupanyon for /groups/65520
20:52:44:670 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 7, node: 0x5E30
20:52:44:737 schedule 4 diff 7636, Thu Dec 17 23:00:00 2020
20:52:44:739 enqueue event config/localtime for /config/
20:52:44:753 rule event /config/localtime: 20:52:43.738 -> 20:52:44.739 (1001ms)
20:52:45:297 poll node 00:17:88:01:02:6f:ca:05-0b
20:52:45:298 Poll light node Master Bedroom 2
20:52:45:346 read attributes of 0x00178801026FCA05 cluster: 0x0006: [ 
20:52:45:347 0x0000 
20:52:45:348 ]
20:52:45:349 add task 203 type 19 to 0x00178801026FCA05 cluster 0x0006 req.id 63
20:52:45:349 Poll APS request 63 to 0x00178801026FCA05 cluster: 0x0006
20:52:45:379 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 7, node: 0x1519
20:52:45:488 Poll APS confirm 63 status: 0x00
20:52:45:489 Erase task req-id: 63, type: 19 zcl seqno: 42 send time 0, profileId: 0x0104, clusterId: 0x0006
20:52:45:525 Node data 0x00178801026fca05 profileId: 0x0104, clusterId: 0x0006
20:52:45:527 0x00178801026FCA05: update ZCL value 0x0B/0x0006/0x0000 after 0 s
20:52:45:529 enqueue event event/checkgroupanyon for /groups/25424
20:52:45:530 enqueue event event/checkgroupanyon for /groups/5
20:52:45:531 enqueue event event/checkgroupanyon for /groups/65520
20:52:45:737 schedule 4 diff 7635, Thu Dec 17 23:00:00 2020
20:52:45:739 enqueue event config/localtime for /config/
20:52:45:758 Node data 0x841826000005dc12 profileId: 0x0104, clusterId: 0x0006
20:52:45:761 0x841826000005DC12: update ZCL value 0x03/0x0006/0x0000 after 0 s
20:52:45:763 enqueue event event/checkgroupanyon for /groups/12
20:52:45:765 enqueue event event/checkgroupanyon for /groups/65520
20:52:45:766 enqueue event event/checkgroupanyon for /groups/22137
20:52:45:767 ZCL attribute report 0x841826000005DC12 for cluster: 0x0006, ep: 0x03, frame control: 0x18, mfcode: 0x0000 
20:52:45:769 	payload: 00001000
20:52:45:771 rule event /config/localtime: 20:52:44.739 -> 20:52:45.739 (1000ms)
20:52:46:589 poll node 00:17:88:01:02:47:c4:1c-0b
20:52:46:590 Poll light node Dining Room 1
20:52:46:639 read attributes of 0x001788010247C41C cluster: 0x0006: [ 
20:52:46:640 0x0000 
20:52:46:642 ]
20:52:46:643 add task 209 type 19 to 0x001788010247C41C cluster 0x0006 req.id 74
20:52:46:644 Poll APS request 74 to 0x001788010247C41C cluster: 0x0006
20:52:46:737 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
20:52:46:740 schedule 4 diff 7634, Thu Dec 17 23:00:00 2020
20:52:46:741 enqueue event config/localtime for /config/
20:52:46:780 rule event /config/localtime: 20:52:45.739 -> 20:52:46.741 (1002ms)
20:52:46:847 Poll APS confirm 74 status: 0x00
20:52:46:849 Erase task req-id: 74, type: 19 zcl seqno: 43 send time 0, profileId: 0x0104, clusterId: 0x0006
20:52:46:891 Node data 0x001788010247c41c profileId: 0x0104, clusterId: 0x0006
20:52:46:893 0x001788010247C41C: update ZCL value 0x0B/0x0006/0x0000 after 0 s
20:52:46:895 enqueue event event/checkgroupanyon for /groups/6996
20:52:46:897 enqueue event event/checkgroupanyon for /groups/6
20:52:46:899 enqueue event event/checkgroupanyon for /groups/65520
[... 474 lines cut ...]
20:53:09:740 schedule 4 diff 7611, Thu Dec 17 23:00:00 2020
20:53:09:741 enqueue event config/localtime for /config/
20:53:09:742 poll node 00:17:88:01:03:39:b7:1a-0b
20:53:09:743 Poll light node Basement 2
20:53:09:744 Idle timer triggered
20:53:09:745 Force binding of attribute reporting for node Basement 2
20:53:09:746 Force read attributes for ZHATemperature SensorNode Motion Sensor
20:53:09:782 rule event /config/localtime: 20:53:08.743 -> 20:53:09.741 (998ms)
20:53:10:038 read attributes of 0x001788010339B71A cluster: 0x0006: [ 
20:53:10:039 0x0000 
20:53:10:040 ]
20:53:10:040 add task 328 type 19 to 0x001788010339B71A cluster 0x0006 req.id 31
20:53:10:041 Poll APS request 31 to 0x001788010339B71A cluster: 0x0006
20:53:10:189 Poll APS confirm 31 status: 0x00
[... cut ...]
20:53:10:221 enqueue event event/checkgroupanyon for /groups/12
20:53:10:223 enqueue event event/checkgroupanyon for /groups/22137
20:53:10:737 schedule 4 diff 7610, Thu Dec 17 23:00:00 2020
20:53:10:739 enqueue event config/localtime for /config/
20:53:10:764 rule event /config/localtime: 20:53:09.741 -> 20:53:10.739 (998ms)
20:53:10:876 Bind response success for 0x7cb03eaa00a6b24c ep: 0x03 cluster: 0x0006
20:53:10:878 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x7CB03EAA00A6B24C (seems to be active)

stevenwfoley, 2020-12-18:
Full set of screenshots:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

stevenwfoley, 2020-12-18:
REST API:

{
  "config": {
    "battery": 100,
    "group": "30",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "f055d820d4380cc445a16f055b86082d",
  "lastseen": "2020-12-18T02:08Z",
  "manufacturername": "sengled",
  "mode": 1,
  "modelid": "E1E-G7F",
  "name": "sengled",
  "state": {
    "buttonevent": null,
    "lastupdated": "none"
  },
  "type": "ZHASwitch",
  "uniqueid": "b0:ce:18:14:03:5e:08:01-01-0006"
}

stevenwfoley, 2020-12-18:
Anything else I can provide to help support this device?

kyletpugh, 2021-01-23:
I am also trying to get support for this device.

SwoopX, 2021-01-23:
Well, tbh, I got no clue if you can configure the device to work like "any other switch" so you can directly steer arbitrary lights. In the current situation, this presumably works for Sengled lights only. If you guys are up to some play in this regard, lemme know.

Meanwhile, the best we can make out of it is creating a button map, so that at least such events are emitted via websocket and usable in whatever home automation system. This here is a good starting point to get it basically working. However, there mit be more dwelling in the depths, meaning long presses and double presses. You guys would need to provide that information, and best not inly the `[INFO] - XYZ` messages, but better also the 15 leading and trailing messages + noting down what you did on what button.

kyletpugh, 2021-01-23:
I have connected this to two other hubs without issue, I have this switch controlling 3 Sengled LEDs, a Hue Ambiance, and a Samsung LED. I am trying to move over to my own hub. The only requirement for the switch is a zigbee hub, I am not sure why the buttons do not send the proper messages though. Everything is like @stevenwfoley said except the on and off buttons can be double pressed and held as well. The brighten and dim buttons just repeat on a second interval if held. Here are the messages.
On, Press: `[INFO] - No button map for: E1E-G7F endpoint: 0x01 cluster: 0xFC10 command: 0x00 payload[0]: 001`
On, Double Press: `[INFO] - No button map for: E1E-G7F endpoint: 0x01 cluster: 0xFC10 command: 0x00 payload[0]: 005`
On, Hold: `[INFO] - No button map for: E1E-G7F endpoint: 0x01 cluster: 0xFC10 command: 0x00 payload[0]: 006`
Off, Press: `[INFO] - No button map for: E1E-G7F endpoint: 0x01 cluster: 0xFC10 command: 0x00 payload[0]: 004`
Off, Double Press: `[INFO] - No button map for: E1E-G7F endpoint: 0x01 cluster: 0xFC10 command: 0x00 payload[0]: 007`
Off, Hold: `[INFO] - No button map for: E1E-G7F endpoint: 0x01 cluster: 0xFC10 command: 0x00 payload[0]: 008`
Brighten: `[INFO] - No button map for: E1E-G7F endpoint: 0x01 cluster: 0xFC10 command: 0x00 payload[0]: 002`
Dim: `[INFO] - No button map for: E1E-G7F endpoint: 0x01 cluster: 0xFC10 command: 0x00 payload[0]: 003`
Thanks.

SwoopX, 2021-01-23:
Although it feels really odd that there's only the FC10 cluster creating those info messages (when you can control other brands as well, then it should use the expected commands as broadcasts), I've created a corresponding button map.
Usually, you would be able to add that directly to the file yourself, but unfortunately, a too tight check in the code prevents that. However, should be all available with the next release.

MoshiBin, 2021-01-25:
I tried creating a button map for this, but since cluster FC10 is such a high value, read_files.cpp refuses to load it: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/77856034258ab2fe649178abcb825ec4b21f0d50/read_files.cpp#L77

I still haven't had the time to recompile with this check modified.

In button_maps.json I've added this to `clusters`:
```
"SENGLED": 64528
```
And this to `commands`:
```
        "SENGLED": {
            "CLICK": 0
        }
```

So this is the map:
```
        "sengledMap": {
            "vendor": "sengled",
            "doc": "Sengled Smart Light Switch E1E-G7F",
            "modelids": ["E1E-G7F"],
            "map": [
                [1, "0x01", "SENGLED", "CLICK", "1", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "SENGLED", "CLICK", "2", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "DimUp Press"],
                [1, "0x01", "SENGLED", "CLICK", "3", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "DimDown Press"],
                [1, "0x01", "SENGLED", "CLICK", "4", "S_BUTTON_4", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x01", "SENGLED", "CLICK", "5", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "On (0)"],
                [1, "0x01", "SENGLED", "CLICK", "6", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "On (1)"],
                [1, "0x01", "SENGLED", "CLICK", "7", "S_BUTTON_4", "S_BUTTON_ACTION_DOUBLE_PRESS", "Off (0)"],
                [1, "0x01", "SENGLED", "CLICK", "8", "S_BUTTON_4", "S_BUTTON_ACTION_HOLD", "Off (1)"]
            ]
        }
```

kyletpugh, 2021-01-26:
I just realized that this device creates some sort of broken group that is not useable or visible. I tried to add lights via the REST API but it shows up in HA as an unknown device and does not even show up in phoscon app.

```
  "14": {
    "action": {
      "alert": "none",
      "bri": 127,
      "colormode": "hs",
      "ct": 0,
      "effect": "none",
      "hue": 0,
      "on": false,
      "sat": 127,
      "scene": null,
      "xy": [
        0,
        0
      ]
    },
    "devicemembership": [
      "2"
    ],
    "etag": "3955b6bd049f75782555dcf3784d7348",
    "id": "14",
    "lights": [],
    "name": "E1E-G7F",
    "scenes": [],
    "state": {
      "all_on": false,
      "any_on": false
    },
    "type": "LightGroup",
    "uniqueid": "b0:ce:18:14:03:5e:03:10"
  },
```

SwoopX, 2021-01-26:
You'd need to make the device assignment in the old webapp.

kyletpugh, 2021-01-26:
The group does nothing, I already added the lights in the old app but removed them since it doesn't work. But I can not remove this group. It shows up in HA as an unknown device, it doesn't show up in the phoscon app.

kyletpugh, 2021-02-12:
This fix send to send events but it doesn't allow the switch to be set to control a group and the event is always the same one no matter what you press on the switch. Is there more fixes coming?