dresden-elektronik/deconz-rest-plugin #3309: Instability with 0x26650700 firmware update

ReX1983, 2020-09-24:
## Describe the bug

Since the update to the 0x26650700 firmware and Phoscon 2.05.82 my network became much less stable. 
One of my Aqara wall switches (lumi.ctrl_neutral2, 11-25-2017) toggled on its own 3 times in 2 days.
Many other switches (same brand and model) became not reachable and then started working again, after some time (without the need for a powercycle).

The most affected switches are the ones far away from the ConBee II.

Is it possible the instability is caused by a lower power transmission setting? 

## Steps to reproduce the behavior

1. Update to 0x26650700 firmware and Phoscon 2.05.82
2. Wait some time and try to turn on/turn off the switches from Home Assistant

## Expected behavior

The Switches are reachable 24/7.

## Screenshots

## Environment

 - Host system: (Raspberry Pi 3)
 - Running method: (Marthoc Docker container)
 - Firmware version: (26650700)
 - deCONZ version: (2.05.82 y)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes) -- only relevant for ConBee I/II

### deCONZ Logs

## Additional context

FunkOdyssey2001, 2020-09-25:
Same problems here. 
All bulbs have a lot of "green connections" in  deCONZ, but they are often suddenly not reachable.

ReX1983, 2020-09-25:
@Mimiix, I think this should be flagged as Bug report, as the the connectivity issues started with this firmware.  My question was just a suggestion on what could be a cause (but I start thinking this might be due to other reasons, e.g. source routing).

LeoeLeoeL, 2020-09-27:
I confirm the instability.
Some devices (always the same ones) change to unavailable and then come back available.
A lamp switches off without any command.
When I switch on the kitchen by phisical switch the Osram TW3 switches on too!!!

Mimiix, 2020-09-28:
I forgot to reply here after my label change. Sorry!

Any reports of problems with a firmware update should be reported in the general issue. Please check https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3260 .

iz3man, 2020-09-29:
I see similar issues. After a restart of homeassistant ALL of my repeater capable devices become unavailable. SOMEHOW. They are still reporting wattage etc, but the SWITCH functionality is gone. It's just not available anymore.
This includes: IKEA Tradfri bulbs, INNR SP120 power sockets, CC2591 repeater, and also the Blitzwolf power sockets.

[image]

[image]

Mimiix, 2020-09-29:
@iz3man Same goes for you:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3260

(If this is related to the firmware update). Otherwise: Open your own issue.

LeoeLeoeL, 2020-09-30:
I think the issue (or a part) is due Ikea bulbs.

rasssta, 2020-10-02:
I've had issues as well with the ConBee II. Had to send pretty much every command twice for it to go through with 26650700. Reverted to 26580700 and it works like a charm again.

alekslyse, 2020-10-06:
I can almost guarantee 100% the latest firmware is buggy. Firstly when I updated one of the stick is will not be recognized at all in unraid (its slackware) - not supported so that is acceptable, but the stick wont even register as with older firmwares. Thats a problem as so many use unraid as a NAS with docker or VM to run deconz. 

Then to rule out the the slacware issue I put it in hassos on a NUC and the builtin deconz - works fine as recognised, but deconz with 26650700 the stick can never decide routes so mesh break down after 5-10 minutes. What I did today was to take my second stick (production stick) with the 26580700 and plugged into the hassos. No routing issues. Tried the newest firmware on both sticks with same results and now over many installations. Going to let this stick run now for a couple of days on the old firmware (old = 26580700) and see if its stable.

For record purposes its the 26650700 that makes the issues

This is now tested in the following os and not at the exactly same spot in the house on
- Unraid (unsupported, but acceptable, even worked on old firmware)
- Docker marthoc - mesh breakdown
- Hassos on bare NUC (no underlaying OS) - mesh breakdown
- Windows 10 - mesh breakdown

Tetsted on two conbee II to rule out stick issues

All four switching to previous firmware no mesh breakdown

LeoeLeoeL, 2020-10-14:
I updated to .84 and new firmware but the issue is still present.

LeoeLeoeL, 2020-10-16:
I realized the issue (probably in my case) is unleashed by physical buttons.
When I switch on a lamp by physical switch (giving it current), some lamps light on and others light off. There is also a smartplug that switched off.

ReX1983, 2020-10-17:
> I realized the issue (probably in my case) is unleashed by physical buttons.
> When I switch on a lamp by physical switch (giving it current), some lamps light on and others light off. There is also a smartplug that switched off.

I think you found the root cause. 

I have noticed that when a device that acts as router reconnect to the network, other devices trigger. 

In my case it happens when trådfri lights go offline and return online. After this event few xiaomi switches trigger turn on.

manup, 2020-10-18:
That's really strange behavior, we need to to more testing to recreate the effect. If it can be recreated it seems to have more to do with the deCONZ than the firmware, as the firmware doesn't do anything on it's own.

Is this "ghost" toggling only seen for the Xiaomi switches (lumi.ctrl_neutral2, ...)?

LeoeLeoeL, 2020-10-18:
Device affected are in the same room of Conbee II and are:
Osram adapter plug (V 1.05.09); It switched off.
Ledvance Undercabinet TW Z3 (00103101); It switched on.
icasa ICZB-IW11D (2.5.3_r47) - a rebrand of Sunricher; It switched off.

LeoeLeoeL, 2020-10-19:
Yesterday I began a test. I unplugged my 3 Ikea signal repeaters E1746.
it passed 19 hours and the issue didn't show.

manup, 2020-10-19:
Please also check out latest 2.5.86 release, it has a change for IKEA lights which might be related.

ReX1983, 2020-10-19:
> Please also check out latest 2.5.86 release, it has a change for IKEA lights which might be related.

I'll try it, thanks. Is 0x26650700 still the latest firmware for Conbee II?

manup, 2020-10-19:
Almost, 0x26660700 is the latest version. The most recent versions are always referenced in https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

ReX1983, 2020-10-19:
> Almost, 0x26660700 is the latest version. The most recent versions are always referenced in https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Thanks. Just updated to the latest firmware and 2.05.85 (2.5.86 is not yet available as Marthoc Docker Image).

ReX1983, 2020-10-19:
On 2.05.85 all my battery operated aqara wall switches stopped working. Is this the bug addressed by 2.05.86?

manup, 2020-10-19:
Are there any lights/routers powered off? Aqara devices are very picky to their parent routers, when they are powered off commands might not be routed to the coordinator anymore.

ReX1983, 2020-10-19:
> Are there any lights/routers powered off? Aqara devices are very picky to their parent routers, when they are powered off commands might not be routed to the coordinator anymore.

I have tried all the lights, switches and buttons and can confirm that all my Aqara battery operated wall switches stopped working (both single button and double). The no neutral Aqara wall switches, the Xiaomi magic buttons and the Aqara plugs work as usual. Also the Ikea lights work.

All the routers (ikea lights and aqara plugs) are on and work correctly. 

Any idea?

SwoopX, 2020-10-19:
Which one? This is like saying my car is broke 🤷‍♂️

ReX1983, 2020-10-19:
> Which one? This is like saying my car is broke 🤷‍♂️

As above. Specifically:
```
Vendor Xiaomi
Product Transmitter 2-gang WXKG02LM
Version 20160517
```

and

```
Vendor Xiaomi
Product Transmitter 1-gang WXKG03LM
Version 20160517
```

Don't send any event to Home Assistant since I updated to 2.05.85.

Smanar, 2020-10-19:
> 
> 
> Which one? This is like saying my car is broke 🤷‍♂️

Take the bus ^^

LeoeLeoeL, 2020-10-19:
> Please also check out latest 2.5.86 release, it has a change for IKEA lights which might be related.

It seems fixed.
I paired again 2 Ikea dimming bulb too.

SwoopX, 2020-10-19:
Yes, those 2 switches were affected. I wonder that a re-pair was necessary, should have worked without it.

ReX1983, 2020-10-19:
> As above. Specifically:
>
> ```
> Vendor Xiaomi
> Product Transmitter 2-gang WXKG02LM
> Version 20160517
> ```
>
> and
>
> ```
> Vendor Xiaomi
> Product Transmitter 1-gang WXKG03LM
> Version 20160517
> ```
>
> Don't send any [...]

Downgraded to .84 and the switches returned to work.

LeoeLeoeL, 2020-10-20:
> > Please also check out latest 2.5.86 release, it has a change for IKEA lights which might be related.
> 
> It seems fixed.
> I paired again 2 Ikea dimming bulb too.

Not fixed.
Today at 8.43.29 I switched on (by physical switch) a hue bulb and 5 seconds later Ledvance Undercabinet TW Z3 lit on.

Mimiix, 2020-10-20:
@LeoeLeoeL Can you open a own issue on this? Not a lot of info to go on on this one. 

Please add logs if you can ^^

LeoeLeoeL, 2020-10-20:
How can I get logs ? Deconz runs on Windows 10.

LeoeLeoeL, 2020-11-15:
Updated to .88.
Now some lamps switch on alone too.
Few minutes ago the icasa ICZB-IW11D started to flash as the IDENTIFY command but non only for 5 times.

LeoeLeoeL, 2020-12-08:
After so long time I can say:
1) HUE bulbs are not affected by this issue. Devices affected (in my network) are: Osram/Ledvance bulbs and plugs, icasa ICZB-IW11D and now (just bought) 3A Smart Home DE (ModelLXN59-2S7LX1.0). IKEA bulbs and repeaters are not in the network for the moment.
2) The issue doesn't tell some devices to switch on but probably gives the command to power on.
I suppose this because I have configured ICasa dimmer to stay OFF after a blackout and, when the issue appears, the lamp of the dimmer switches off.

closed 2021-02-08 by Mimiix