dresden-elektronik/deconz-rest-plugin #3310: After FW upgrade, conbee II stops working with message 'failed to set dtr/rts'

drdxenakis, 2020-09-24:
## Describe the bug
After upgrading Conbee II firmware, it stops communicating.  So nothing works...<br/>

## Steps to reproduce the behavior

1. After updating Deconz software, I went to the Phoscon application to check that everything was OK. Indeed it was, but the Phoscon application showed that there is a newer version of firmware for the Conbee II . <br/>
2. I proceeded as instructed and it finished the process of upgrading with a **Success** message. I noticed that the new version on the Phoscon page was 0000002 (not sure how many zeros) which was peculiar. <br/>
3. Immediately, all lights disappeared from Phoscon, sensors and switches remained but not functional. <br/>
4. Rebooted, noticed that Phoscon showed that the Conbee was not connected<br/> 
5. followed instructions on https://www.phoscon.de/en/support#conbee2-connection-issues but nothing.
6. While trying to follow instructions above, I noticed that /dev/ttyACM0 appears and disappears. That led me to check dmesg which gave repeated messages (in section below the full typical output)  `failed to set dtr/rts`.
7. Since then I have removed and re-inserted numerous times, removed all USB (except the keyboard dongle), inserted the Conbee II in another computer (same message `failed to set dtr/rts`)
8. Conbee II is not functioning.
 

## Expected behavior

## Screenshots

[image]

## Environment

 - Host system: PC
 - Running method: Ubuntu
 - Firmware version: was 264A0700
 - deCONZ version:  2.05.81
 - Device: ConBee II
 - Do you use an USB extension cable: yes

### deCONZ Logs

## Additional context

**dmesg**
```
$dmesg -H -w
[  +8,513145] usb 2-3.3: new full-speed USB device number 114 using ehci-pci
[  +0,210511] usb 2-3.3: New USB device found, idVendor=1cf1, idProduct=0030
[  +0,000003] usb 2-3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0,000001] usb 2-3.3: Product: ConBee II
[  +0,000002] usb 2-3.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[  +0,000002] usb 2-3.3: SerialNumber: DE2123643
[  +0,000436] cdc_acm 2-3.3:1.0: ttyACM1: USB ACM device
[  +4,585809] usb 2-3.3: USB disconnect, device number 114
[  +0,002219] cdc_acm 2-3.3:1.0: failed to set dtr/rts
```
**lsusb**
```
$ lsusb
Bus 002 Device 005: ID 045e:0745 Microsoft Corp. Nano Transceiver v1.0 for Bluetooth
Bus 002 Device 004: ID 048d:1336 Integrated Technology Express, Inc. SD/MMC Cardreader
Bus 002 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 012 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 011 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 009 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 010 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```
Attempted to re-install the firmware (several times) with this message:
```
$ GCFFlasher_internal -d /dev/ttyACM0 -f ~/tmp/deCONZ_ConBeeII_0x26580700.bin.GCF -x 3
GCFFlasher V3_15 (c) dresden elektronik ingenieurtechnik gmbh
19:23:43:564 using firmware file: /home/nkdx/tmp/deCONZ_ConBeeII_0x26580700.bin.GCF 
19:23:43:607 ls dev: /dev/ttyACM0 (0x1CF1/0x0030) sn: DE2123643
19:23:43:607 ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
Reboot device /dev/ttyACM0 (ConBee II)
19:23:43:613 query bootloader v1 ID after 5 ms
19:23:43:625 RX 60 bytes ASCII 
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
, 08:48:53
 after 17 ms
19:23:43:625 bootloader start after 17 ms
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
19:23:43:626 GCF_ResetDeviceDone
19:23:43:627 bootloader v1 update firmware
flashing 160930 bytes: |===error: timeout flashing firmware after 3001 ms
```

Smanar, 2020-09-24:
Hello, can you try with the command that use "-t 60" ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

drdxenakis, 2020-09-24:
Thank you very much, I executed `sudo systemctl stop ModemManager` as instructed in https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-ubuntu and managed to update the firmware successfully. ヽ(•‿•)ノ
Maybe it's only me having this problem, but in case I am not, I hope whoever has control over the https://www.phoscon.de/en/support#conbee2-connection-issues page may consider adding this!!!

Thanks again Smanar, I won't need to switch on the coffee machine in the morning, phoscon will manage! 

> Hello, can you try with the command that use "-t 60" ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

closed 2020-09-24 by drdxenakis