dresden-elektronik/deconz-rest-plugin #3315: OSRAM Smart+ Candle E14 (B40 DIM Z3) - loosing coordinator after power off

SAOPP, 2020-09-25:
Hi!

Guys, OSRAM E14 model B40 DIM Z3. The first time I ran into a problem a few weeks ago - turning off this bulb and turning it on again, I had to reconnect it. Tonight there was a short power outage, and again, the lamp was like from the factory... I had to re-pair it again.

In this regard, the question is, does anyone have the same behaviors with this device too? And the question to the developers, what data is worth sharing based on the above?

Right now, the device is connected to the conbee2 with phoscon 2.05.82 / 14.09.2020 and firmware is 264A0700 in docker at x86.

[image]

[image]

Thanks.

Mimiix, 2020-09-25:
Could it be that the device is broken? Maybe it doesn't safe the settings on itself?

SAOPP, 2020-09-25:
I thought about it too. I do not have a second such lamp, there are two with warm white light. This device is dimmable warm only.

closed 2020-10-25 by stale[bot]