dresden-elektronik/deconz-rest-plugin #3319: ptvo switch relay

moshiko2312, 2020-09-26:
Hey
First of all thank you so much for the work you do it is just amazing

I use an 8/4 Relay box
My main use is in the Home Assistant system with a CONBEE 2 adapter
* I manage to pair the product and even see it configured in the system
The problem is when I turn on the relay after 30 seconds the on-screen indication shows that the relay is off (actually the relay is still on)
It is possible to get a solution to this issue
Thank you

I enclose some links on the relevant product

[https://www.zigbee2mqtt.io/devices/ptvo.switch.html](url)

[https://ptvo.info/zigbee-switch-configurable-firmware-router-199/](url)

[https://ptvo.info/zigbee-switch-configurable-firmware-v2-210/](url)
[image]
[image]
[image]
[image]
[image]

kachalex, 2020-09-26:
I have this problem too,  status does not show correctly.
I will be very glad, if you solve this problems.
With ziqbee2mqqt works well.

Mimiix, 2020-09-26:
@moshiko2312 I am really tempted to close this one.

Please fill the template details and i am willing to add the label.

moshiko2312, 2020-09-27:
Hey
I'm really sorry, this is the second time I'm opening, I do not really know or can change the label.
Can I have an explanation?

closed 2020-09-27 by moshiko2312

Mimiix, 2020-09-28:
@moshiko2312 When you create a issue, it presents you with 4 options. After selecting the right one for your needs, it already contains text. That text needs to be adjusted to your information.