dresden-elektronik/deconz-rest-plugin #3321: How to get optimal fluency when dimming?

DIYtechie, 2020-09-26:
## Describe the question or issue you are having
I currently use a [dimming flow in Node-RED](https://flows.nodered.org/flow/1949af43de3db8133b039dac706372bd/in/YsSoz9IDM4Z_) which reads the current brightness and sends a payload with a calculated new brightness level every 500 ms, and while it works fluently there must be a simpler way.

IKEA and Hue remotes can only send on, off, start increasing brightness, stop increasing, start decreasing brightness and stop decreasing. This must mean that all Zigbee bulbs have a built in dimmer that understands these commands. How do I send these specific dimming commands directly from my ConBee II? (e.g. start increasing without setting the final brigtness level in advance).

## Screenshots
N/A

## Environment
 - Host system: Raspberry Pi 4 8GB
 - Running method: Marthoc Docker container on Raspbery pi OS
 - Firmware version: 264A0700
 - deCONZ version: 2.05.76
 - Device:  ConBee II
 - Do you use an USB extension cable: yes

### deCONZ Logs
N/A

## Additional context
N/A

Smanar, 2020-09-26:
Hello, perhaps using the command "move to level with step", but I don't think this command will be available throught the API.

DIYtechie, 2020-09-26:
HI :) Not sure I understand the step to level part - I do not want to move to a specific level. I just want the bulb to start dimming and then I would like to tell it to stop with a separate command. Without using triggers or loops. Exacly like the remotes do when paired directly.

Smanar, 2020-09-26:
This command permit to set a level based on previous one

[image]

DIYtechie, 2020-09-26:
> This command permit to set a level based on previous one
> 
> [image]

Looks like halfway there. It's an improvement that I wouldn't need to get the current brightness first, but I would like to have fluent dimming without needing to send a "step up with step size 20" every 500 ms. Is there a "step up until stopped" command? Can I use the step command from Node-RED?

ebaauw, 2020-09-26:
The wireless remotes send a _Move_ command.  This command take a rate and direction (_Up_ or _Down_) as parameters.  You can send it from the GUI but not from the API.  Setting `state.bri` sends a _Move to Level_; setting `state.bri_inc` sends a _Step_.

For most smooth dimming, let the dimmer control the lights directly (by adding them to the associated group).

The Hue app creates Hue bridge rules with an almost as smooth experience.  They set `bri_inc` on _Initial Press_ (x000) and on every _Hold_ (x001) event, with some optimised values for the increment and for the transitiontime.  That will only work for the Hue dimmer and the Hue button; no other switches send repeated _Hold_ events.

DIYtechie, 2020-09-29:
> The wireless remotes send a _Move_ command. This command takes a rate and direction (_Up_ or _Down_) as parameters. You can send it from the GUI but not from the API. Setting `state.bri` sends a _Move to Level_; setting `state.bri_inc` sends a _Step_.
> 
> For most smooth dimming, let the dimmer control the lights directly (by adding them to the associated group).
> 
> The Hue app creates Hue bridge rules with an almost as smooth experience. They set `bri_inc` on _Initial Press_ (x000) and on every _Hold_ (x001) event, with some optimised values for the increment and for the transitiontime. That will only work for the Hue dimmer and the Hue button; no other switches send repeated _Hold_ events.

Thank you for the explanation. I understand it to mean that I cannot exactly imitate a wireless remote from Node-RED since the API cannot send a _move_ command. How about _move to level_  (state.bri)? Is that possible from the API and how to stop a _move_ from the API (to stop the up or down movement)?

The reason I can't just let the _dimmer remote_ control the lights directly is that I use non-Zigbee smart wall switches to control my Zigbee lights.

ebaauw, 2020-09-29:
> Is that possible from the API and how to stop a move from the API (to stop the up or down movement)?

Set `bri_inc` to 0.  The _Stop_ aborts a  _Move_ in progress as well as a _Step_.  I never tried it on a _Move to Level_ (with a longer transition time), but it might just work.

> The reason I can't just let the dimmer remote control the lights directly is that I use non-Zigbee smart wall switches to control my Zigbee lights.

If they fire continuously while held, you could issue `bri_inc`, cf the Hue app rules.  If they only fire on press and release, you’d need to set `bri` to 254 or 0 (depending on up vs down) on press, computing `transitiontime` from the delta between current and target brightness; and `bri_inc` to 0 on release.

DIYtechie, 2020-09-30:
Thanks. Looking very much forward to trying this out 👍🏻

DIYtechie, 2020-10-01:
> Set `bri_inc` to 0. The _Stop_ aborts a _Move_ in progress as well as a _Step_. I never tried it on a _Move to Level_ (with a longer transition time), but it might just work.

I can confirm that this works :) I'll do some comparisons of smoothness, but in any event, please add the _Move_ command to the API :) This would just make smooth dimming so much easier. No need for calculating transition times.

inconsx, 2020-10-15:
> The wireless remotes send a _Move_ command. This command takes a rate and direction (_Up_ or _Down_) as parameters. You can send it from the GUI but not from the API. Setting `state.bri` sends a [...]

Does somebody has the best setup for dimming hue bulbs with hue smart button?
Dimming is a mess for me right now. When I dimm up (long press) the brightness is flipping between up and down.

Thanks!

Mimiix, 2020-10-15:
@inconsx You are free to open your own issue on this one, or ask in #advice on discord. I'd like to keep issues clean.

closed 2020-10-15 by DIYtechie