dresden-elektronik/deconz-rest-plugin #3323: Device Add support for Stelpro Orleans Fan (SORB)

goldriver2, 2020-09-26:
## Device

- Product name: Stelpro Orleans Fan
- Manufacturer: Stelpro
- Model identifier: SORB
- Device type :  Thermostat
  

## Screenshots
[image]
[image]
[image]

Mimiix, 2020-09-28:
Missing cluster info /screens

goldriver2, 2020-09-28:
Sorry I missed the other cluster... here they are:

[image]
[image]
[image]
[image]
[image]
[image]

Mimiix, 2020-09-28:
@SwoopX Is this complete?

goldriver2, 2020-10-02:
Is there anything missing in the provided information that prevent the device to be added ?

SwoopX, 2020-10-02:
The thermostat cluster is incomplete. For the rest, I'm good at guessing which cluster that data comes from...

goldriver2, 2020-10-03:
sorry I missed the rest of the thermostat cluster, here are all the thermostat cluster informations
[image]
[image]
[image]
[image]
[image]

goldriver2, 2020-10-06:
> The thermostat cluster is incomplete. For the rest, I'm good at guessing which cluster that data comes from...

sorry I missed the rest of the thermostat cluster, here are all the thermostat cluster informations
[image]
[image]
[image]
[image]
[image]

SwoopX, 2020-11-22:
Closing this one as it is supported with version 2.06.00.

closed 2020-11-22 by SwoopX