dresden-elektronik/deconz-rest-plugin #3324: PTVO Switch Relay 4/8

moshiko2312, 2020-09-27:
Hey
First of all thank you so much for the work you do it is just amazing

I use an 8/4 Relay box
My main use is in the Home Assistant system with a CONBEE 2 adapter
* I manage to pair the product and even see it configured in the system
The problem is when I turn on the relay after 30 seconds the on-screen indication shows that the relay is off (actually the relay is still on)
It is possible to get a solution to this issue
Thank you

I enclose some links on the relevant product

[https://www.zigbee2mqtt.io/devices/ptvo.switch.html](url)

[https://ptvo.info/zigbee-switch-configurable-firmware-router-199/](url)

[https://ptvo.info/zigbee-switch-configurable-firmware-v2-210/](url)
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

kachalex, 2020-09-27:
I have this problem too, status does not show correctly.
I will be very glad, if you solve this problems.
With ziqbee2mqqt works well.

Mimiix, 2020-09-28:
Same as #3319 , But now you need to provide all correct information. 
Please check https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3319#issuecomment-699828336

I am missing all clusters read and your deconz version info.

kachalex, 2020-09-28:
deconz version info - 2.05.82 / 14.09.2020
firmware - 26580700
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

kachalex, 2020-09-28:
If i do attributes read - takes the correct status.

[image]

What should be done? To automatically read the status?

[image]

Mimiix, 2020-09-28:
Attributes need to be read ;)

kachalex, 2020-09-28:
I have more than 30 more devices and I don't have to go in and read the status with VNC every time after each change.

SwoopX, 2020-09-30:
You're still not following the template... Regardless.

What is the issue here? You wrote:

> The problem is when I turn on the relay after 30 seconds the on-screen indication shows that the relay is off (actually the relay is still on)

What on-screen display? On the device? Does it work from a functionality perspective?

moshiko2312, 2020-09-30:
Hey
Functionally it works but there is no display in Home Assistant
The button returns to the off position
It does not perform a read mode of the Riley box

SwoopX, 2020-10-02:
This may be a HA thing where we cannot help. However, does the state get right after like 10 mins without touching it?

kachalex, 2020-10-03:
This problem starts back in Phoscon App, before forwarding to HA. The state does not become correct: not after 10 minutes, not after an hour and a day without touching it. After each status change, you must manually read attributes.

kachalex, 2020-10-03:
Here is an example how is displayed relay 8 in zigbee2mqtt:
[image]
[image]
[image]
[image]
[image]

 

0x00124b001fb3df83

Router
Description
[Multi-channel relay switch](https://ptvo.info/zigbee-switch-configurable-firmware-router-199/)
Supports
hold, single, double and triple click, on/off, type, rssi
Firmware build date
2019-06-21
Firmware version
2019-06-21
Vendor
Custom devices (DiY)
Model
ptvo.switch
0x00124b001fb3df83
Endpoint 1
Output clusters
genBasic
haDiagnostic
genOnOff
Input clusters
genBasic
haDiagnostic
genOnOff
Endpoint 2
Output clusters
genOnOff
Input clusters
genOnOff
genMultistateInput
Endpoint 3
Output clusters
genOnOff
Input clusters
genOnOff
genMultistateInput
Endpoint 4
Output clusters
genOnOff
Input clusters
genOnOff
genMultistateInput
Endpoint 5
Output clusters
genOnOff
Input clusters
genOnOff
genMultistateInput
Endpoint 6
Input clusters
genAnalogInput
Endpoint 7
Output clusters
genOnOff
Input clusters
genOnOff
genMultistateInput
Endpoint 8
Output clusters
genOnOff
Input clusters
genOnOff

SwoopX, 2020-10-04:
Eventually, the attribute reporting is missing/not working. What happens, if you follow https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to...#manually-bind-server-clusters for on/off cluster and this https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to...#manually-configure-attribute-reporting for attribute 0x0000 of the on/off cluster?

moshiko2312, 2020-10-14:
Hey
I really no longer know what to do, I have tried everything I know
I tried to perform the actions according to the pictures you attached in the link, but without success
* I would love it if someone could connect to my computer and show me exactly what I need to do.
I really want this product to be set up perfectly with the system.
Thanks for all the help
[image]
[image]
[image]
[image]

SwoopX, 2020-10-15:
It is called bind dropbox, because you can drag&drop the clusters/endpoints.

As for your last screenshot, why are you taking a different cluster than I asked for? The on/off cluster is important here and attribute 0x000.

kachalex, 2020-10-16:
Reading config file
[image]

ICTSMART, 2020-10-16:
Hello all, I thought this issue was related to my specific setup, nevertheless I have exactly the same findings.
I am running the following releases / firmware:

RaspBee II
Version 2.05.81 / 9/14/2020
Firmware 26610700
DeDONZ is running as a container on Hassio

When I toggle attribute '006 On/Off' attribute in deCONZ VNC the corresponding port on the device responds  correctly. Unfortunately there is no feedback to Phoscon at all.
Only after reading the attributes in VNC, Phoscon updates its status.

Please let me know what other info you require.

Rgds.

SwoopX, 2020-10-16:
@kachalex Have you previsouly successfully done the binding? In your last screenshot, was that the server on/off cluster (in blue letters)?

kachalex, 2020-10-17:
I have not done a binding before. I do not know how to do this.

P.S
In your last screenshot, was that the server on/off cluster (in blue letters)? - Yes.

SwoopX, 2020-10-17:
If you are not able to drag&drop the server on/off cluster of the device on the left and the Endpoint 1 of the coordinator on the right then I cannot help you.

kachalex, 2020-10-17:
Can you help me how to do this?

SwoopX, 2020-10-17:
I did [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3324#issuecomment-703307230) and [here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3324#issuecomment-710814755).

ICTSMART, 2020-11-08:
Hello SwoopX, 

I've read the instructions on how to do the manually binding and did so for this firmware (PTVO). So in my case I used (probably) the same firmware (configured 1 input and 1 output via the PTVO tool). The Output is visible as cluster 0006 On/Off. So sending an off or on command does update the status of the device, what seems missing is what you mentioned above: "...Eventually, the attribute reporting is missing/not working". Therefore I've enabled all debug options in deCONZ (running as add-on in Hassio), hoping this provides further information for you guys to dive in. Please let me know how I can assist further.
I saw Mimiix is also waiting for further info - so let me know what makes debugging easier for you.

[logging_raspbeeII_PTVO.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5506773/logging_raspbeeII_PTVO.txt)
[image]

SwoopX, 2020-11-08:
The binding looks alright, but you also need to configure attribute reporting for 0x0000 of the on/off cluster

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to...#manually-configure-attribute-reporting

ICTSMART, 2020-11-09:
Hello, I probably missed that before. New attempt with all logging enabled and report interval set as depicted in attachment.
Maybe good to know that the firmware tool used here, allows you to set 8 inputs and 8 outputs. In this case I've just defined 1 input and 1 output and 'linked' them via 'Link to out 1'. Without 'linking' them in the tool, the node shows up with both a '0006 On/Off' cluster and a '0012 Multistate input (basic) (2nd picture).
[image]
[image]
[image]
[logging_attribute_report.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5510499/logging_attribute_report.txt)

ICTSMART, 2020-11-09:
Now when reflashing with the 'unlinked' output 1 and input 1, I still can set the attributes on 0006 On/Off. When I try to do the same for the 0012 Multistate inout (Basic) and try to modify the attribute 0x0055 "present value" of which I suspect should be reflecting the status of the input, I get a UNSUPPORTED_ATTRIBUTE back from deCONZ. Same happens for instance for attribute 0x006F. See pictures below. Nevertheless I've created a log when generating an input signal from towards the coordinator. 
[image]
[image]

Report corresponding to 'unlinked' and a triggerend 'on/off'command' initiated on the device itself: 
[logging_attribute_report_unlinked.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5510641/logging_attribute_report_unlinked.txt)

ICTSMART, 2020-11-09:
According to its documentation, the firmware should report the state of an independent button via the “ZCL_CLUSTER_ID_GEN_MULTISTATE_INPUT_BASIC” cluster and the “PRESENT_VALUE” attribute:
1 – single click.
2 – double click.
3 – triple-click.
4 – long click.

SwoopX, 2020-11-09:
Ok, I guess we're getting closer now. In the 2nd log file, I can see that the device is reporting for cluster 0x0006 and 0x0012, so that's good.

For the on/off events, are those now reflected in Phoscon (or in the REST API) as well?
The multi state input basic reports are currently not taken into account, this will probably require some code changes.

Has the debug log taken with `dbg-info=2`? I'd expected to see some more information regarding the attribute reporting, the exact payload to be precise. I'd generally recommend to collect like 10mins of log data, play around with the device and share the full log, as there's more usefull data dwelling in the depths.

Ideally, you'd also be able to compile a test version of the REST API plugin, given you do not run any HA or docker.

ICTSMART, 2020-11-09:
Ok. Let me change the dbg-info to 2 and I will play around with the device in order to collect a full log.
As I am running the official deCONZ docker add-on for Home Assistion, I guess I am limited for what I can do there.
I'll be back shortly with better logging.

ICTSMART, 2020-11-09:
Please find attached an assembly of multiple log snippets. As I am not aware of a complete logfile in the deCONZ container, I tried to copy/paste relevant information here. I adjusted the debugging level to 2.
[logging_snippets_DBG_lvl2.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5512751/logging_snippets_DBG_lvl2.txt)

sdhawade, 2020-12-03:
Hi I am also facing the same problem, the data appears to reach co-ordinator, I can see the on and off status as 1001 and 1000 respectively in the payload, but no status change in Phoscon. Added log below
-- Light On --
20:29:04:800 Node data 0x00124b001fc64555 profileId: 0x0104, clusterId: 0x0006
20:29:04:801 0x00124B001FC64555: update ZCL value 0x04/0x0006/0x0000 after 0 s
20:29:04:801 ZCL attribute report 0x00124B001FC64555 for cluster: 0x0006, ep: 0x04, frame control: 0x00, mfcode: 0x0000
20:29:04:801    payload: 00001001

--Light Off--
20:34:44:803 Node data 0x00124b001fc64555 profileId: 0x0104, clusterId: 0x0006
20:34:44:803 0x00124B001FC64555: update ZCL value 0x04/0x0006/0x0000 after 0 s
20:34:44:804 ZCL attribute report 0x00124B001FC64555 for cluster: 0x0006, ep: 0x04, frame control: 0x00, mfcode: 0x0000
20:34:44:804    payload: 00001000

ICTSMART, 2020-12-25:
Not sure whether this topic should be closed yet, as there's quite some interest in having PTVO support in Deconz.
@SwoopX any suggestions on how to continue?

sdhawade, 2020-12-26:
I had a brief look at the code, looks like there would be multiple changes required to add this to the list of supported devices.  Can anyone point me to raise a request to add support ?

ICTSMART, 2021-01-13:
@SwoopX , @Mimiix What can I do to provide you guys with further info or testing? Please let me know how I can contribute in getting the PTVO firmware supported.

SwoopX, 2021-01-15:
Well, I pretty much did what I could from here. What else do you expect or what is not working respectively? As I recall, this was initially about the state changes in deconz.

The challenge with this device is, that it seems to be highly configurable and I do not have any clue what you need to do to "have it working as expected". To have the ultimate insight, somebody from the devs willing to have this covered would probably need to have the device in hands and do some traffic sniffing.

ICTSMART, 2021-01-15:
@SwoopX You are absolutely right, this software is indeed highly configurable. Nevertheless from what I have read - and based on my own experience is that 'inputs' from this device are not reflected in Deconz. Just that would be a major step forward as we can have the actual status fed back towards Deconz. Who from the devs would you suggest I contact? As I am using these devices for quite some time now, especially in combination with the Raspbee I and Raspbee II, my wild guess is that I should be able to provide the necessary sniffing input. Are you aware of the fact that this device is fully supported in Zigbee2Mqtt? I could imagine some info is reusable to make this also work for Deconz (all be it with limited functionality).

wvuyk, 2021-02-08:
Her I have an user with the 4 relais version of the device. What I am seeing here is that the webhook keeps sending a second event each time an "on":true command has been send: 

```
11:27:34.7002860 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"31","r":"lights","state":{"alert":null,"on":true,"reachable":true},"t":"event","uniqueid":"00:12:4b:00:22:02:80:97-03"} new queue size is 0 on 00212EFFFF05ED63
11:27:34.7537991 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"31","r":"lights","state":{"alert":null,"on":false,"reachable":true},"t":"event","uniqueid":"00:12:4b:00:22:02:80:97-03"} new queue size is 0 on 00212EFFFF05ED63

```

Any idea why does this happen?

OskarL83, 2021-03-05:
Hi,
I also have one of these PTVO relay units. 
I'm using it in Home Assistant with Conbee II as coordinator - but not with deCONZ/Phoscon in between. I'm using the ZHA integration instead, which is using zigpy-deconz library to communicate with the Conbee.
With this setup it's working a bit better than with deCONZ/Phoscon. The state of the relay outputs are mostly reported correctly, but sometimes the state changes to off in HA after some minutes, even if the switch is on and has not changed. Don't know why.

OskarL83, 2021-03-06:
Any progress on this?

mamrai1, 2021-03-06:
It would be great if deconz supported also the analog inputs like device temperature and source voltage.. I use 2 of them as routers with backup battery power and they are working great
[image]
[image]

mamrai1, 2021-03-08:
i have just found this issue -  General ZBMini issue. #4016 - It seems that the problem was similar with the on/of cluster and attribute 0x0000!!

closed 2021-04-10 by github-actions[bot]

rafaeledi, 2021-05-18:
hello, I have a 4 relays device, but unfortunately it does not report the status of the relay (deconz 6.9.0.). has someone already solved the problem without going to the competition. I've tried everything but still no solution.